WEC-Sim/WEC-Sim #1077: Issue with installing BEMIO

mprevisic1, 2023-06-28:
**Describe the bug**
I cloned Bemio and try to run install.py, but it gives me an error message that it cannot find a module called bemio._version_

**WEC-Sim file(s)**
NA

**To Reproduce**
1. I installed the latest version of python (I am on 3.11.4)
2. I used pip install to add the required libraries including: numpy, script, matplotlib, progressbar2, astrophy, h5py
3. I cloned Bemio from GitHub using the git install command
3. I used the command line to execute install.py.

**Expected behavior**
Installation

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10 Pro

**Additional context**
NA

mprevisic1, 2023-06-28:
I downloaded instead of cloning the repository and now it seems to find the required file during install (the download version seems to have many more files then the cloned version). However, now I have issues running the install_test.py file. Here is the error:
[image]

dav-og, 2023-06-29:
Hi @mprevisic1
Its been a while since we supported the Python version of BEMIO - we've only been working on the MATLAB version over the past couple of years. However, I have heard from a few people recently that they would benefit from having the Python BEMIO "revived" - is that something you need in your project? or could you just use the MATLAB version of BEMIO?
Cheers,
Dave

mprevisic1, 2023-06-29:
David,

I did not know that there was a Matlab version. Where would I find that?

Thanks,

Mirko

mprevisic1, 2023-06-29:
> Hey Mirko, its included with WEC-Sim here:
> https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions/BEMIO
>
> You should be able to use BEMIO after installing WEC-Sim with the
> addWecSimSource.m script.
>
> You can test it by running some of the examples in WEC-Sim/examples/BEMIO
> (e.g. this one
> ).
>
> Let me know if you get that working ok :)
>
> Cheers,
> Dave

Dave,

I was wondering why Bemio was a python code - seemed like unnecessary
complexity :-) I would suggest you update the Bemio documentation online (
http://wec-sim.github.io/bemio/) and point to the fact that Bemio can now
be found in Matlab.

Thanks,

Mirko

closed 2023-07-12 by kmruehl