WEC-Sim/WEC-Sim #1078: Adding Nemoh input files for examples

mprevisic1, 2023-06-30:
**Is your feature request related to a problem? Please describe.**
I am trying to do a 2-body case in Nemoh. The solution runs, but then I have Bemio compatibility issues. Specifically, Nemoh does not generate the _0 and _1 files. I suspect this is because I do not setup my 2-body case in the same way as you have. I am trying to run a case that is very similar to the nemoh "cubes" case.

**Describe the solution you'd like**
It would be nice to have the Nemoh matlab codes that were used to generate the mesh and do the Nemoh run. I only see the format of the input to Bemio. Unfortunately the Nemoh documentation itself does not include how that case was setup using their matlab wrapper.

nathanmtom, 2023-07-12:
@mprevisic1 Thank you for your question about the Nemoh example cases that we have in WEC-Sim and I'll do my best to support your questions. 

Based on this issue thread on NEMOH's Github, https://github.com/LHEEA/Nemoh/issues/6, the latest version of NEMOH will only generate the hydrostatics after using the built-in mesh generator (Mesh). 

Please let us know if you are still having difficulty getting BEMIO to accept Nemoh outputs after looking at NEMOH's issues board.

Cheers, 
Nathan

nathanmtom, 2023-07-27:
@mprevisic1 I wanted to reach out to ask if my response above helped to resolve your issue? If not, please provide an update to this thread or else we will plan to close this issue at the end of the week.

closed 2023-07-28 by nathanmtom