WEC-Sim/WEC-Sim #1082: Importing Nemoh files[BUG]

mprevisic1, 2023-07-07:
**Describe the bug**
Fails to import Nemoh files. readNEMOH.m line 173 seems to have an error. The code opens the Excitationforce.tec file, but then you seem to be searching for 'Diffraction force' term. Of-course it does not find that terms and returns an empty output and then fails on line 182. I replaced 'Diffraction force' with 'Excitation force' and it seems to run. Can you check if that is the correct change? Here is the error I am getting:
[image]

After correcting this issue, I get an index error in normalizeBem on line 32.
[image]

I am running Nemoh with a 2-body system that is similar to your cubes example. I tried to run the bemio.m file in the cubes example as well and I seem to be getting the same problem. It seems like the indexing issue has to do with having a two body system instead of 1? 

**WEC-Sim file(s)**
readNEMOH.m

**To Reproduce**
See above

**Expected behavior**
No errors

**Screenshots**
See above

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB Release: 9.12.0.2170939 (R2022a) Update 6
 - [WEC-Sim Release] NA

**Additional context**
For some reason, it does not accept my files in the submission below, but I am happy to pass along all my input files so you can reproduce the issue.

nathanmtom, 2023-07-12:
@mprevisic1 Thank you for the question. I'll review the files you are pointing to but will need more information from you. 

1. What version of NEMOH are you using? 
2. What version of BEMIO are you using?

nathanmtom, 2023-07-12:
Thanks for that information @mprevisic1, I have a few other questions then: 

- In the error message above, you appear to be running our 'Cubes' example but wanted to confirm the error is coming from using our examples or you re-ran the same input files in the latest version of NEMOH and are getting this error. Also just to confirm are you using the 'master' or 'dev' branches.
- The WEC-Sim NEMOH examples were generated several years ago and we have not updated or verified that with the recent release of NEMOH v3.0 has not implemented changes in language that would break our readNEMOH parser. The image below is the top sections of the ExcitationForce.tec and DiffractionForce.tec files in the 'Cubes' example. Both do list "Diffraction force" and the new release might have changed the ExcitationForce.tec to "Excitation force'.
   - Please confirm if this is what you are experiencing when you run NEMOH and reviewing your output files. 
 
[image]

- If the point above is indeed the issue you are encountering then I would agree that the first parser language would need to be updated to reflect changes in NEMOH output files based on the v3.0 release. 
   - Also if this is the case, would you be willing to share the sample output files with us as we currently do not have sample output files to cross reference.

Cheers, 
Nathan

mprevisic1, 2023-07-18:
Nathan,

Please find my responses below.

Thanks,

Mirko

> Thanks for that information @mprevisic1 <https://github.com/mprevisic1>,
> I have a few other questions then:
>
>    - In the error message above, you appear to be running our 'Cubes'
>    example but wanted to confirm the error is coming from using our examples
>    or you re-ran the same input files in the latest version of NEMOH and are
>    getting this error. Also just to confirm are you using the 'master' or
>    'dev' branches.
>
> Response: Yes, I am using the master branch. The same error appears with
the cubes and my own example. For consistency, I am including the Nemoh
matlab code for my own example.

>
>    - The WEC-Sim NEMOH examples were generated several years ago and we
>    have not updated or verified that with the recent release of NEMOH v3.0 has
>    not implemented changes in language that would break our readNEMOH parser.
>    The image below is the top sections of the ExcitationForce.tec and
>    DiffractionForce.tec files in the 'Cubes' example. Both do list
>    "Diffraction force" and the new release might have changed the
>    ExcitationForce.tec to "Excitation force'.
>       - Please confirm if this is what you are experiencing when you run
>       NEMOH and reviewing your output files.
>
> Response: Yes, that is correct. The ExcitationForce.tec file now says
"Excitation Force".

>
>    -
>
> [image: image]
> <https://user-images.githubusercontent.com/7635064/253112723-c7389723-efdf-4ce4-aee0-97d5b6552d55.png>
>
>    - If the point above is indeed the issue you are encountering then I
>    would agree that the first parser language would need to be updated to
>    reflect changes in NEMOH output files based on the v3.0 release.
>       - Also if this is the case, would you be willing to share the
>       sample output files with us as we currently do not have sample output files
>       to cross reference.
>
> Response: I was not entirely clear which Nemoh outputs you are utilizing
in the parser. I am including the Nemoh input file that will allow you to
run the case and produce the 2-body outputs.

Cheers,

> Nathan

%% Runs Nemoh for a 2-Body hinged barge with two rectangular absorber bodies. Includes hydrodynamic interaction terms for 12 x 12 output matrix (diagonals)
%% Created by: Mirko Previsic
%% Date: 6-22-23

clc
clear all
close all

[pathstr,~,~] = fileparts(mfilename('fullpath'));
addpath(genpath(pathstr));                          % Include the subfolders in the Matlab PATH

ID_PLOT_RESULTS = 1;
ID_QTF = 0;                                         % Flag to activate QTF computation (0 or 1)

outputdir = ['C:\Data\nemoh-windows-x64-v3.0.1' filesep 'output'];   % Update this output files location
projdir   = [outputdir,filesep,'Hinged_Barge'];

% Check that Nemoh is available
assert(FindingNemoh(ID_QTF, true))

ID_HydrosCal=1;     % A switch,1 computes hydrostatics, inertia, kH.
                    % if RAO will be computed, see input in
                    % Nemoh.cal, set ID_Hydroscal=1

% Define Body Dimensions for Box
L = 30;            % Length x axis
W = 15;            % Width y axis
Draft= 2;          % Draft of Box
  
%% First Barge Body
OL = 0;            % Length offset to move box in +X direction
Body = 1;          % Set body index
% left face
X(Body,1,:,:)=[-L/2+OL,0,-Draft;
    -L/2+OL,-W/2,-Draft;
    -L/2+OL,-W/2,0;
    -L/2+OL,0,0];
% Bottom face
X(Body,2,:,:)=[-L/2+OL,-W/2,-Draft;
    -L/2+OL,0,-Draft
    L/2+OL,0,-Draft;
    L/2+OL,-W/2,-Draft;];
% right face
X(Body,3,:,:)=[L/2+OL,0,-Draft;
    L/2+OL,0,0;
    L/2+OL,-W/2,0;
    L/2+OL,-W/2,-Draft] ;
% front face

X(Body,4,:,:)=[-L/2+OL,-W/2,-Draft;
    L/2+OL,-W/2,-Draft;
    L/2+OL,-W/2,0;
    -L/2+OL,-W/2,0];
      
ncoarse(Body)=length(X(1,:,1,1));          % number of coarse panels
tX(Body)=1;                                % 0 no translation applied to the Mesh
xyzCoG(Body,:)=[0,0,-Draft/2];             % position of gravity centre
nfobj(Body)=600;                           % target number of panels 

%% Second Barge Body
OL = 32;                                    % Length offset to move box
Body = 2;                                   % Increment Body index
% left face
X(Body,1,:,:)=[-L/2+OL,0,-Draft;
    -L/2+OL,-W/2,-Draft;
    -L/2+OL,-W/2,0;
    -L/2+OL,0,0];
% Bottom face
X(Body,2,:,:)=[-L/2+OL,-W/2,-Draft;
    -L/2+OL,0,-Draft
    L/2+OL,0,-Draft;
    L/2+OL,-W/2,-Draft;];
% right face
X(Body,3,:,:)=[L/2+OL,0,-Draft;
    L/2+OL,0,0;
    L/2+OL,-W/2,0;
    L/2+OL,-W/2,-Draft] ;
% front face

X(Body,4,:,:)=[-L/2+OL,-W/2,-Draft;
    L/2+OL,-W/2,-Draft;
    L/2+OL,-W/2,0;
    -L/2+OL,-W/2,0];

ncoarse(Body)=length(X(1,:,1,1));          % number of coarse panels
tX(Body)=1;                                % 0 no translation applied to the Mesh
xyzCoG(Body,:)=[OL,0,-Draft/2];            % position of gravity centre
nfobj(Body)=600;                           % target number of panels 

%% Set Nemoh Run Parameters
nBodies = 2;
wmax=3;                                     % Max omega in rad/s 3 rad/s ~ 2s
nbfreq=200;                                 % number frequency
w= linspace(wmax/nbfreq,wmax,nbfreq)';      % min w [rad/s], max w [rad/s], Nw
wavedir=0;                                  % angle of the incident waves
depth=100;                                  % water depth (m)
QTFInput=[ID_QTF,2];                        %[Switch,Contrib]
Mesh(nBodies,ncoarse,X,tX,xyzCoG,nfobj,depth,w,wavedir,QTFInput,projdir);

% Output Prompts
disp('Meshing done!')
disp('Check and adjust the produced Nemoh.cal in the project directory!')
disp('Press enter for the NEMOH computation')
pause;

% %-------Launch Nemoh Calculation------------
[Idw,w,A,B,Fe]=Nemoh(projdir,ID_HydrosCal,ID_QTF); % Call the function Nemoh.m

nathanmtom, 2023-07-27:
@mprevisic1 Thank you for clarifying and providing those files. We have installed the latest version of NEMOH and can confirm the change in file headers. We will now work on updating the BEMIO Nemoh parser to check for NEMOH version and adjust prior to importing the hydrodynamics. We will notify you when an update has been released.

nathanmtom, 2023-07-28:
@mprevisic1 I wanted to provide an update that PR#1087 includes an updated readNEMOH.m file to reflect changes to the output and input files used in this script. However, I did rerun our cubes example in NEMOH v3.0 but did not encounter the indexing error you have mentioned receiving. Perhaps the other edits made in the readNEMOH.m file will resolve the issues, but cannot comment as unsure what other modifications you might have made to readNEMOH.m. 

Or if you are still experiencing issues with importing the files, if you can share your input/output files we can try to identify the continuing issue.

mprevisic1, 2023-07-30:
I am not sure if subsequent errors are the result of my own erroneous edits
to your codebase or are a real issue. There are also several un-documented
process steps (going from Nemoh output to Bemio) and it is certainly
possible that I missed a specific step in my process. The other issue is
that Nemoh only provides the output files for the two-body cubes example
(no Matlab input wrapper file) and it is unclear how exactly these
outputs were generated - so again it is entirely possible that I messed up
on a process step.

I previously supplied my nemoh input matlab wrapper. This should allow you
to run my case and generate the appropriate Nemoh outputs (and respective
input files to Bemio) and allow you to trace any potential errors.
Alternatively, if you do have Nemoh input files for the cubes example
(ideally the matlab wrapper file), I could use that as an example to guide
my processing steps.

Thanks,

Mirko

nathanmtom, 2023-08-09:
@mprevisic1 Thank you for the feedback to my questions. 

Based on running your code and reviewing the script, I see a potential issue if you were running the mesh function for both bodies and not relabeling the hydrostatics and KH.dat files. Per our readNEMOH documentation, 
http://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#functions.BEMIO.readNEMOH, we need to rename these files if you have multiple bodies and if two are not present there can be an issue with body indices. Are you able to confirm if this process was completed or not? If it was completed and the above issue is still present then I'll continue trying to identify what might be wrong in the workflow.

mprevisic1, 2023-08-09:
Yes, I did rename the files as instructed in the Nemoh documentation. Nemoh
seems to create hydrostatics and KH.dat files with values that are valid
for a single composite body. I did not fully check this, but it seemed like
it was for example calculating the Cg for the two bodies as if they were a
single body and rigidly connected. So I manually changed the values by
editing the files. The obvious question becomes if there are two bodies
that have different properties, just copying these files will create the
wrong values as inputs to your hydrostatics/system dynamics, so it is not
documented how to manipulate these two files to reflect reality.

Thanks,

Mirko

mprevisic1, 2023-08-22:
Looks like all the issues have to do with changes in the output format in the latest Nemoh version. I had to fix the following issues:
1. in readNemoh.m, line 95 I had to change the selection process, reading in 4 instead of 3 temp variables and utilizing temp 2-4 instead of 1-3. The new Nemoh output format adds an additonal value to select the input format, which messes up the order in which it reads in values. 
2. in readNemoh.m, line 172, replace Diffraction Force with Excitation Force.
3. in the Nemoh.cal output file, add the description "!Name of Mesh File". The new output seems to have dropped that. 

Once these three items are corrected, the code works fine, including with the included Nemoh matlab examples. 

You may want to fix this in a better way so that Bemio automatically detects the Nemoh version and adjusts the parser accordingly.

nathanmtom, 2023-08-22:
@mprevisic1 Thank you for providing the steps you completed to resolve the readNEMOH.m issues. I wanted to confirm the above points as I was able to run your script yesterday which highlighted the NEMOH v3.0 auto-generation of the NEMOH.cal file does leave out '! Name of Mesh File' which did not come up previously when updating WEC-Sim's NEMOH input files. 

I want to mention that points 1 and points 2 above are implemented in PR #1105 and the parser does look to see if there is 3 or 4 variables in line 95 and updates accordingly. 

The concern with '! Name of Mesh File' will be flagged and our team will decide if the parser might be updated to search for the .dat file label or we can request a PR into the NEMOH GitLab account to add the comment automatically. 

Based on this discussion and Issue #1103, the current issue has been resolved and the new readNEMOH.m is compatible with the new NEMOH v3.0 with the caveat about the '! Name of Mesh File' comment in the NEMOH.cal. Please let us know if we can now close this issue.

mprevisic1, 2023-08-22:
Yes, please close this out. Thanks for the update on PR #1105.

Mirko

closed 2023-08-22 by nathanmtom