WEC-Sim/WEC-Sim #1083: Reactive control in WECSim

Amartinezperurena, 2023-07-10:
Hi! 
I would like to share with you the problem that I´m having in WECSim and openly recieve your comments or help. In the following sentences I describe the application and the exact problem. 
**Adjustment of PTO parameters for reactive control  **
**Issue**
_I was trying to implement a reactive control in an Ellipsoid point absorber using a transational PTO. But WECSim doesn´t allow me to introduce negative stiffness coefficients (see the error in the screenshot bellow)._
**To Reproduce**
1. _Set a possitive damping and negative damping coefficients in the initialization process._
2. _Run the simulation to send the coefficients to the transational PTO._
3. _Scroll down in MATLAB command window and see the error shown in the scrrenshot,._
**Expected behavior**
_What I expect to get after setting the coefficients in the controller is the WEC's hydrodynamic forces, displacement and power profiles._
**Screenshots**
_Please find attached a screenshot of the error message from the MATLAB command line._
[image]
**Workstation Information :**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2022b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v5.0.1]
Thank you very much in advance.

nathanmtom, 2023-07-10:
@Amartinezperurena Thank you for your question and I will do my best to respond. 

You are correct, the PTO/Translational PCC (Prismatic Joint) does not accept negative spring coefficients that could be used to implement complex conjugate control. In order to implement complex conjugate control with a negative spring stiffness you will need to use the Translational PTO Actuation Force Block (circled in red below) and implement your own feedback loop to go from motion to PTO force.

[image]

A template to use for such an implementation is our WECCCOMP model which can be found in the WEC-Sim Applications Repository:

https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/WECCCOMP/WECCCOMP

Cheers, 
Nathan

jtgrasb, 2023-07-12:
@Amartinezperurena We also recently developed some controls applications cases. They are currently on the [dev branch of WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/dev/Controls). These examples include implementation of a reactive controller with complex conjugate control demonstrated for mechanical power maximization. The [WEC-Sim dev documentation ](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#controller-features) explains the control examples.

nathanmtom, 2023-07-12:
@Amartinezperurena Please let us know if you have any further questions. If we do not hear from you by the end of this week we are going to close this issue.

closed 2023-07-14 by nathanmtom