WEC-Sim/WEC-Sim #1084: [Theory or Implementation] Significant oscillations due to check valves

TiphaineBr, 2023-07-24:
**Is your question request related to a problem? Please describe.**

Significant oscillations are observed for flows and pressures due to the check valves in the hydraulic model.

**Personal project context**

I’m currently attempting to adapt the WEC-Sim code of the latest version (v 5.01) to model a wave powered machine consisting of a float with an hydraulic cylinder actioned by a PTO using a rope (represented as a cylinder on the WEC-sim model). This machine has already been modeled with WEC-Sim v 2.0 
However ,I am facing some challenges with certain functionalities in this new version.

For the simulation :
- The float is considered as a hydrodynamic body (simu.nonlineardt = 2), with viscous drag and nonlinear calculation turned on (body(1).nonlinearHydro=2)
- The cylinder is treated as a non-hydrodynamic body (no .h5 file), with any hydrodynamic forces acting on it (as its volume is very small compared to the buoy).

**Describe the theory or implementation approach that you have a question**

My hydraulic model is using check valves that I tried to dimension to accurately represent reality. To achieve this, I referred to their technical information (head loss / flow curve) to adjust the check valve block parameters.

With these parameters, the overall performances and results seem to align with what I had previously obtained with WEC-Sim v 2.0. However, I observe significant oscillations in flow and pressure due to the check valves. These oscillations are affecting the simulations, resulting in deviations from the model and leading to aborted simulations (aborted simulation after 169s with dt = 0.00005s).

Here is an example of these oscillations, observed during a simulation with the following parameters : 
simu.dt = 0.00005; simu.nonlinearDt = 0.01; simu.rampTime = 0.5; simu.stateSpace = 1   ; waves = waveClass('irregular')

[image]
[image]

Despite attempting to address the oscillations by adjusting the nonlinear time step and nonlinear hydrodynamic forces calculation, it appears that these changes do not have any significant influence. However, I did observe that reducing the time step can help mitigate the oscillations, but this approach results in unacceptably long simulation times. To achieve acceptable simulation times, I aim to find an effective solution that allows me to run the simulation with a a time step simu.dt = 0.0001. 

Below are the results with a smaller time step (simu.dt = 0.00001) : 

[image]
[image]

**Describe the type of conclusion or resolution you are looking for**

As of now, I am seeking insights into the source of these oscillations to manage them effectively.
Any suggestions or explanations regarding the origins of these oscillations and potential strategies to address them would be greatly appreciated. Additionally, I am curious to understand the guidelines for choosing check valve parameters (k1, k2, Amax, Cv). I have already attempted to dimension them based on technical information, but I am uncertain if this is the correct approach.

**Additional context**

N/A

dforbush2, 2023-07-26:
Hi @TiphaineBr 

This is a tough question. Can you please share all the model files from your case directory? Please also include the *h5 file if able. This will allow me to explore this directly. 

If you are unable to share this model, I do have some general suggestions that might help. In general, the action of a check valve is very numerically stiff: as the valve opens, one expects the pressure to drop significantly, and the reverse is true on valve closure. This can create a situation where, even at very small time-steps, the valve is "open" at one instant and "closed" the next, resulting in the large oscillations you are seeing. 

* use a non-zero closure/opening time and a ramp in opening area. see https://www.mathworks.com/help/hydro/ref/checkvalve.html for a discussion here. 
* if you have the necessary license, try the 'daessc' solver, which is especially effective with Simscape components. If not, I recommend you try a stiff (e.g., ode15s) or a moderately stiff (ode23tb) solver which usually handles this sort of dynamic better. 
* it is not usually a robust solution, but it is worth trying a fixed-step solver (i.e., ode4) at a range of time steps. This can avoid the problem of a variable step solver getting "stuck" at very small time steps near a discontinuity like a check valve closure.

TiphaineBr, 2023-08-01:
Hi @dforbush2 ! 

Thank you very much for helping on this issue. Unfortunately I'm not able to share my model files at this time.

I have started implementing the troubleshooting techniques you suggested, but I still have some questions regarding them : 
- When you mention "non-zero closure/opening time," could you please provide more clarification? I have attempted to use classical check valves with ramp in opening areas instead of WEC-sim check valves with the tan function, but it hasn't improve the results.
- About trying different solvers, I was using "ode4" (ode3 and ode8 are working as well). However, when attempting to switch to a variable time step solver such as ode45, ode15s, ode23tb, ode23t, or daessc, the simulation consistently crashes. There are no error messages, but the floater is pushed out, and the simulation either does not run or runs extremely slowly. I am uncertain about the root cause of this issue, so any insights you have would be greatly appreciated.

For the BEM I use NEMOH, but I am uncertain if the problem is related to it.

Once again, thank you for your assistance on this issue !
Tiphaine

dforbush2, 2023-08-01:
* This is what I meant. Are you using the PTO-sim or the Simscape check valve in your model? You can usually define a leakage area, a cracking pressure, and a full-open pressure, with a variable area. The "more gradually" this valve actuation occurs the easier it will be for your solver.

* This is very strange and I have not encountered this behavior before. It is possible that adjusting the variable step solver parameters (see https://www.mathworks.com/help/simulink/ug/variable-step-solvers-in-simulink-1.html to start) but it is odd that the parameters suggested automatically by Simulink are resulting in this undesirable behavior. Frankly I'm not optimistic that this will solve your problem but it is worth a try. 

When faced with very unusual solver behavior in a complex model it can be helpful to use the Simulink Profiler and Solver Profiler. These are found in Simulink --> Debug (a tab on the top of the window) as drop-down options under Performance Advisor (on the far left in 2022b). To use these tools for a WEC-Sim model typically run from the command window:
1)  >>edit wecSim.m
2) In wecSim.m, place a debug point on the line beginning with "sim(...". For me this is line 45.
3) From command window, >>wecSim.m
4) this will load all your parameters into your workspace and initialize everything needed to run the model.
5) you can now open your *slx file and run using the Profiler interfaces. 

* If you cannot share your model, perhaps you can share your BEM results? It is possible cleaning up BEM results will improve performance as well. As a check: if you remove the complex PTO model and replace it with, for example, the translational PTO block with stiffness and damping values selected that are representative of your PTO system, does the model run successfully for various solver types (fixed and variable step) at a variety of time steps? If the answer to this is NO, then the problem is likely your BEM data. If the answer is YES, then we will keep working on the PTO model. 

In the past, there has been need to simplify very complex PTOs into dynamically similar but reduced forms to solve in a timely manner. Without seeing your model I cannot offer specific advice, but consider reducing hydraulic/electrical systems into nearly equivalent circuits. Look for opportunities to reduce similar parallel elements with single-element approximations. Consider reducing multiple sources of model discontinuities (e.g., check valves, static friction, logic checks) to single elements where possible. 

It is often useful to make a stand-alone version of your PTO model (just the PTO driven by ideal periodic signals, no WEC bodies) to test/verify that the simplified system still captures the desired system dynamics with the necessary accuracy.

Hope that helps.

dforbush2, 2023-08-01:
I'm going to reference this issue for future related inquiries so I will use it to share some additional general advice that might not be specifically relevant for you. 

If you are using Simscape, there is another trick I got from the developers at MW themselves specific to hydraulic systems using the idea of a "dead node". Recall that unless you specifically model sections of pipe in a hydraulic network, Simscape does not assume that there is any volume in between the connected elements. Consider a hydraulic network like this: 

[image]

Under "normal" operations, both check valves (equivalently, this could be modeled as a single pressure-relief valve between top and bottom lines) are closed and actuation at the piston (left) drives fluid through the pump (right). However, to prevent over-pressure at the pump or piston, the valves offer pressure relief and bypasses the over-pressure side. At the instant a valve opens, even if we assume a compressible fluid, Simscape does not presume any available fluid volume to provide the additional flow through the valve, which requires it to re-solve the continuity equations at each node: which, given infinitesimal available volume, will subsequently result in a solution where the pressure across the valve would not have been sufficient to open it, resulting in the oscillation you were seeing in your previous results. A solution is to add two constant volume chambers to the node including the actuating valve.

[image]

What this is functionally doing is adding a bit of compliance to the fluid system that makes things easier on the solver. Physically, we can think of it as "lumping" the fluid volume of these components into a single block. Be warned: depending on the specified volumes and fluid properties this MIGHT actually have a significant dynamic effect on your system, so it is advisable to treat dead-node volume as a tuning parameter. This can significantly speed up simulations and increase allowable solver step sizes.

TiphaineBr, 2023-08-03:
Thanks again for your help and responsiveness.

I’m using only PTO-Sim components. I have attempted to adjust the parameters (cracking pressure pMin, full-open pressure pMax, minimum and maximum open area Amin and Amax, Cd, coefficients k1 and k2) without any improvement in the oscillations. 

Following your advice, I replaced the complex PTO with a transitional PTO and the model ran successfully with both fixed and variable steps ! The problem is then likely coming from the PTO model itself. 
I also ran the solver profiler, which suggested that the ratio of solver resets to solver steps was excessively high. The profiler recommends checking the model for any continuous blocks driven by discrete blocks. Below are presented the solver reset details : 
[image]

Regarding your last advice, could you please clarify what you mean exactly? I'm not quite sure how to create a stand-alone version of my PTO model without any WEC bodies, and I'd appreciate further guidance on this matter.

Thanks, 
Tiphaine

dforbush2, 2023-08-03:
Hi Tiphaine,

Lots of solver resets are very common near anything that looks like a logic check: in this case an actuating valve looks like a logic check, so I'm not surprised to see this but I also don't have any great ideas to fix it beyond making these valves operate as "smoothly" as possible. Tagging developer @jleonqu to see if he might have any more insight here. Its hard to give any more specific advice without seeing your exact model. Generally, you're on the right track, the dynamics you are modeling are numerically difficult. 

With respect to the stand-alone model, recall that the variables into your PTO are the forces/motions of the connected WEC bodies. The advice is to remove these bodies and replace these input variables to your PTO with simulink sources, particularly oscillatory sources like sine waves in the mean/amplitude ranges you expect from your WEC. Then, placing the debug point in WEC-Sim as above, you can run the PTO-only model using the simulink GUI (you can of course also make your own initialization script if you'd prefer): you'll want to make sure you are using data logging blocks (like "to workspace") to save data for later analysis since wecSim post processing won't likely work once the model finishes. You may also need to use Simulink-Physical signal blocks (or their opposite) https://www.mathworks.com/help/simscape/ref/simulinkpsconverter.html to connect things up. 

I am also just noticing that you are using a tether connection to the sea-bed: how are you modeling this? If you are using the cable block, try, as a debug step, replacing it with translational PTO  and see if that improves performance. If it does, let me know and we can replace the cable block with something else.

jleonqu, 2023-08-04:
Hello @TiphaineBr 

Can you share more details about your check valve? What are the values that you are using for your valve model? In addition to these values, it would be useful if you can share more details like the maximum flow rate you are expecting, the maximum pressure in each side of the valve, the delta pressure that you are expecting in the valve, etc.

Some general recommendations:
1. Plot the area of your valve as a function of the pressure difference (see figure 5 in this [reference](https://energy.sandia.gov/wp-content/uploads/2014/06/SAND2015-2069C.pdf) ). Change the value of k1 and k2 (equation 4 of the same reference) to adjust the transition time between a fully closed and a fully open valve.
2. I am checking the pressure plot that you have in your first comment, the one with dt = 0.00001, and it looks like the pressure difference across the valve is very small, that's why I would like to see what values you are using for your valve. Try to calculate the valve area that you need to have a pressure drop of 10 psi across the valve. According to your plot, you are just losing ~2 psi, which could be causing the issues, because a small change in the pressure is either closing or opening the valve in your model.

TiphaineBr, 2023-08-08:
Thank you both for your responses!

In order to facilitate your understanding of my model, I made adjustments to the geometry and hydraulic PTO properties to be able to share the model. I used “RM3_cHydraulic_PTO” from WEC-Sim applications as a basis, with : 
- Rescaled RM3 properties and geometry, to align with my own (I regenerated the .h5 file using Nemoh v3)
- My hydraulic PTO model and mechanism, with some confidential values modified
I have added you as contributors, so please confirm if you can access it !

Doing these changes, I also found out why the model resulted in a crashed simulation with variable time step solvers : I was trying to run the simulations with a lower non linear time step (simu.nonlinearDt = 0.01) than the fixed one, and it turns out that it crashed the simulation. 
I was then able to run the model with various solvers : ode 45 ; ode13s ; ode23s ; ode23t ; ode113 ; ode23 ; ode23tb ; daessc… but it didn’t help unfortunately. 

@dforbush2  
I have not yet explored the stand-alone model you proposed, but I now understand what you mean. Concerning the sea bed connection I should have used a cable block indeed, but instead I employed two constraints (rotational) and a transnational PTO (so any changes to be done in this area I suppose). 

@jleonqu 
When selecting the parameters for the check valves, I relied on the head loss / flow curve provided in the datasheet of the utilized check valves' manufacturer. I fine-tuned the parameters of the WEC-Sim check valve block to closely match real behavior. I also attempted to minimize the k2 value to achieve a smoother opening, yet this adjustment did not help.

To see below, the check valves behavior : 
[image]
[image]

I also checked the pressure drop, that seems to be significant enough. But I might misunderstand your meaning on this part… let me know if I’m mistaken reviewing my model. 

Best, 
Tiphaine

jleonqu, 2023-08-14:
Hello @TiphaineBr 

Thanks for providing more information. One more thing, can you upload a 2022b version of your Simulink model?

Regarding the plots that you presented in the comment above:

1. The delta P of your valve between fully closed or fully open is ~2.5e4 Pa (~3.6 psi). I would recommend adjusting the values that you are using to increase that delta P to at least 10 psi or more (depending on your application). This is what is happening: in a given time step your delta P is 3 psi, which is enough to open the valve, but in the next time step, your delta P is -1 psi, this is a change of just 4 psi which is easily achievable for a hydraulic system in a short period of time. Then your valve is showing a high-frequency on-off behavior and that's what you can see in your results. A possible solution for this is to change the valve area to increase the delta P of the valve.
2. What's the maximum flow rate that you are expecting in your valves? You can use that max. flow value to design a valve that has the desired delta P.

TiphaineBr, 2023-08-14:
Hello @jleonqu, 

Thanks for your recommendations, I'll look at it shortly. 

In the meantime, I'd like to make sure that you were able to access the documents I uploaded (I added you as collaborators to my repository, with @dforbush2  ? 
Also, what do you mean by "a 2022b" version of my Simulink model ? I'm currently working with 2023a, and the "meca.slx" file I uploaded has been made with that version. 
Before, I was using Matlab R2020b with WEC-Sim V4.4, is that this first Simulink model you want me to upload ?

jleonqu, 2023-08-29:
Hello @TiphaineBr 
I have access to the documents and files in your repository.
I am using Matlab 2022b. It will save me some time if you upload the 2022b version of your model. In Matlab 2023a you can export your Simulink file to a previous version.

TiphaineBr, 2023-08-30:
Hello @jleonqu 
I exported the Simulink file to the 2022b version. Let me know if it suits you ! 
Many thanks for your help on this subject

jleonqu, 2023-09-01:
Hello @TiphaineBr 
Thanks for uploading the 2022b version of your model. I was playing with some values for the check valve area, and as I mentioned before, by decreasing the valve area I was able to reduce the oscillations. How is your progress with this issue?

jleonqu, 2023-09-06:
Hello @TiphaineBr 
I multiply the valve area by 0.5 and I was able to run your model.

nathanmtom, 2023-10-27:
Closing because of inactivity.

closed 2023-10-27 by nathanmtom