WEC-Sim/WEC-Sim #1086: [BUG] Hardstops ignored by WEC-Sim

antbog, 2023-07-26:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
I'm trying to implement hard stops for my WEC but the parameters I set in the wecSimInputFile.m seem to be ignored while running the simulation. It's almost exactly the same problem as in issue #574 .

**WEC-Sim file(s)**
wecSimInputFile.m and Simulink.

**To Reproduce**
% Translational PTO
pto(1) = ptoClass('PTO1');    % Initialize ptoClass for PTO1
pto(1).stiffness = raideur;   % PTO Stiffness [N/m], optimum found during simulations
pto(1).damping = amortissement ;      % PTO Damping [N/(m/s)], one of 2 optima
pto(1).location = [0 0 0];            % PTO Location [m]

%End-stop PTO
pto(2) = ptoClass('PTO2');    % Initialize ptoClass for PTO2
pto(2).hardStops.upperLimitSpecify = 'on';
pto(2).hardStops.upperLimitBound = 2; %[m] upper limit of the stroke
pto(2).hardStops.upperLimitStiffness = 0; %[N/m]
pto(2).hardStops.upperLimitDamping = 1e9; %[Ns/m]
pto(2).hardStops.upperLimitTransitionRegionWidth = 0.5; % [m]

pto(2).hardStops.lowerLimitSpecify = 'on'; 
pto(2).hardStops.lowerLimitBound = - 2; %[m] lower limit of the stroke
pto(2).hardStops.lowerLimitStiffness = 0; %[N/m]
pto(2).hardStops.lowerLimitDamping = 1e9; %[Ns/m]
pto(2).hardStops.lowerLimitTransitionRegionWidth = 0.5; %[m]
pto(2).location = [0 0 0];  %location at the CoG

The 1st PTO seems to take the values I gave into account, but not the 2nd one.

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
>>>wecSim 

**Expected behavior**
I want the float of the RM3 not to exceed +-2m of stroke.

**Screenshots**
No error message, just ignore my inputs. See the screenshot.

**Workstation Information (this section MUST BE completed):**
OS: Windows 10 Enterprise, Version 22H2
[MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 9.14.0.2254940 R2023a
[WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v5.0.1

**Additional context**
N/A

jtgrasb, 2023-07-26:
Hi @antbog,

Thank you for inquiring about the hard stops. It is hard to be sure of what is going wrong here without knowing what your Simulink model looks like, but there are two potential reasons I suspect:
1. Are your PTOs in series with each other? Although PTO(2) would be constrained, PTO(1) would be free to translate and lead to greater overall motions of the float.
2. The PTO is not constraining motion of the float, but is instead constraining the relative motion between the float and the spar. This is likely not that different of a result because the spar doesn't move much but will lead to at least some difference.

If you are still getting unexpected results after checking these two potential issues, please include the Simulink model in your result so we can better understand your model.

antbog, 2023-07-27:
Hi, 
The second reason was the one! If I look at the relative motion, it's indeed limited as stated in my input file. Sorry for my confusion.

Thank you for the support!