WEC-Sim/WEC-Sim #1088: [BUG]As for the RM3 example, I found that the z coordinates of plate.stl file is not consistant with that in the manual, is that a Bug?

chuzhubei, 2023-07-29:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
As for the RM3 example, I found that the z coordinates of plate.stl file is not consistant with that in the manual, is that a bug? 

**WEC-Sim file(s)**
plate.stl

**To Reproduce**
N/A
**Expected behavior**
N/A
**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
N/A
**Additional context**
N/A

nathanmtom, 2023-07-31:
@chuzhubei Thank you for your question and I'll do my best to respond. 

The .stl files for visualization requires the origin of the .stl file to align with the center of gravity. Then the SimMechanics animation will shift those .stl files based on the defined center of gravity in the .h5 file (from you hydrodynamic data). As highlighted by the spar, a lot more of the geometry is below the water line (z=0 plane) in the SimMechanics Animation which is consistent with the center of gravity being much lower so the .stl is shifted down.

Therefore this is intended and I hope the explanation above explains why. 

Please let us know if you still have questions.

Cheers, 
Nathan

chuzhubei, 2023-08-01:
Thank you for your comprehensive response. I sincerely appreciate it, and I now have a clear understanding.