WEC-Sim/WEC-Sim #1089: [Theory or Implementation]

benhopkinson, 2023-07-30:
Hi guys, 

I am new to WEC-sim and am trying to load the example OSWEC files in matlab, I am coming across these error messages and I am stumpt. apologies for the simplicity but if someone could give me a hand with this it would be great. Cheers!
[image]

nathanmtom, 2023-07-31:
Thank you for your question at @benhopkinson . We have begun requiring users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "**XXXXX**" template and include more details on your setup and error message, I am happy to help. Please maintain the template structure and the issue type in the title. When creating an issue, please use the GitHub templates:
[image]
[image]
[image]

This information is important for us to accurately assess what an issue is, how it occurs, and most quickly create a solution for you. It also helps us categorize our issues to determine areas of future improvement. Please maintain the template structure (bold headings). ​

nathanmtom, 2023-08-07:
@benhopkinson We are still waiting back for you about this issue. Please provide an update by this Friday, August 11th or else we will close this issue due to inactivity.

closed 2023-08-11 by nathanmtom

benhopkinson, 2023-08-13:
Apologies @nathanmtom
I will make sure to follow this format from now on. FYI My error occured due to matlab simulink not being installed properly. 
Cheers