WEC-Sim/WEC-Sim #1090: [BUG]

DeminLiLi, 2023-07-31:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**
The  phase of wave excitation force in two-body system in "readAQWA.m"
**Describe the bug**
_A clear and concise description of what the perceived bug is._
I am currently using version 5.0.1 of wecsim. When calculating the double body, I found that the heave relative motion of the double body is different from the test results. I used version 4.3 as a verification and found that the calculation results of version 4.3 are consistent with the test values. After It was identified that there was a problem with the phase of the wave excitation force in version 5.0.1.
When using wecsim version 5.0.1 for two-body calculation, the phase difference in the heave direction between the inner float and the wave increases by 180°, resulting in a 180° phase difference in the relative motion of the heave between the two floats. Directional motion becomes counter-directional motion, and eventually the relative motion amplitude of the heave becomes the superposition of the amplitudes of the two floats. This is obviously wrong. After inspection, it was found that the inner float in the heave motion did not call hydro(F).ex_ph(3,ind(2),ind(3)) but called  hydro(F).ex_ph(((ind(1)-1)*6+1):(ind(1)*6),ind(2),ind(3))  in "readAQWA.m", but the outer float was called by the correct wave excitation force.
**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._
The line 168 and 169
[image]
I compared the experimental results with the calculation results of different versions. The blue line, orange line and gray line are the phase difference of outer float and wave, the inner float and wave, and the phase difference between the two floats calculated with the 5.0.1 version of readAQWA.m. The phase difference results of the yellow line, the cyan line and the green line are calculated with the 4.3 version of readAQWA.m, and the black line is the phase difference of the two floats in the test. It can be seen from the figure that the phase difference between the inner float and the relative motion calculated by the wave excitation force phase of version 4.3 is correct and consistent with the test results. In the modified readAQWA.m, the phase  of the original wave excitation force is updated to the phase of line 168 which is same with the version 4.3.
[image]
[image]

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to '...'_
2. _Click on '....'_
3. _Scroll down to '....'_
4. _See error_

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]
Windows 11  MATLAB R2022b  WEC-Sim v5.0.1
**Additional context**
_Add any other context about the problem here._

salhus, 2023-08-02:
Hi @DeminLiLi,

Thanks for your observations. Would you be able to share some information on the BEM setup? e.g., the output files from AQWA that BEMIO is reading. If possible, could you share the mesh files used as input? I would like to compare the AQWA results with other BEM software.
The direction of the incoming wave has a different convention in AQWA when compared to that used in other BEM software and in WEC-Sim (wave direction is positive Left to Right). The difference of conventions has been discussed in #855 #865 
A quick check would be to set, 
waves.direction = 180

Let me know, how the results look when you set waves.direction = 180 (for better or for worse)

Cheers,
Sal

DeminLiLi, 2023-08-03:
Hi Sal,
Many thanks for your reply, I am very happy to share my aqwa output file. I am very sorry that I did not notice the previous discussion about the direction of the incident wave. I have not redefined the direction of the wave in WEC-Sim. According to the default value, it should be the 0 direction of the input. In the attachment it is the result of aqwa, and I will further verify the result of the 180° moment.
[aqwa results.zip](https://github.com/WEC-Sim/WEC-Sim/files/12250025/aqwa.results.zip)

salhus, 2023-08-03:
Hi Demin @DeminLiLi,

Could you also share your WEC-Sim model (the folder with WEC-Sim files).
I am not sure what you mean by the changes you made. Did you change the direction in the 'WEC-Sim' input file (used to run the WEC-Sim model) or the AQWA file?
The WEC-Sim suite of code can be seen as two-staged; 1. Post-processing the BEM results, and 2. WEC-Sim simulation itself. My suggestion was to change the waves.direction in the second phase, i.e., in the wecsim input file.
Changing the waves.direction = 180 was a recommendation for accounting for the differences in conventions between WEC-Sim and AQWA. The BEMIO run is merely reading the AQWA files, so as to account for the different convention AQWA used. The convention AQWA uses may not be the same as the one in the reference data/ exp. data.

I don't mind running the same system in other BEM software to compare the AQWA outputs. However, it is difficult to understand without having the geometric mesh files and the WEC-Sim model.

DeminLiLi, 2023-08-03:
Hi sal，
Many thanks for your reply, and I will share the WEC-Sim model with you.  In the above reply, I changed the direction in  'WEC-Sim' input file. Because the file is so large, so I send email to you. Please check it.

Best wishes,
Demin

salhus, 2023-08-04:
Given that different post-processing tools are interpreting phase the same way. The readAQWA.m seems to work correctly and no changes are necessary. The user was having issues in how they defined the PTO between the two bodies they were modeling. Discussion on differences in sign conventions can be found at #846 which was resolved in the PR https://github.com/WEC-Sim/WEC-Sim/pull/855

hachikoi1, 2023-08-05:
> Hi Demin @DeminLiLi,
> 
> Could you also share your WEC-Sim model (the folder with WEC-Sim files). I am not sure what you mean by the changes you made. Did you change the direction in the 'WEC-Sim' input file (used to run the WEC-Sim model) or the AQWA file? The WEC-Sim suite of code can be seen as two-staged; 1. Post-processing the BEM results, and 2. WEC-Sim simulation itself. My suggestion was to change the waves.direction in the second phase, i.e., in the wecsim input file. Changing the waves.direction = 180 was a recommendation for accounting for the differences in conventions between WEC-Sim and AQWA. The BEMIO run is merely reading the AQWA files, so as to account for the different convention AQWA used. The convention AQWA uses may not be the same as the one in the reference data/ exp. data.
> 
> I don't mind running the same system in other BEM software to compare the AQWA outputs. However, it is difficult to understand without having the geometric mesh files and the WEC-Sim model.
> 
> The waveclass.m does not require any changes. A PR has already addressed #846
> 
> I post-processed the AQWA files you shared in BEMIO and in BEM Rosetta (a completely different post-processing software), and seem to get the same results. So seems like both BEMIO and BEM Rosetta are interpreting the phase information in the same way - verifying the implementation in WEC-Sim v5.0.1 [image]
> 
> [image] [image]
> 
> Cheers, Sal

Dear [salhus](https://github.com/salhus)
Nice to meet you again. I'd like to point out that, BEM Rosetta seems to convert the aqwa phase using -ph*pi/180, as documented in Line 197. 
Best regards,
Tianyuan
[image]

DeminLiLi, 2023-08-05:
Dear Sal,

Sorry, I don't agree with you that my numerical model is about the pto setting, so I think it is necessary to reopen this thread for a full discussion.  The first thing I put forward is that the definitions of the phase angle of the wave excitation force by the two softwares of BEMIO and bemrosetta are completely different. Bemrosetta does not perform two coordinate transformations on the phase angle, but only performs negative value processing on the phase.  Modifications made in BEMIO.  In the screenshot you gave, BEMIO uses -180 degrees, and bemrosetta uses 0 degrees. It is a comparison of two directions. The same result only shows that BEMIO does not need to perform phase angle coordinate conversion.  This is exactly what I explained to you in the email. I think you can think about my explanation. In addition, I will further verify whether the waves of wamit and wecsim you mentioned are from left to right.
KG
Demin

salhus, 2023-08-05:
Hi @DeminLiLi @hachikoi1,

Nice to see you back too @hachikoi1.

@DeminLiLi In the screenshot you showed when the issue was opened, your correction of commenting out line 169 is identical to what line 169 is doing. The only difference is that the angle is being wrapped to avoid values outside $[-\pi,\pi]$ range.
[image]

The so-called new implmentation is treating all modes differently to the heave mode, notice line 168 does a flipping operation (explained below) while line 169 over-writes 168 for the heave mode with just a negative sign and no additional $\pi$. 
By using that negative sign and finding it to be the correct approach you verify that the implementation is correct for the heave mode, which is same as BEM Roetta as shown by @hachikoi1. 

The $-\pi$ phase difference is happening for modes other than heave. 

Now for other modes, the issue is twofold,
1. Whether the phase from AQWA needs correction?
2. How the correction must be done?
---------
1. Whether the phase from AQWA needs correction?

Here is the WEC-Sim convention,

[image]

Here is the AQWA convention,

[image]

The current implementation is as follows,

[image]

As it would become clear from the conventions shown above, oscillations in AQWA will have to be rotated about the vertical $z-axis$. Which implies, the vertical oscillations in $z-axis$ will change their sign, but oscillations in $x-axis$, and rotations about the $z-axis$ and $x-axis$ will have to be flipped -- i.e., a change in sign and additional phase of $-\pi$. 

As the screenshot shows in the third line, the heave phase is only being changed by a negative sign which is same as BEM Roetta as shown by @hachikoi1.
Another way of thinking about this transformation is that the AQWA results are being rotated about the vertical. 
Implying, the displacement along $z-axis$ remains the same, while the displacement in $x-axis$ needs two transformations:
 1. Change in the coordinate position,
 2. Change in the sense of motion. 
 ----------
 2. How the correction must be done?

The change in coordinate position is accomplished by the rotation about the $z-axis$ -- the addition of phase $-pi$, while change in direction is accomplished by the negative sign. This means the phase needs to be flipped (negative sign and additional phase of $-\pi)$ for oscillations along $x-axis$, $y-axis$, and rotations about $x-axis$, $y-axis$, $z-axis$. Oscillations _along_ $z-axis$ however, doesn't change its coordinate, only its direction, hence only a negative sign.

Put simply, the need for additional $-\pi$ phase difference for NON-Heave modes comes from the fact that the coordinate system has to be rotated in order to convert what AQWA treats as positive to what WEC-Sim treats as positive.  

Now to verify this, I ran bemio on the cylinder example for wamit, and aqwa. I compared them using the compare feature in BEMIO. If we compare the surge modes, the phases in aqwa have the opposite sign -- If onlynegative sign was used to correct all modes.
Here is a comparison of excitation force phases from wamit and aqwa in surge mode (oscillations about $x-axis$)

[image]

and here is the same comparison between the excitation phases of wamit and aqwa with the current implementation,

[image]

This shows why NON-Heave modes require a negative sign and a phase difference of $-\pi$ to be converted to WEC-Sim convention. While established above, the Heave mode just needs a negative sign.

Cheers,
Sal

DeminLiLi, 2023-08-07:
Dear Sal,
I have a question about the coordinate agreement between aqwa and wecsim. In my opinion, there is a negative sign difference between the time item and the space item of the cos function in the two softwares, but the phase angle is the same. After extracting the minus sign, cos(-z)=cos(z), only the minus sign difference about the phase angle, regardless of z-axis flip. I don't know if my understanding is correct, please explain it for me.

KR,
Demin

salhus, 2023-08-07:
Hi Demin,

Refer Euler's identity to see that how equation you show becomes the screenshot shared which has the real part of what you shared, please refer,
https://en.wikipedia.org/wiki/Euler%27s_identity

The X and Y take a negative sign because the coordinate frame is flpped.

salhus, 2023-08-07:
@DeminLiLi 

I would reiterate please refer https://github.com/WEC-Sim/WEC-Sim/issues/865
https://github.com/WEC-Sim/WEC-Sim/issues/865#issuecomment-1578912003
https://github.com/WEC-Sim/WEC-Sim/issues/865#issuecomment-1579308388
https://github.com/WEC-Sim/WEC-Sim/issues/865#issuecomment-1579809942
https://github.com/WEC-Sim/WEC-Sim/issues/865#issuecomment-1579949133
https://github.com/WEC-Sim/WEC-Sim/issues/865#issuecomment-1579960696

DeminLiLi, 2023-08-07:
Dear Sal,
Thank you for your reply, I understand the Euler transformation,I am sorry that I'm sorry I misread a symbol. I am aware of the modifications you made in 5.0.1. I personally think that all degrees of freedom do not need to add Π, in other words, no coordinate transformation is required. For my numerical model, my method can match the experimental data very well, but the numerical model established by your method is wrong. Modifications in the non-heave direction cause my model calculation errors in the heave direction.  The reference BEMrosetta also does not carry out the coordinate transformation of the phase angle. Thank you for your positive communication with me, and I am very sorry to bother you for so long.

KR,
Demin

salhus, 2023-08-07:
@DeminLiLi @hachikoi1  
I think you are right. The examples I was using to compare had 3 wave headings with the set [-180, 0, 180]. When I compared these examples to the corresponding WAMIT example, I was making a mistake. I am making a PR that removes the additional pi phase. Thanks for bringing it to my attention and for the careful discussion.

DeminLiLi, 2023-08-08:
Dear Sal,
Thank you very much for your reply, I think our discussion was positive and useful. Thank you for your revisions and your efforts for all researchers who use WEC-Sim. I hope we can speak freely when we encounter problems in the future. thanks again.
KR,
Demin

salhus, 2023-08-08:
Hi @DeminLiLi,
I appreciate your patience with me. I am glad you found the discussion helpful, and I would be happy to talk again in the future.
I have made a PR #1096 feel free to review :)

Cheers,
Sal

kmruehl, 2023-08-10:
resolved by [1096](https://github.com/WEC-Sim/WEC-Sim/pull/1096[)](https://github.com/WEC-Sim/WEC-Sim/commit/8db4dccccfc9c0841d8608bb9510ecb2bb198b48)

closed 2023-08-10 by kmruehl