WEC-Sim/WEC-Sim #1094: [Feature request] BEMIO plot RAO function

ckberinger, 2023-08-04:
**Is your feature request related to a problem? Please describe.**
I would like to compare the body response amplitude operator values from WAMIT to that found from other methods like experiments and tuned WEC-Sim models. I have turned on the RAO option in the WAMIT frc file and I can see the RAO values in the .out file. But it seems like the h5 file from BEMIO does not include this data and it cannot be visualized in the HDFviewer either. I do not see a function to plot the RAOs in BEMIO.

**Describe the solution you'd like**
I would like a plotRAO function that plots the RAO values in the desired degrees of freedom. And a matrix that lists the RAO values for each frequency similar to how the other coefficients are saved. 

**Describe alternatives you've considered**
I have considered manually extracting the RAOs for each frequency from the .out file. Or coding something similar to the readWAMIT function. 

**Personal project context**
I would like to have plots comparing the WAMIT RAOs of the float and spar of the LUPA WEC with those found in experiments and a tuned WEC-Sim model.

**Additional context**
NA

salhus, 2023-08-04:
Hi @ckberinger,

You can look at the RAOs by enabling the 'rao' flag when running BEMIO. 
[image]
The description of readWAMIT also shows this, 
[image]
Check out that functionality but if there is something missing I can work on this functionality. Also, if you like, it would be great to have more user participation and I would highly encourage you developing that.  

Additionally, a much easier approach would be to use BEM Rosetta.

ckberinger, 2023-08-07:
Hi Sal,

If I am understanding correctly, using the approach you outlined would produce just the _excitation force coefficient_ as an RAO. Is it possible to represent the _body motion_ RAO from BEM outputs in the frequency domain?
Such as with an equation like this: 
[image]
taken from here: https://en.wikipedia.org/wiki/Response_amplitude_operator

I do see now that the Response Amplitude Operators values given from WAMIT are connected to the excitation force after adding the 'rao' flag to readWAMIT. This is not that clear in the .out file or the manual. Please clarify if I am not understanding this correctly. 
[image]

salhus, 2023-08-07:
Hi @ckberinger,

The readWAMIT script, looks for the RAO (be checking for RESPONSE AMPLITUDE OPERATOR string) in the output file. 
Additionally, here is some simple code I wrote that can read the RAO from the #.4 files.

`clc;clear;close all

mode = 3;

fname = 'FILENAME';

%%

n_angle = 1;
n_body = 1;    %number of bodies
modes_tmp = [1 2 3 4 5 6];
modes = [];
for i = 1:n_body
    modes = [modes modes_tmp];
end

fileID = fopen([fname '.4']);
rao_out = textscan(fileID,'%f %f %f %f %f %f %f %f','HeaderLines',1);
fclose(fileID);
rao.data = cell2mat(rao_out);
n_modes = find(and(rao.data(2:end,3)==rao.data(1,3), ...
                   rao.data(2:end,4)==rao.data(1,4)),1);

%% Package it all up

rao.w       = 2*pi./rao.data(mode:length(modes):end,1);
rao.mod      = rao.data(mode:length(modes):end,4);
rao.phase    = rao.data(mode:length(modes):end,5);

close all
clearvars  -except rao

%%
figure()
plot(rao.w,rao.mod,'-r','LineWidth',2)
grid on
legend('RAO Magnitude','Location','best')
figure()
plot(rao.w,rao.phase,'-b','LineWidth',2)
grid on
hold on
legend('RAO Phase','Location','best')`

ckberinger, 2023-08-07:
Thank you for sending that, it worked well. Is this RAO the 'amplitude of body motion/amplitude of the wave'?

salhus, 2023-08-07:
Yes, it is. 
WAMIT generates output files with the extensions corresponding to the IOPTN list you posted.

ckberinger, 2023-08-07:
I meant to say "amplitude of body divided by the amplitude of the wave" but WAMIT uses a unit wave height so it is the same as saying the "motion of the body". I think I understand now, thank you for all your help!

salhus, 2023-08-07:
Exactly.
So we can close the issue?

ckberinger, 2023-08-07:
Yes, thank you!

closed 2023-08-07 by salhus