WEC-Sim/WEC-Sim #1103: [WEC-Sim Applications]

benhopkinson, 2023-08-13:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
writeBEMIOH5()

**Describe the bug or issue**
I am attempting to run bemio.m for the Nemoh files that I have attached in the sphere.zip file below. Everything runs when I update readNEMOH.m with help from issue [#1082], but when the 'writeBEMIOH5()' function tries to run, the error message 'Unrecognized field name "body"' occurs. Screenshots are attached.

**To Reproduce**
I believe this file contains all the required Nemoh files to run bemio as per wec-sim tutorials.
[Sphere.zip](https://github.com/WEC-Sim/WEC-Sim/files/12329546/Sphere.zip)

**Expected behavior**
No Error and the .h5 file be created

**Screenshots**
[image]

[image]

**Workstation Information (this section MUST BE completed):**
 - OS: macOS Monterey
 - MATLAB/Simulink 2022a
 - WEC-Sim Release v5.0.1

**Additional context**
N/A

nathanmtom, 2023-08-16:
@benhopkinson Thank you for rising this issue and will do my best to answer. 

After downloading your files, the issue has been identified that you deleted the '! Name of mesh file` from the Nemoh.cal on the line where the axisym.dat was defined (see image below where I replaced the comment. 

[image]

You can see from the readNEMOH.m file on line 87, see image below, that our parser is looking for the exact wording 'Name of mesh file' or else it will not assign a name to hydro.body.

[image]

After replacing '! Name of mesh file` in your Nemoh.cal, both writeBEMIOH5 and plotBEMIO functions ran without issue. 

Please confirm this fix addresses works for your example so we can close this issue.

- Nathan
National Renewable Energy Laboratory

benhopkinson, 2023-08-19:
Thank you so much @nathanmtom, everything works perfect now. Just so you are aware the '! Name of mesh file` comment from the Nemoh.cal file does not appear when I generate a new mesh in Nemoh. I have tried to generate multiple files and each time the comment is missing. I am not sure if this is something you guys may want to look into for readNEMOH.m or if it is just me, thank you regardless I am all up and working now

nathanmtom, 2023-08-21:
@benhopkinson Thank you for confirming that by adding the comment line `! Name of mesh file` that readNEMOH.m is working for you. 

To clarify you second statement, `when I generate a new mesh in Nemoh` there is a function being called in a NEMOH script of function that autogenerates the Nemoh.cal file? If this is the case, my assumption is that the file could be amended if the other comments in the file are autogenerated. If you can confirm this is how you were generating Nemoh.cal files then I'll add a project board card to explore this further or determine if a modification to readNEMOH.m is a better option.

mprevisic1, 2023-08-22:
I encountered the same issue. Nemoh 3 produces a different output format. I solved the issue by adding the "! Name of Mesh file" comment back inthe the Nemoh.cal file and it runs.

nathanmtom, 2023-08-28:
Based on the additional feedback and recent PRs #1087 #1105, I am going to close this issue. We have identified that the main issue on this thread was the missing `! Name of mesh file` in the `NEMOH.cal` input which is not included when using the NEMOH v3.0 autogeneration feature for making 'NEMOH.cal'.

closed 2023-08-28 by nathanmtom