WEC-Sim/WEC-Sim #1104: [Theory or Implementation]

brittlyd, 2023-08-16:
**Is your question request related to a problem? Please describe.**
My question is related to the NMPC example on the WECCCOMP device. Specifically, in the Simulink model of that example.

**Describe the theory or implementation approach that you have a question**
In the Simulink model for the NMPC example on the WECCCOMP device, there are certain Goto blocks that are highlighted in green and represent values that are not directly included in the Simulink model (i.e. excForce, shown in the first screenshot) and they have no Goto source (shown in the second screenshot). How are these variables accessed, and can I do this for other variables I want to use in the control scheme (total force/torque, wave field measurements, etc.). Also, is there a direct way to access past measurements of these variables, or is it best to just concatenate the most recent value to a matrix of past values?

**Personal project context**
I am trying to employ a MPC scheme that needs other values that WEC-Sim calculates for its states and am looking for how to access them in Simulink.

**Describe the type of conclusion or resolution you are looking for**
I am looking to be able to access other variables in Simulink that aren't the standard output from the PTO. 

**Additional context**
[image]
[image]

brittlyd, 2023-08-17:
Hi Nathan,

Thanks for your quick response! So if I wanted to do a similar thing to the OSWEC example (utilize other values like excitation torque, etc), I wouldn't be able to because I can't access them using a standard WEC-Sim Body block, is that correct? Is there a way to apply this to the OSWEC example?

Thanks!
Brittany

nathanmtom, 2023-08-17:
@brittlyd Thanks for the follow up and apologies for not being clearer in my initial response. 

You can complete this for your model by following the same steps of [breaking the library link](https://www.mathworks.com/help/simulink/ug/disable-links-to-library-blocks.html) and inserting the "GoTo" blocks to make the wave excitation vector available for your control algorithm. The Simulink blocks and Matlab scripts are customizable and this is an example of a feature that most users do not require, but can be implemented fairly quickly. 

I would just caution that the forces and moments you will have access to will be about the center of gravity of the OSWEC, while I assume you will want the forces about the OSWEC rotation point. If this is the case, you will need to complete a coordinate system transform to get your desired values. 

- Nathan

nathanmtom, 2023-08-28:
@brittlyd I wanted to check to see if my last response provided sufficient explanation to implement your own version of the WECCCOMP NMPC? If not please let me know what additional help you might need. If we do not hear from you by the end of this week we will plan to close the issue and can reopen if updates become available.

Cheers, 
Nathan

closed 2023-09-01 by nathanmtom