WEC-Sim/WEC-Sim #1109: [Theory or Implementation] MoorDyn Preprocessing

yman95, 2023-08-23:
**Is your question request related to a problem? Please describe.**
I'm testing out a hexagon geometry under regular wave condition and it has been functioning properly, i.e. without MoorDyn, the project works fine. The next step for me is trying to implement MoorDyn into the project. The mooring line I'm trying to test is a simple straight line with 11 nodes with the bottom fixed at the seabed and the top attached to the geometry. Currently, I'm having problems getting MATLAB to preprocess MoorDyn. 

The error message is as follows,
[image]

With incomplete error message, it is difficult to identify the cause. It seems like MATLAB or MoorDyn has trouble reading the input file, rather than the coding or values of the actual contents.

- MoorDyn was downloaded and extracted to `WEC-Sim\source\functions\moorDyn`, and MinGW-w64 was installed and added to the environment PATH. 
- MoorDyn input file lines.txt was put into a mooring folder under the project folder
- MoorDyn .slx block was added 
- Line types and properties from the input file were followed from the RM3 application

**Describe the type of conclusion or resolution you are looking for**
MoorDyn functions normally

**Additional context**
PS: the userDefinedFunctions.m prints 7 graphs, so you may want to delete those codes for testing.
Project folder:
[hexagon_10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/12417769/hexagon_10m.zip)

yman95, 2023-08-26:
Hi @dav-og

I can't actually seem to solve this problem about rotating/flipping.

I tried the following:

- Change the constraint(1) in wecSimInputFile location to (0,0,-1.875)
- Change the CG to (0,0,-1.875) and rerun NEMOH for the h5 file
- Change the center of motion in BEMRosetta to (0,0,-1.875) ((Though I'm not sure if the .dat file exported from BEMRosetta remebers the change in the center of motion))
- Change Nemoh.cal file rotating point to (0,0,-1.875)
- Change Nemoh.cal file rotating point and CG to (0,0,-1.875)
- Change Nemoh.cal file rotating point, CG, and constraint(1) to (0,0,-1.875)

None of them seem to stop the geometry from rotating on its own. Any idea how to solve this?

One thing I've noticed is that running Nemoh's hydroscal.exe will always set CG back to (0,0,0) because neither the .dat and .stl file can remember the location of the CG. I will have to manually edit the CG in Hydrostatics.dat file. Maybe the inertias were calculated based on CG at (0,0,0) despite the change of it afterwards. I'm not entirely sure. I've attached a zip file of the last case (Nemoh.cal file rotating point, CG in \mesh\hydrostatics.dat, and constraint(1) in wecSimInputFile to (0,0,-1.875)) of Nemoh side of the project here. Feel free to let me know if there are additional files/info you need.
[hexagon_10m_v2.zip](https://github.com/WEC-Sim/WEC-Sim/files/12444746/hexagon_10m_v2.zip)

Appreciate the help,
Yi

yman95, 2023-09-05:
Hello @dav-og,

I've reconstructed my model and rerun all the hydrodynamics. Now the model is stable under noWaveCIC condition. However, the model will still pitch over 360 degrees (rotate) under regularWaveCIC. I assume this might be reasonable for the geometry but still not desirable for the model. Since the geometry is floating on the surface with its CG almost at the surface, not moored, it is prone to rotation. Is this behavior reasonable? 

The updated file is attached. In this version, the corrected CB is at (0, 0, -1.625). In the hydroData folder, the h5 files for different different fixated point are included. cg is for center of gravity, cm refers to the fixed point in NEMOH's nemoh.cal file. '_low' means they are set to (0, 0, -1,625), and '_0' means they are set to (0, 0, 0)
[image]

I ran the moorDyn plugin, and it gives the same error message. Please let me know where you think the problem is. Thanks!
[hexagon_10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/12521431/hexagon_10m.zip)

Best,
Yi

yman95, 2023-09-11:
I tried the MoorDyn example from WEC-Sim/Application/Mooring, and I got the same error message, so I believe there must be something wrong with how MoorDyn functions or installed on my end. 
[image]

Here's how I installed MoorDyn. I double checked but can't seem to see if I have installed it wrong.
MoorDyn was downloaded from its repository:
[image]

MinGW-w64 was installed and its path were added to the system environment
[image]

lines.txt was put into 'Mooring' folder. 

Let me know what the problem could be. Thanks.

Yi

yman95, 2023-09-12:
Small update @dav-og :
Reinstalled MinGW-w64 to be version 6.3.0 which is the version compatible to MATLAB R2022b.
Set mex to C++ compiler 
[image]

WEC-Sim still has problem simulating but it's no longer 'Error in preprocessing,' instead,
[image]

Please advise,
Yi

mattEhall, 2023-09-17:
Hi Yi -- @RyanDavies19 and @jtgrasb may have some advice on this!

RyanDavies19, 2023-09-18:
@yman95 we are in the process of putting together an update that makes this coupling much easier in the future. It should be live on the dev branch by the end of the month. 

Could you share your lines.txt input file for MoorDyn? MoorDyn version 1 (what is currently implemented in WECSim) does not have the best error handling. I'm curious if there is a formatting issue there.

yman95, 2023-09-20:
Thanks for the responses @mattEhall @RyanDavies19 @jtgrasb!

The news is exciting! I'll be thrilled to try!

Here is the input file for MoorDyn. I pretty much copied the properties from the RM3 example. If you need other data for WEC-Sim, the entire updated folder was uploaded in the reply two weeks ago.
[lines.txt](https://github.com/WEC-Sim/WEC-Sim/files/12667726/lines.txt)

dav-og, 2023-09-20:
Hi @yman95, apologies for missing your responses,

> However, the model will still pitch over 360 degrees (rotate) under regularWaveCIC. I assume this might be reasonable for the geometry but still not desirable for the model.

As I mentioned, we can only simulate small amplitude motion in WEC-Sim - so even under wave forcing, the body should only really pitch up to 20 degrees or so. If its rotating 360 degrees, this could cause issues with your MoorDyn coupling.

I've noticed in your latest file that the CG and CB values for the floating body are co-located - this could be the reason why its flipping over - you may want to try putting the CG below the CB, to help with stability.

yman95, 2023-09-21:
Hi @dav-og @RyanDavies19 

Understood. After trying to set the CG at z= -3, -4, and -10 (geometry has z=-3.25 bottom), the geometry still rotates 360 degrees, and it confuses me. I don't think geometry is supposed to rotate with CG at -10.
[image]

Maybe it has something to do with the weight of my model and its MOI? Or maybe I messed up with NEMOH's calculations themselves? Here are my NEMOH and WEC-Sim folder:
[NEMOH_hexagon_10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/12685483/NEMOH_hexagon_10m.zip)
[WEC-Sim_hexagon_10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/12685509/WEC-Sim_hexagon_10m.zip)

dav-og, 2023-09-21:
Yes, it shouldn't be rotating 360 degs...

I thought your suggestion of MOI was good, but I checked the values in your input files and they seem quite reasonable to me...and the RIRFs look reasonable.

I think it could be the hydrostatic stiffness matrix:

[image]

These values seems a bit small to me - they might need scaling or something. I'm not sure how you computed these values, but I would suggest looking into that more to make sure you have the correct units etc.

If they are too small they won't provide a sufficient restoring moment for your floating body, which could explain why it keeps spinning.

yman95, 2023-09-22:
@dav-og You are right. After a quick investigation, it seems like the stiffness is not read properly. Here is the KH file in NEMOH.
[image]

It seems like the data were mistaken by a factor of E+04, e.g., the E+06 number at [2,2] were read as 64.951 instead of 64951.9 (small discrepancies between the h5 data and the KH.dat data from NEMOH). I went to check readNEMOH.m, and I can't find a reason for such behavior; the code didn't have any data manipulation. Later on in the code, the KH values were used to calculate RAO, but the values weren't changed. I'm confused on this part.

However, when I manually multiplied every value by E+04 in KH.dat, the simulation will crash as the geometry 'flies' vertically out of the screen after just 1.95 seconds.
[image]

Any thoughts on this?

mattEhall, 2023-09-22:
Those pitch and roll hydrostatic stiffness values look negative! (Assuming
I'm interpreting this correctly, as a normal hydrostatic stiffness matrix.)
I'm not sure if this matrix includes the effect of CG or just hydrostatic
pressure. I'm also not sure if it's relative to the waterline or the cg.
But either way it hints that the metacenter is below the reference point,
and potentially below the I'm too.

You mentioned CG location. Have you also checked the CB or metacenter
location and ensured it is above the CG?

yman95, 2023-09-22:
Hi @mattEhall 

Thanks for replying. Yes, the CB is above CG's location. As NEMOH was taking care of the calculation, I only set the CG at z=-10 then proceeded to run the solver of NEMOH. CB is supposed to be at z=-1.625 with the geometry bottom at z=-3.25. These values refer to positions below the waterline. 

There's one uncertainty with NEMOH. I'm using the executables from them, and it goes like preproc.exe -> hydroscal.exe -> solver.exe -> postproc.exe. I normally set the CG location before running the executables, but after hydroscal.exe, the CG will be reset to (0,0,0) and I have to manually set the CG location in .../mesh/Hydrostatics.dat to the desired location (in this case, z=-10). I'm not sure if the reset of CG from hydroscal.exe affects any other coefficients, i.e., if other coefficients were calculated based on CG at z=0. 

In their manual, the hydroscal.exe is responsible for the .dat and .tec files in .../mesh folder, and the files in .../mechanics.
[image]

mattEhall, 2023-09-22:
Okay that's good that the CB is above the CG :)
I'd suggest confirming the meaning of this stiffness matrix (is weight
included, is it about CG or waterline?) and then doing some hand
calculations to confirm that it's got the correct values.

yman95, 2023-09-26:
Hi @RyanDavies19 @jtgrasb @dav-og,

I think the values from KH.dat were divided/normalized by the (rho*g). 653107.4/(1025*9.81)=64.952, which is the value in the h5 file. Is it normal for the KH.dat value for a 10m circumdiameter, 7.5m tall, hexagon to be around 6.5*10^5?

Also as @mattEhall mentioned, the calculations could be based on the water level. After some investigation, I think that might be the case. After hydroscal.exe, NEMOH calculates the coefficients, including KH.dat. The CG will always be reset to (0,0,0), and at the same time, KH.dat values are calculated, which makes believe that all the coefficients are calculated based on CG at (0,0,0). Since both the .dat and .stl geometry don't remember the CG location when saved, I currently don't have a solution to this problem, other than maybe recompile NEMOH and modify the source code instead of using the .exe for calculations. 

Any suggestions?
Yi

yman95, 2023-10-10:
Update on the coefficients from NEMOH.

I contacted a developer from Nemoh and he suggested to change CG in mesh.cal since hydroscal.exe only takes mesh.cal as an input. Once I edited mesh.cal, the geometry becomes stable from WEC-Sim's simulation with CG at (0,0,-3) and (0,0,-4). 

[hexagon_10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/12857750/hexagon_10m.zip)

closed 2023-10-17 by dav-og