WEC-Sim/WEC-Sim #1110: [BUG] RM3 WAMIT example: strange peaks

antbog, 2023-08-23:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
I launched a WAMIT run for the RM3 WEC provided in the tutorials. I used the input files given in this GitHub repository (examples > BEMIO > WAMIT > RM3) and changed only the IOPTN(4) from 0 to 2 in the .frc file.

When I display the frequency-dependent quantities (using bemio.m or my own routines), it appears that strange peaks that weren't in the tutorial's .out appeared. They could be due to irregular frequencies (but the option IRR=1 still stands in .cfg) or resonances. Anyway, I don't understand the emergence of these peaks and especially why you don't have it in the tutorial hydroData figures.

Moreover, even when I don't change the IOPTN(4) (ie I keep all the exact settings provided in the examples > BEMIO > WAMIT > RM3 files), the peaks still appear.

**WEC-Sim file(s)**
examples > BEMIO > WAMIT > RM3 all input files (*.wam, *.gdf, *.dat, , *.cfg, *.frc, *.pot)

**To Reproduce**
1. Launch WAMIT with these files as input
2. Use the .out file as input for bemio.m or for any routine allowing to visualize the frequency-dependent quantities, or even the response functions (it this case, we observe beat instead of peaks).

**Expected behavior**
The peaks shouldn't be there.

**Screenshots**
Plots from the RM3 tutorial:
[image]

Plot that I obtain after WAMIT calculations with the input files provided in examples > BEMIO > WAMIT > RM3
[image]

**Workstation Information (this section MUST BE completed):**
OS: Windows 10 Enterprise, Version 22H2
[MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 9.14.0.2254940 R2023a
[WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v5.0.1
WAMIT Version 7.4(x64)

**Additional context**
N/A

dforbush2, 2023-08-23:
Hi @salhus and @antbog 

It looks like @antbog still does have IRR=1 in his input file since he is using our input files in examples/BEMIO/WAMIT/RM3. The question is why do these peaks emerge here, but are NOT present in the tutorial BEM result which allegedly come from the same results. Changing IOPTN(4) should not affect any thing here, its just toggling an RAO output on/off. 

In this directory IRR =1, so I would expect @antbog to be getting the same results as in the tutorial, especially as he is using the same WAMIT version we are.

antbog, 2023-08-24:
Hi @salhus and @dforbush2 

Yes, I'm aware of the existence of this option removing the effects of irregular frequencies and as @dforbush2 said, I'm using your input files so in particular, IRR=1 is in the .cfg and the .gdf contains correctly the interior free surface panels. 

Here is the .out I got if you want to check. 
[rm3out.txt](https://github.com/WEC-Sim/WEC-Sim/files/12426107/rm3out.txt)

I superimposed my curves with the ones from the tutorial .out and it shows that apart from the peaks, they are similar.

I wonder if it has something to do with the contact between the float hole panels surrounding some of the spar panels:
[image]

But in any case I don't understand why and don't get the same results as in the tutorial.

Cheers,
Antoine Bogenschutz

dav-og, 2023-08-24:
Yes, the agreement does look good apart from the irregular frequencies:

[image]

For some reason the irregular frequency removal option is not working properly in your case. However, your .out file shows the option is on. Hence, it could be something related to the meshes (e.g. maybe the mesh of the interior free surface is not being created properly).

In any case, there must be a discrepancy somewhere in your input files (and/or meshes) vs the input files in `examples/BEMIO/WAMIT/RM3`. I've tried to compare the two in BEMRosetta, but nothing jumps out:

[image]

(same CB, CG, Vsub, etc)

Without access to the rest of your input files and meshes its impossible to say.

Cheers,
Dave

antbog, 2023-08-24:
Hi, 

Here are all the input files and meshes. 

[0823_inutFiles.zip](https://github.com/WEC-Sim/WEC-Sim/files/12428364/0823_inutFiles.zip)

They are supposed to be the same than those on the Git repository as I copy-pasted them from there.

Cheers,
Antoine

dav-og, 2023-08-24:
Its a strange one - nothing jumps out to me...

However, I can see that the files you shared are not _exactly_ the same as in the examples folder. I've noticed some slight differences in the .pot and .cfg files for example:

[image]

[image]

So it looks like the input files have been modified. I don't see how any of these changes would introduce irregular frequencies in your results...but there could be another difference that I haven't caught.

Hence, I'd suggest just starting again ensuring that you have the _exact_ same files from the examples folder...run WAMIT with those files and see if you still get irregular frequencies.

Cheers,
Dave

dav-og, 2023-08-24:
I didn't re-run WAMIT - I used some existing outputs. Maybe @salhus or @dforbush2 can re-run the input files from `examples/BEMIO/WAMIT/RM3` and verify that the results still have no irregular frequencies.

salhus, 2023-08-24:
@antbog 
Hi Antoine,
Apologies for not reading your issue description carefully. Let me get back to you soon. As others have pointed, this does seem to be an unexpected issue.
Cheers,
Sal

dforbush2, 2023-08-24:
I'm running this case as we speak and will follow up as soon as it finishes. A few remarks:
1) @dav-og good eye. As far as I can see the differences here shouldn't make any difference w.r.t the calculation method, the change in the PER vector just removes the infinite frequency calculation.
2 @antbog though its counter-intuitive if you are used to the  conventions of other BEM solvers, the WAMIT convention (for use with WEC-Sim at least) is that we model each body with its local origin at the CG and then specify its global position in the *pot file. So your plot of the superimposed gdf that shows interference is not actually what wamit should be seen by WAMIT. 

Is it possible this is a version issue? I see version 7.2 in our example output file... and note you're running v7.4.

dav-og, 2023-08-24:
I wondered the same (v7.2 vs v7.4) but nothing obvious from the release notes:

[image]

[image]

https://www.wamit.com/manualv7.4/wamit_v74manualch1.html#x2-70001.6

dforbush2, 2023-08-24:
Checking the repo, it is the EXACT same WAMIT run that lives in examples/RM3, examples/BEMO/WAMIT/RM3/ and the RM3 tutorial (you can tell by the time stamps in the *out file). And running BEMIO on all, they are the exact same. These indicate that they are v7.2 outputs and they DO NOT have these spikes. 

Comparing to my fresh v7.4 WAMIT run of the inputs provided by @antbog to those found in  examples/BEMO/WAMIT/RM3/,  the outputs are the same and the spikes DO appear.

It is _possible_ that the *out file we have in the repo was long ago manually cleaned up (by a developer whose identity has faded into legend) to remove those spikes: my finishing v7.2 run will confirm this. If is shows the spikes, then this is what happened.  

**If if does NOT have spikes, then this is a WAMIT version issue that surfaces in v7.4. Developers should be notified. This has some serious implications and we should get to the bottom of it.**

@antbog Note that the internal "hole" of the float will cause some singularities that won't be corrected by the IRR. Assuming deep water, the radian wave frequency corresponding to the diameter of this hole is ~3.2 rad/s, which is about where the mess is. 

@salhus I do note that in your comparison the off-diagonal K terms are significantly different between the two runs, but they are the same in both of my outputs. That's odd and worth investigating.  

I still have a v7.2 run finishing, and will confirm when it finishes (I had to run this locally).

salhus, 2023-08-24:
Hi @dforbush2,
I think the above comparison is between @antbog 's out file and wecsim's existing rm3 out file. 
I ran a new run using the mesh files provided by @antbog (which is I think identical), and this is what I got 
[image]

I am about to finish another run where I set the IRR=2, will update soon.

salhus, 2023-08-24:
Also, I don't think IOPTN(4) is relevant to the spikes. Here is why.
I used the infinite frequency option and still got the spikes, and looking at the IRF based infinite added mass calculations the spikes do show up 
[image]

However, the inf. added mass calculated by enabling IOPTN(4) gives similar values,
[image]

dforbush2, 2023-08-28:
We are in correspondence with WAMIT w.r.t this issue. While we do not yet have an understanding of why v7.2 differs from v7.4/v7.4.1, we have learned that IRR > 0 may not work reliably for bodies with contacting surfaces, as we have in this case near the free surface between the float and spar (as @antbog pointed out). 

Based on our correspondence: it seems best practice for WAMIT with N multiple contacting bodies of M DOF each may be:
1) Run each body separately (i.e., N different BEM runs) with IRR = 1. These should be used as the M x M block-diagonal matrices of the full (N*M) x (N*M) matrices of radiation damping and added mass terms, and should correctly handle IRR = 1. 
2) Run all bodies together. These should be used to populate the off-diagonal blocks of the radiation damping and added mass terms not included with the above. 

Note that IRR =1 DOES appear to work for this case in v7.2: we're still working on understanding why this is, and, at least for the RM3 geometry, IRR = 1 in v7.4 does not appear to be working correctly even for a float only run.

salhus, 2023-08-28:
@dforbush2 
Running them separately doesn't make sense as the hydrodynamic couplings will be absent. 
I think running them separately was a way to check if the IRR is working not a prescription for running such a case.
I think like I said in a separate conversation at OREC,  I would recommend introducing a small gap between the float and the spar. As that way there won't be any intersecting normals. 

@antbog 
Either way, we would update the example case soon. This thread will be updated.

Cheers,
Sal

dforbush2, 2023-08-30:
We have gained more insight into the results of 7.4 and 7.4.1. 
[image]

These remaining frequencies are irregular frequencies that are not removed due to the contacting surfaces (exterior of spar cylinder, interior of float moonpool) causing the problem to be ill-conditioned. IRR = 1 "worked" in 7.2, but, according to WAMIT developers, this was not a robust result and this meshing/modeling approach is no longer best practice in v7.4 and v7.4.1. 

A study was performed in which the spar diameter was reduced. The results show that these irregular frequencies are removed, but a low frequency "pumping mode" resonance due to the float moonpool is NOT removed via IRR. It is "physical" in the sense that it is a real free surface, but is not realistic because in reality it will be sharply attenuated by viscosity/vortex formation etc. This feature is qualitatively robust to different spar diameter selections.
[image]

This implies that a "correct" approach will 1) not have interfering panels such that IRRs can be removed  and 2) represent the float in such a way that the pumping (and sloshing modes in surge and pitch, equally pronounced and problematic) are not present.

An attempt was performed with a shortened spar, such that there were no contacting panels between the float and spar. This resulted in the following: which does not appear improved.
[image]

Work continues and will inform future best practices. In the mean time @antbog see PR #1076 for a function that may be useful to you.

hachikoi1, 2023-09-03:
Dear [antbog](https://github.com/antbog), [salhus](https://github.com/salhus), [dforbush2](https://github.com/dforbush2) and [dav-og](https://github.com/dav-og)

I mentioned this issue in https://github.com/WEC-Sim/WEC-Sim/issues/1114#issuecomment-1704300869.
I hope my answer resolves your doubts

BTW,  [antbog](https://github.com/antbog), could you guide me on how to create circular mesh faces in Rhinoceros like this?

[image]

Best regards,
Tianyuan

antbog, 2023-09-04:
Hi @hachikoi1, 
I didn't create the circular mesh on Rhino, I only opened the .gdf files provided by WEC-Sim (examples > BEMIO > WAMIT > RM3)  on this software. So unfortunately I don't have answer to your question.

@dav-og @dforbush2 @salhus thank you for these updates, I'm looking forward for future ones and will look at #1076 function.

Cheers,
Antoine

salhus, 2023-09-28:
Hi Antoine @antbog,

Thanks for brining this issue. We are investigating this and developing a better example. Sorry about the long development time. Hopefully, this should be resolved in the next few weeks. This thread will be updated when the development efforts are complete.

Cheers,
Sal

hachikoi1, 2023-09-29:
Hi [salhus](https://github.com/salhus)
Here I share some manuscripts regarding the strange peaks in the excitation loads and radiaton coefficients
[A viscous damping model for piston mode resona.pdf](https://github.com/WEC-Sim/WEC-Sim/files/12755989/A.viscous.damping.model.for.piston.mode.resona.pdf)
[Relative Heave Motion Responses of Floating Du.pdf](https://github.com/WEC-Sim/WEC-Sim/files/12755990/Relative.Heave.Motion.Responses.of.Floating.Du.pdf)
[Hydrodynamic coefficients in heave of a moonpo.pdf](https://github.com/WEC-Sim/WEC-Sim/files/12755992/Hydrodynamic.coefficients.in.heave.of.a.moonpo.pdf)
From these three papers, the strange peaks are caused by the narrow gap between two bodies and they do exist in physics. The radiation coefficients calculated based on CFD also have strange peaks, see the third paper.
A general solution to this issue is to add a lid (dampling) to the gap (on the water surface) to avoid overestimating the water elevation in the gap. But the damping coefficient is based on the experimental or CFD results.
I hope this helps, and I'm looking forward to your deveploment.
Best regards,
Tianyuan

salhus, 2023-09-29:
Hi Tianyuan @hachikoi1 ,

Thanks for sharing those papers. And it is rather fascinating that the peaks are in fact physical. We will try to finalize if any changes are needed for this example, and update this thread in the coming weeks.

Cheers,
Sal

hachikoi1, 2023-10-30:
Hi 

Any progress on this issue? It's a question that also bothers me. 

From what I understand, in the linear frequency-domain calculations, the wave-excitation loads are obtained by fixing the two objects without any motion. And the wave-radiation loads are obtained by forcing one of the objects to move with unit ampitude in one direction.

For example, added mass matrix of the two objects has a size of 12*12, the added mass A(1,1) is obtained by forcing object 1 to move with unit ampitude in the surge direction, and A(1,7) is the added mass imposed on object 2 by the forced motion of object 1 in the surge direction. Likewise, A(7,7) is obtained by forcing object 2 to move in the surge direction, and A(7,1) is the added mass imposed on object 1 by the forced motion of object 2 in the surge direction.

Considering the existence of rigid constraints between the two bodies, should the radiation coefficients be calculated in the surge, sway, roll and pitch directions? For the outer bodies with a moonpool, the forced motions of one of the two objects in the four directions would cause large-amplitude radiation waves, especially if the gap between the two objects is small.

I am now worrying whether the strange peaks in the surge, sway, roll and pitch directions are caused by the narrow gap resonance? Or generated by the forced motions? If the strange peaks (in the surge, sway, roll and pitch  directions) are caused by the relative motion of two floating bodies during the forced motions (they don’t actually exist because of the constraints), what should be done with it?

Cheers,
Tianyuan

dforbush2, 2024-02-21:
Hi All,
[WAMITHelp_public.pdf](https://github.com/WEC-Sim/WEC-Sim/files/14364974/WAMITHelp_public.pdf)

I'm attaching a pdf of a powerpoint that shows our current understanding here. The quick summary is that the way we had meshed RM3 was ill-advised but "worked" in WAMIT v7.2. But in the v7.4 and later updates, the previous meshing approach results in these peaks that are undesirable for WEC-Sim, but these later WAMIT versions are more robust implementations of the solver so these results are apparently improved in this respect. I wish I had a clean, final, answer here but I think it will be a work in progress for some time. 

Best practices summary:
1) do not include moonpools unless they are a necessary, physical aspect of your device. There will be associated resonances related to its draft and radius that, without applying damping in the BEM model, will look like singularities. In the RM3, there isn't actually a moonpool, just a mated spar and float. In the improved mesh, the buoy is modeled as a solid.

2) Do not have interfering or overlapping mesh panels. In our example mesh the mated surface between the spar and the float interfered and this resulted in a poorly posed problem from the perspective of BEM.

3) Check with and without IRR: this appears to be acting counter-intuitively. 

4) increase mesh resolution. For the RM3 example, this was 2-3 times what is presented in the example files.

5) Use the badBEM_fix.m (on DEV Branch) cleanup function to remove things that look like singularities

hachikoi1, 2024-02-22:
Hi [dforbush2](https://github.com/dforbush2) @dforbush2 
Many thanks for your detailed discussions. A question that may sound silly: How to create circular mesh faces in Rhinoceros like this? Could you please give me some references?
[image]

Best regards,
Tianyuan

hachikoi1, 2024-02-22:
Hi [dforbush2](https://github.com/dforbush2) @dforbush2

I tested the codes you shared. A powerful tool! It seems to be developed based on an interpolation method, and is effective in eliminating sparsely distributed singularities singularities within a frequency range.

I've encountered bugs and made slight modifications. For example, Line 161:
```
for kkk=1:length(BLocs)
    if BLocs(kkk) > 2 & BLocs(kkk) < length(hydro.w)-2
        BRep = interp1([hydro.w(BLocs(kkk)-2);hydro.w(BLocs(kkk)-1);hydro.w(BLocs(kkk)+1);hydro.w(BLocs(kkk)+2)],[hydro.B(k,kk,BLocs(kkk)-2);hydro.B(k,kk,BLocs(kkk)-1);hydro.B(k,kk,BLocs(kkk)+1);hydro.B(k,kk,BLocs(kkk)+2)],hydro.w(BLocs(kkk)),'linear');
        hydro.B(k,kk,BLocs(kkk))=BRep;
    end
end
```
If a singularity occurs at the boundary of the studied frequency range, no interpolation is performed.

kmruehl, 2024-03-20:
@dforbush2 and @salhus migrate to discussion board