WEC-Sim/WEC-Sim #1111: [Theory or Implementation] Rotational instability

AudeMulard, 2023-08-23:
**Is your question request related to a problem? Please describe.**
I am trying to calculate the loads on a floating structure. 
- I have managed to get stable and physical answers for translations.
- When transitioning from 3DOF to 6DOF, the structure starts to have unstable and exponential accelerations in the no wave case. They stay very low (10^-8), but I think they are responsible for the drift I observe at regular wave case.
- In the regular wave case, my module starts stable and then drifts.

**Describe the theory or implementation approach that you have a question**
Looking at the documentation about the numerical test cases, I saw it was a problem linked to the mass distribution, and I have set "body.mass='equilibrium'". I am not sure what to do next (there are other tests of course, but as both nowaves tests seem to not fully be stable, I thought it best to solve them first). I am wondering whether it is a simulation or a problem with my design, and would be grateful for your opinion and help.

**Personal project context**
My project is to calculate the loads on a hollow hexagonal structure surrounded by a hexagonal floater. From these loads I am hoping to be able to dimension the different parts of my structure (added to that, I have a question about how to extract loads on the structure from the forces at the center of gravity without nonlinear simulation).

**Describe the type of conclusion or resolution you are looking for**
I am looking for a way to avoid these instabilities as I think they might lead to false results in irregular waves and nonlinear simulations, or dangerous behavior in the ocean.

Thank you for your help!!

**Additional context**
noWaveCIC: [image]
noWave (6s period)
[image]
Regular waves 3DOF:
[image]
Regular wave 6DOF: 
[image]

jtgrasb, 2023-08-23:
Hi @AudeMulard, thank you for your question. It is difficult to provide specific guidance on your case without having access to the case files, but I will do my best. Overall, your approach is relatively sound and its important to address the noWave issues first. The oscillations present for the noWave cases are very small though and they look like they're settling out for the noWave (6s period) case. For the noWaveCIC case though, the response is increasing over time - I would be curious to see if it continues to increase to a more significant level after more time. 

One thing to look into is the BEMIO plots for radiation and excitation IRF. I would think that the IRFs (especially for sway) may not converge to 0, which would indicate some instability issues with the BEM data. 

As for extracting the forces on the body, these can be found in the output struct after the simulation is complete. If you are looking for the forces acting on individual parts/panels of the body, the non-linear hydrodynamics option must be used.

jtgrasb, 2023-09-06:
@AudeMulard I am closing this issue due to inactivity, please let me know if you have further questions.

AudeMulard, 2023-09-08:
Hello @jtgrasb , thank you very much for your answer! Apologies for the delay in the answer. We changed our design quite a bit, so there was a lot to update.
I am happy to send the case files, the only issue is that it is a company project, so I can't share them here, would there be any other way to send them to you?

I am now trying to get working the 6DOF non-linear hydro irregular wave case, as it seems like the best way to extract the values of the pressure on the different surfaces. For that, I applied the same build-up process as previously and extended the time of simulation as you asked:
- Regular wave, linear Hydro, 3DOF:
[image]
- Regular wave nonlinear hydro 3DOF: problem in surge
[image]
- noWave CIC, linear hydro 6DOF (extended to 2000s from 400):
[image]
- noWave 6s, 6DOF:
[image]
- Regular wave linear hydro 6DOF:
[image]
- noWave 6s, nonlinearhydro, 6DOF:
[image]
- noWaveCIC, 6DOF, nonlinear hydro:
[image]

Here are my BEM data, which is not very pretty. I think it is because I am looking at a structure with a hole in the middle that is not dealt very well with by Capytaine, but I don't have access to Wamit. I am also not sure how to plot the other DOFs without messing up the source code.
[image]
[image]
[image]
[image]
[image]
[image]

Sorry for the many many pictures, I think I am a bit lost on how to solve the problem, and what information is relevant..

jtgrasb, 2023-09-18:
Thank you @AudeMulard for sharing the 6DOF example files. I was able to run them and replicate your results. To answer your questions:
•	"For the linear 6DOF, regular waves, there are some small movements in the rotation DOFs, but they are limited to under 10e-3 in terms of amplitude. Can I ignore them?"
It is not unexpected that you would experience some small motion in rotational dofs. The results are small and still stable in this case, so I would not be concerned with it.

•	"For the nonlinear 6DOF, regular waves, there are large yaw movements: it seems like there is an increasing velocity, and I can't understand why. Could it be because of the wave orientation and its interaction with my geometry? I have put two STLs with different sizes. The velocity does decrease with the small mesh (lower velocity), but the position change still happens, and I was thinking there might be something else going on."
The result seems to be based on a rotational instability in yaw as previously mentioned. Although a finer mesh likely helps, it may just be an imperfection in the mesh that causes some excitation in the yaw direction. Since there is no restoring force in yaw, this can increase exponentially, Thus, I believe it is a small numerical issue. One way to solve this would be to add some stiffness and damping in the yaw direction the mooring system. Another option would be to stick with a 3DOF constraint. For an axisymmetric body (which it seems your device is) with waves at 0 degrees, we do not expect significant motion in the sway, roll, and yaw directions, so any motion in these directions is likely to be numerical noise. This old issue #52  explains some of the relevant concepts.

closed 2023-10-02 by jtgrasb