WEC-Sim/WEC-Sim #1112: [Theory or Implementation]

mprevisic1, 2023-08-23:
**Is your question request related to a problem? Please describe.**
I implemented a two-body system that consists of two barges connected by a rotational PTO in the center. The system is freely floating. I am naming the two bodies fore and aft barge. It seems to me that I should be able to place the PTO in the center of the two bodies, but no matter what I do, it seems to me that the hinge point does not move to between the two bodies. The overall system seems to pitch around a fixed point at xyz = [0 0 0], no matter what I do (as shown in the Mechanics Explorer). I tried to replace the rotational PTO with a linear PTO, but am finding the same issue. 

**Describe the theory or implementation approach that you have a question**
Could you clarify if the setup is correct and if this might be a bug in the Simulink library?

**Personal project context**
Trying to simulate a hinged barge. 

**Describe the type of conclusion or resolution you are looking for**
I need to be able to re-locate the X position of the rotational PTO.

**Additional context**
Please find enclosed simulation files. 
[Hinged_Barge.zip](https://github.com/WEC-Sim/WEC-Sim/files/12423590/Hinged_Barge.zip)

dav-og, 2023-08-24:
Hello @mprevisic1,

Thank you for providing your input files - I was able to reproduce the problem.

I think there are a few things going on here.

Checking the fore body:

[image]

The mesh is 15000 meters long - so that needs to be remade/rescaled by a factor of 1/1000 so that the mesh is 15 meters long. (Same goes for the aft body).

I'm not sure if this is just affecting the .stl meshes or if its affecting your hydrodynamic meshes as well - you'll have to check if they are ok (I assume those meshes are ok).

The other thing is that in your .h5 file the CG of the fore body is at x=0:

[image]

However, the mesh goes from 0 to 15m - so its like the CG is on the nose of the fore body. I assume that should be in the middle of the fore body (~7.5m)? If so, you'll probably need to re-run the BEM code for a mesh extending from 0 to 15m along the axis, with CG at 7.5m. Alternatively, you could keep the CGx at 0.0m and use a mesh that goes from -7.5m to +7.5m - whatever works best for you.

I hope that makes sense - let me know if you need further clarification 👍

Cheers,
Dave

mprevisic1, 2023-08-24:
Thanks for this feedback. I created the geometry in Solidworks and just
exported the STL file. My unit was set to m (for a 15m x 15m body), but I
seem to vaguely remember now that there was an issue with conversion/units.

Mirko

mprevisic1, 2023-08-25:
I recreated the STL files to properly match my Nemoh input geometry. Specifically, the system consists of two barges measuring 30m length x 15m width x 2m draft. The fore barge is centered around XYZ = [0 0 0] so that the Center of Buoyancy is at [0 0 -1]. The aft barge has the same dimensions, but is offset in X by 32m, so that the Center of Buoyancy is at [32 0 -1], leaving a gap of 2m between the two bodies. The pto sits between the two bodies at [16 0 0].

When I run the model in WEC-Sim it shows the following in the Visualization:
[image]

You can see the bodys reference frame that seems to start at XYZ = [0 0 0] as well as the PTO location between the two reference frames. Here is the strange thing:
1. The reference frames still do not line up with my geometry. I am assuming the yellow bodies are generated from the .stl file. 
2. The center of mass is located at [0 0 0] of body 1. That center of mass should be in the center of the body, not at the edge. 
3. When I change the location of the rotary PTO, it does not change position in the visualization. 

Do I still have some issues with the geometry definition? I generated the stl file in Solidworks using mm instead of m - a scaling factor of 1:1000. Digging a bit deeper, I looked at the hydro file that is generated using Bemio. It shows the following:
[image]

It seems that the two bodies have the proper cg. However, I am not clear why the Vo values for the two bodies are 1800m^3. They should be 15*30*2 = 900m^3. It seems like it's reporting the combined value of both bodies. Related to the same, looking at the hydro stiffness matrices for the two bodies I show the following:
[image]
As a check, the hydrostatic stiffness in Z for each body should be  15 x 30 x 1025 x 9.81 = 4.5MN/m or 4.5 e6. Looking at the KH.dat Nemoh output file, I see the same values reported (minus normalization), but it is 9e7 in 3/3 (Z direction of body 1) of the matrix. So in a similar fashion, it seems to report the hydrostatics for the 2-body system combined. The WEC-Sim instructions seem to suggest that we just copy the KH.dat file into the KH_1.dat and KH_2.dat files which would obviously lead to this error. Do I need to compute these matrices manually because Nemoh doesn't seem to handle that properly? 

I am including an updated version of the WEC_Sim input files. 
[Hinged_Barge2.zip](https://github.com/WEC-Sim/WEC-Sim/files/12443514/Hinged_Barge2.zip)

mprevisic1, 2023-08-28:
I went through this exercise one more time, step by step. Now the model displays properly in the SimMechanics explorer. I did have to utilize Meshmagic to compute the hydrostatic spring-stiffness matrices and moments of inertia for the two bodies. Here is what I learned:
1. The Nemoh outputs for KH_0 and KH_1 are not accurate and I relied on Meshmagic instead to compute values.
2. The Nemoh outputs for Hydrostatics_0 and Hydrostatics_1 are not accurate and I instead used values from Meshmagic.

The body coordinate system for the stl file provided to the SimMechanics explorer is a local coordinate system, not a global one. This led to a bit of confusion when setting up the system. It's probably somewhere in your documentation, but I did not find the reference. 

Feel free to close this issue as resolved.

closed 2023-08-28 by mprevisic1