WEC-Sim/WEC-Sim #1113: [Theory or Implementation] Why is the total force not equal to the sum of separate forces?

Dadidal, 2023-08-24:
**Is your question request related to a problem? Please describe.**
N/A

**Describe the theory or implementation approach that you have a question**
I conducted the simple simulation using WEC-Sim and tried to compare output forces. I summarized the particular forces (restoring, added mass, radiation damping, linear additional damping and wave excitation) and compared with output output.bodies.forceTotal.
And I observed that by simple summation with "plus" signs we cannot obtain the total forces provided in output.bodies.forceTotal. 

Then I looked to the manual and understood that added mass, radiation damping and linear additional damping forces has to be with "minus" sign. After that I used "minus" sign for abovementioned forces and "plus" sign for restoring excitation forces and result converged with output.bodies.forceTotal.

BUT for rotational DOFs (Pitch, Roll, Yaw) the convergence can be obtained only if use use the "minus" sign for restoring force too. Why?

**Personal project context**
N/A

**Describe the type of conclusion or resolution you are looking for**
I want to understand why the total force is not just the sum of all forces and signs of particular forces has to be changed for rotational DOFs. Also why the restoring force have to be summed with "plus" sign for translational DOFs and with "minus" sing for rotational DOFs.

**Additional context**
N/A

jtgrasb, 2023-08-24:
Hi @Dadidal thank you for your question. When combining the forces in Simulink, the excitation is added while the other forces are all subtracted. This is the case for all dofs. Perhaps the restoring forces for the translational dofs were not significant enough to affect the total force? It is hard to tell without seeing your exact case, but hope this helps! 

Let me know if the excitation force minus all the other forces does not equal the totalForce.

Dadidal, 2023-08-25:
Hello @jtgrasb! Thank you for your explanation. You were right, I forgot that restoring forces are 0 for surge and sway DOFs. I have one more little question:
Where is the origin of the coordinate system to which all the forces are applied? Center of Buoyancy or Center of Gravity of the hydrodynamic body?
I want to clarify that moment because it seems that displacements, velocity, and accelerations used to calculate the forces are tracked for CoG, but hydrodynamic forces have to be applied to CoB (am I correct?).

jtgrasb, 2023-08-25:
Hi @Dadidal. WEC-Sim calculates all forces about the center of gravity. Similarly, any BEM inputs (hydrodynamic coefficients) are required to be calculated at the center of gravity. Although the hydrodynamic forces do act on the center of buoyancy, BEM converts results to be about the cog so that WEC-Sim/Simulink can properly handle the forces.

jtgrasb, 2023-09-06:
@Dadidal Closing this issue - let me know if you have further questions.

closed 2023-09-06 by jtgrasb