WEC-Sim/WEC-Sim #1114: [Theory or Implementation] Questions about hydrodynamic modelling in WAMIT

hachikoi1, 2023-08-26:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_A clear and concise description of what the problem is. Ex. I do not understand why I get 'X' result when running 'Y' simulation [...]_

**Describe the theory or implementation approach that you have a question**
_Please clearly describe the assumptions, equations, or code script that you would like clarification. Also, please provide references to other research papers, open-source material, or other information that may be relevant in the WEC-Sim developers understanding your question._

Dear [salhus](https://github.com/salhus):

I recently encounter some problems with the hydrodynamic modelling in WAMIT, regarding XCG in .frc file and XBODY in .pot file.
In WAMIT User Manual v7.4 (http://www.wamit.com/manualupdate/v74_manual.pdf), I found almost all the models use XBODY(1) and XBODY(2) as the location of body coordinates in the xy plane, and use VCG as vertical center of gravity (COG) in the z direction. If I understand correctly, the geometric center of a body is defined by XBODY, where XBODY(3) is always zero, and XCG is the COG location. 
(i) For a body with an axisymmetric shape but with an offset COG (-1, -1, -1), XBODY(1) = XBODY(2) = XBODY(3) = 0, XCG(1) = XCG(2) = XCG(3) = -1. 
(ii) However, for the axisymmetric body with the geometric center located at (-1,-1,0) and the vertical COG at (0,0,-1), XBODY(1) = XBODY(2) = -1, XBODY(3) = XCG(1) = XCG(2) = 0, and XCG(3) = -1. 
(iii) For case (ii), XBODY(1) and XBODY(2) should be set to 0 if there is only one body, and XBODY(1) and XBODY(2) should be set to -1 if there are multiple bodies.
Am I right?

I can understand most frequency-domain simulation configurations in the WEC-Sim_Applications-master, but some cases, for example, \WEC-Sim_Applications-master\WECCCOMP\hydroData\wavestar.out, I can't understand why neither XBODY nor XCG is zero.
[image]

Similarly, in \WEC-Sim_Applications-master\Traditional_Morison_Element\hydroData\monopile.out, both XBODY(3) and XCG(3) are -15. Does this mean that the body coordinate system is located at (0,0,-15), and the COG in the body coordinate system is (0,0,-15)? That is, the COG in the global coordinate system is (0,0,-30)?
[image]

For a body, I found that the simulation configuration with XBODY = (0,0,-15) and XCG = (0,0,0) is not equivalent to that with XBODY = (0,0,0) and XCG = (0,0,-15). I'm confused about it, could you give me some guidance?
[image]
[image]

Beat regards, 
Tianyuan

**Personal project context**
_If this request is related to your project, please describe what you are attempting to do in your project so the WEC-Sim development team knows your end goal._

**Describe the type of conclusion or resolution you are looking for**
_[Ex.] I am still unclear on how an irregular wave train is generated within WEC-Sim and would appreciate being pointed to where in the documentation and code where I can find additional information._

**Additional context**
_Add any other context or screenshots about the feature request here._

dforbush2, 2023-08-28:
Hi @hachikoi1 

I conferred with @salhus on this and this is the consensus as to the proper approach. By default, WAMIT assumes that the input is in hydrostatic equilibrium, and WEC-Sim requires that the hydrodynamic coefficients are calculated about the CG.

In combination, this implies that a system that is set up in wamit as follows:
1) the XBODY(3) in *pot should describe offset CoG under the loaded condition. Your GDF file should have this CoG at its origin.
2) Following the definitions in (1), XCG YCG and ZCG = 0 in *frc.

 However, you will need to account for the mooring-exerted forces somehow: if you do this in WEC-Sim, you should NOT do this in WAMIT. However, if you don't model the mooring forces in WEC-Sim, you should do so im WAMIT by inclusion of an EXSTIFFNESS, EXDAMPING, and EXMASS matrix using IALTFRC =2 form of your *frc file.

We recommend the former work flow.

dforbush2, 2023-08-28:
Looking at the files on your repository, it seems that the GDF is consistently defined incorrectly for use with WEC-Sim: the origin of this GDF should be at the loaded CoG of the body, and *pot XBODY should reflect the offset from this point to the free surface, (as it seems to in some of your cases).

I suspect the combination of the incorrect GDF definition with some combinations of POT and FRC explain the variation in hydrostatics that you observe.

salhus, 2023-08-29:
Hi Tianyuan @hachikoi1, 

Nice to see you again. Thanks @dforbush2 for answering and stating everything very clearly. Following are some further comments that may support the above comments.

Traditional maritime conventions place the body-frame at the water surface. In WAMIT, the XBODY location is the location of the body-frame. The body-frame, when it comes to WAMIT is independent of the body CG. However, the body CG is expressed using the XCG which is expressed **_with respect to the body-frame, i.e., the XBODY._** 
[image]

Now, when using WEC-Sim we recommend setting up the geometry file, so it has its CG at $[0, 0, 0]$, the advantage of that being that, 
1. The XCG can then be set to $[0, 0, 0]$, and the XBODY can then be used to move the body to the location it will be with respect to the global reference frame,
2. The hydrodynamics calculations are done with respect to the body-frame. By having the XCG at $[0, 0, 0]$, the hydrodynamics coefficients can then be calculated at the center of gravity -- which is necessary for setting up WEC-Sim in Simscape multibody. All external forces such as hinges, PTOs, etc. can then be suitably setup and necessary geometric transformations such as Euler rotations can then be done in WEC-Sim.

Note, WAMIT assumes that the input geometry is in hydrostatic equilibrium. The hydrostatic equilibrium maybe achieved with external forces such as the mooring pre-tension, but the input geometry needs to be in hydrostatic equilibrium. 
Now the main questions are,
1. How to model a body that needs external forces to get to a target position that will be the hydrostatic equilibrium?
2. How are the hydrostatic coefficients calculated?

1. 
This can be done either, a.) in conjunction with WEC-Sim, or any analytical modeling framework that can impose external forces in the time-domain model), b.) in WAMIT,
   a. In WEC-Sim: Run WAMIT using FRC=1, where WAMIT assumes the body's inertia matrix based on the immersed volume of the input files, i.e., the mass of the body is equal to the mass of the displaced fluid, i.e., the body has the same density as water. The external forces can then be imposed in WEC-Sim, using moorings (using moordyn coupling or the stiffness/damping matrices for the mooring forces), or siply use other joints like cables, constraints or PTOs. The external forces in WEC-Sim will account for any discrepancies in buoyancy,
   b. Everything in WAMIT: Now one should use FRC=2, where WAMIT doesn't assume body density to be the same as water density. The user needs to provide the external forces as EXMASS, EXDAMP, and EXSTIF matrix. 

2. The hydrostatics coefficients,   
Now recall that maritime conventions set the body-frame at the water surface. By providing the VCG functionality, WAMIT allows one to experiment with the effect of changing the metacentric height (physically achieved by changes in ballast) without having to generate a new mesh.
In the image below, the $z_{cg}$ corresponds to VCG.
[image]

The important thing to remember is that the hydrostatics are also being calculated with respect to the body-frame XBODY.  

Here is an excel sheet that explores the effect of choosing XBODY and XCG. I analytically calculate the pitch hydrostatics coefficients and compare them with different WAMIT runs.
[Pitch hydrostatics of a cylinder.xlsx](https://github.com/WEC-Sim/WEC-Sim/files/12459722/Pitch.hydrostatics.of.a.cylinder.xlsx)

Now to your specific questions,

1.  To your first question, 
I think you correctly observed that the intended reference point is where the coefficients are being reported 
"The hydrostatic restoring matrix differences between configurations (iii) and (iv) are caused by the difference in reference point ?"
So yes, the choice of XBODY drives this. 
In configuration (iii) XBODY(1) = XBODY(2) = VCG = 0, XBODY(3) = -14.4, The entire body is 'moved' down as the XBODY drives the location of the body. (I think).  It is a good practice to check what wamit interpreted. WAMIT produces a #_pan.dat file that can be plotted to see what WAMIT sees. If the entire body is immersed, it would explain the heave C33 to nearly 0, and a CG above the CB will explain the negative signs for the rotational C44 and C55. Here is a script that will help you plot the #_pan.dat script. It is written as a function, so just input the part in front of the #_pan.dat file (make sure to enable the setting that makes WAMIT output that file) Here is the code, (save the file as an m file...)

[GDF_Mesh.txt](https://github.com/WEC-Sim/WEC-Sim/files/12459770/GDF_Mesh.txt)

In (iv) XBODY(1) = XBODY(2) = XBODY(3) = 0, VCG = -14.4. No such translations occur but the Zcg is being moved, thereby affecting the rotational hydrostatic coefficients.

The important thing to remember is that different .gdf files will be needed when using XBODY= $[0,0,0]$, and XBODY = $[0,0,-14,4]$ because XBODY _moves_ the geometry before WAMIT is run. So if XBODY = $[0,0,-14.4]$ is to be used (as you would need to, in order to use WEC-Sim) the geometry needs to be translated vertically upwards by $+14.4 m$, which will then be moved by WAMIT vertically downwards. This will achieve two things, 1. Enable hydrostatics and hydrodynamics calculations with respect to CG -- this is because WAMIT calculates everything at XBODY. 2. By this initial translation we force WAMIT to calculate everything.

------------
Q3. The WECCOMP examples.
I will write a seperate note on that.

Cheers,
Sal Husain, Ph.D,
NREL.

hachikoi1, 2023-08-29:
Dear [salhus](https://github.com/salhus) and [salhus](https://github.com/salhus)

Thank you very much for your detailed explanations ! I have tested the simulation configuration with panel model shifted up by +14.4m, XBODY = (0,0,-14.4) and VCG = 0.
As expected, AQWA and WAMIT are in good agreements.
Radiation damping:
[image]

Salman, the code you shared is very useful! 

I will post my case [here](https://github.com/hachikoi1/UMaine.git) for anyone else who needs it

Thanks again for all your help. 

Best regards,

Tianyuan

salhus, 2023-08-29:
Thanks for letting us know Tianyuan ! Thanks for sharing your work.
I am happy you think it helped. Let us know if you have any other questions. If not, we may close the issue?

Cheers,
sal

hachikoi1, 2023-08-30:
Dear [salhus](https://github.com/salhus) and [dforbush2](https://github.com/dforbush2)
    I wish to keep this issue for a few more days. I am testing the multi-body cases with multiple WECs integrated on the floating platform, for example, the W2POWER concept.
[image]
   The calculation speed of multi-body cases is slow and I have no results yet. I have some questions about the calculation settings, could you give me some suggestions?
   According to the WAMIT user manual, e.g.,  test05, I found the panel models are located at (x,y) = (0,0), and WAMIT offsets the models using XBODY. 

Figure 1: test05.gdf
[image]

Figure 2: test05.pot
[image]

% =============================================
Q1:
As for the hydrodynamic modelling for WEC-Sim, I want three toroids placed on the side columns of the platform (Figure 4), and the torus and side column are concentric. The COGs of WECs are on the SWL. When the geometry center of panel model (.gdf) is the same as XBODY (e.g., the geometry center of WEC1 is -51.75,0,0 and XBODY = -51.75,0,0), the resulting '_pan.dat' file is shown in Figure 3.

Figure 3: The geometry center of panel model is the same as XBODY: 
[image]
[image]

I tested two ways to handle this: (i)  The geometry center of WEC1 panel model is (0,0,0), XBODY of WEC1 is (-51.75,0,0) (ii) The geometry center of WEC1 panel model is (-51.75,0,0), XBODY of WEC1 is (0,0,0).
The two settings generate the same '_pan.dat' in Figure 4, I am not sure which one is right for WEC-Sim. Referring to WAMIT's official cases, configuration (i) seems more reasonable. 

Figure 4: The geometry center of WEC1 panel model is (0,0,0), XBODY of WEC1 is (-51.75,0,0) // The geometry center of WEC1 panel model is (-51.75,0,0), XBODY of WEC1 is (0,0,0)
[image]

% =============================
Q2:
Now the WECs have a COG on the SWL. If the COG of WECs offsets downward by -5 m, should I set the origin of WEC1 panel model to (0,0,-5) (move upwards by +5m so that the COG is located the origin), and set XBODY of WEC1 to (-51.75,0,-5).
I executed this setup, but the calculation is not yet complete. It looks correct from '_pan.dat'.

Figure 5: The origin of panel model is located at COG
[image]

% ==============================
Q3: 
I am confused about the reference point of WEC's hydrogynamic coefficients. Is it with respect to the origin of local body coordinate system, or with respect to the origin of global coordinate system? This will affect the hydrogynamic coefficients in the roll, pitch and yaw directions.

[image]
[image]
[image]

I think this is essentially related to the WEC-Sim workflow for body-to-body cases. Does WEC-Sim centralize directly all loads to the global coordinate system origin, and the constraints (e.g., translational constraint) just impose a constraint equation ? Or the loads are in the local body coordinate system, and the constraints also have a function in transferring loads from one local coordinate origin to another ?
[image]

% ==============================
Q4:
It's about the lid method to remove the irregular frequency. For a outside torus WEC with a gap to the inner cylinder, both BEMRosetta and WAMIT (IRR = 3, IOR = 1) generate a lid covering the entire innner diameter of outside torus, including the gap (Figure 6). Is it necessary? Or just add a lid for the torus (Figure 7)

Figure 6
[image]

Figure 7
[image]

I'll post the results here once the calculations are complete.

Looking forwards to hearing from you

Best regards,

Tianyuan

hachikoi1, 2023-08-30:
AQWA results and WAMIT input files are in [here](https://github.com/hachikoi1/Multibody.git) .
In AQWA, the COG of WEC is offset. The COG of WEC1 is (-51.75,0,0). It seems that AQWA outputs the hydrodynamic coefficients with respect to the origin of global coordinate system, because the roll-roll and pitch-pitch coefficients of the three WECs are different. (If the coefficients are with respect to the local WEC coordinate origin, the coefficients should be the same)
[image]

hachikoi1, 2023-09-03:
% ==============================
Q4:
% ========================
Let me add some clarifications on Q4:
Generally, the internal lid method is used to remove the irregular frequency. But for a coaxial double-buoy WEC (e.g. RM3), narrow gap resonance phenomenon must be considered. A common practice is to add a gap-length damping disk to the inner or outer WEC, or add the damping disk as a separate body. I was wondering if it is possible to create a circular surface so that it encloses the moon pool of the outer WEC. But it doesn't seem feasible.

Damping disk on the plaform 
[image]

Damping disk on the WEC
[image]

Separate damping disk
[image]

In WAMIT, a  ring-shaped damping disk is established, with the same area as the gap area. The damping disk is attached on the outer bodies, and the damper coefficient is 0.5. The approach is the same as test02a in WAMIT User Manual.
[image]

A similar damping disk attached on the outer body is established in AQWA
[image]

The results will be posted here as soon as the calculations are complete.
% ========================
I will continue to look for the causes of the differences between AQWA_Platform_COG00144_WEC_COG517500 and WAMIT_Platform00144_XBODY00144_VCG0_WEC000_XBODY517500_VCG0.
It's about the lid method to remove the irregular frequency. For a outside torus WEC with a gap to the inner cylinder, both BEMRosetta and WAMIT (IRR = 3, IOR = 1) generate a lid covering the entire innner diameter of outside torus, including the gap (Figure 6). Is it necessary? Or just add a lid for the torus (Figure 7)

Figure 6
[image]

Figure 7
[image]

I'll post the results here once the calculations are complete.

salhus, 2023-09-06:
@hachikoi1 
Hi Tianyuan,

Sorry about the late follow-up. A general comment, WAMIT evaluates everything at the XBODY, i.e., the local-frame; hence the geometric manipulations so that the hydrodynamics are at the center of gravity. Once we have everything at the center of gravity, the dynamics in WEC-Sim can be calculated -- which are calculated at the center of gravity. To your questions,

Question 1: BEM for multi-body systems in WEC-Sim,
----------------------------------

Option 1. is correct. Basically, define the geometry (.gdf) so that the center of gravity is at $[0, 0, 0]$ and then use XBODY to place them. This is typically the case and applicable when you have the same .gdf file and multiple instances of the same body, e.g., a WEC-array or the tension-leg platform you shared.

Question 2:  Defining geometry and XBODY,
------------------------------------------------
So say the body at $[x, y]$ in the $x-y$-plane, has its center of gravity at $[x,y,-a]$, where $\set{a}>0$. 
1. Now transform the body so that the body center of gravity is at $[0, 0, 0]$, which in this case will imply a transformation of $-1*[x, y, -a]$,
2. Set $XBODY$ to $[x, y, -a]$.

This would then enable WEC-Sim to read the hydrodynamics calculated at center of gravity, and read the intended center of gravity from $XBODY$ and correctly set it to $[x, y, -a]$.

Question 3: Calculation of Dynamics in WEC-Sim,
------------------------------------------------------

As suggested above, WEC-Sim calculates the body-dynamics at the center of gravity. An easy way to check is to look at the body block (the Simscape body block). Notice, that the "R" frame is same as center of gravity.

Question 4: Multibody systems with intersecting normals,
---------------------------------------------------------------

I find your discussion rather impressive. Let me get back to you. I see you appreciate the problem BEMIO will face when introducing additional bodies. 
The main issue is having normals that overlap each other -- and especially intersecting at $\pi rad.$ -- making the Green's Theorem solution ill-conditioned. 
$IRR=1,3$ (or $IRR=0$ with some sort of lids) may all ultimately face the same challenge. I have personally found a somewhat useful, albeit round-about way of handling this. I used $IRR=2$ for one of the bodies (preferably the outer annular body), and the lid solutions for the internal body. This results in a much higher irregular-frequency which can then be simply truncated from IRF calculations, filtered out using BEM-Rosetta, or post-processed in BEMIO using the PR #1076 

Let me know your thoughts.
Cheers,
Sal

hachikoi1, 2023-09-06:
Dear [salhus](https://github.com/salhus)
Many thanks for your reply.

Regarding Q4, I have a feeling that the strange peaks marked by red lines are caused by irregular frequency. For the peaks marked by blue lines, perhaps this is a real physical phenomenon that is related to the resonance of fluids and structures, and has nothing to do with the irregular frequency. 

[image]

Since the BEM method assumes that the fluid is inviscid and irrotational, the wave elevation in the gap is overestimated, and correspondingly, the hydrodynamic coefficients near the body's intrinsic frequency are overestimated. Test03a in the WAMIT User Manual employ a damping disk to suppress the excessively large wave elevation in the gap.

But I can't understand How does WAMIT 7.2 eliminate these resonance results (https://github.com/WEC-Sim/WEC-Sim/issues/1110#issue-1863183587). Or the RM3 results have been filtered using BEM Rosetta.

I look forward to the explanation given by wamit.

Best regards,
Tianyuan

salhus, 2023-09-28:
@hachikoi1 
Hi Tianyuan,

If the strange peaks is the only outstanding issue on this thread, do you mind if I close this issue and direct all strange-peaks discussion to Antoine's thread (@antbog) #1110 

Cheers,
Sal

hachikoi1, 2023-09-29:
Dear [salhus](https://github.com/salhus)
I agree to merge this issue and https://github.com/WEC-Sim/WEC-Sim/issues/1110. I recently have discussed this issue with Lei Tan, who has publised related paper in JFM. I will make some comments in [A](https://github.com/WEC-Sim/WEC-Sim/issues/1110)

salhus, 2023-09-29:
Thanks @hachikoi1 
I will keep you posted on the other thread.

closed 2023-09-29 by salhus