WEC-Sim/WEC-Sim #1115: [Feature request] Simulink Temp-File Issue

mprevisic1, 2023-09-01:
**Is your feature request related to a problem? Please describe.**
I used the wecSimMCR function to sweep a range of variables in a WEC-Sim based parametric study. I noticed that Simulink kept growing it's data inspector temp files by about about 20GB for a series of a few hundred runs. I resolved the issue by modifying the wecSimMCR function as follows:
[image]

As you can see from the above, I am issuing the Simulink.sdi.clear command in the main iterative loop. This command clears out the temp-files created by the Simulink data-inspector. Problem solved.

salhus, 2023-09-06:
@mprevisic1 
This seems like a very helpful suggestion. Would you like to put in a PR for this? 

Cheers,
Sal

mprevisic1, 2023-09-06:
Sal,

I am not sure what PR is.

Mirko

salhus, 2023-09-06:
Hi Mirko @mprevisic1,

No worries at all. I can test out your recommendation. A PR is a pull request -- a formal merger process that ensures compatibility with the source code. https://wec-sim.github.io/WEC-Sim/master/developer/pull_requests.html

However, as I said, I will test out your recommendation and if everything checks out I will put in a PR  and tag you.

I do have a question about the model you are using though.
Do you have any Simulink data logging enabled signals in your model?
As I don't believe we use any data logging in any of the library block. If it is specific to your application, it is good to know you were able to improve computational costs but the recommendation may be not necessary for the general source code.

Cheers,
Sal

mprevisic1, 2023-09-06:
The model I tested this on is very similar to your RM3 model, with the
exception that I use a rotary PTO instead of a linear PTO.

Mirko

akeeste, 2023-10-17:
@salhus @mprevisic1 
So is this issue related to your specific model, which possible has additional logging implemented, or a general WEC-Sim model?

I don't believe we've encountered this issue before in other WEC-Sim MCR cases. If it is model-dependent, @mprevisic1 I recommend that you keep this change locally. However we would not merge it into the source in case it causes issues for other users.

mprevisic1, 2023-10-17:
It would surprise me if this is not a general issue. You can check your
disk space before and after running MCR (excluding any output files you are
archiving) and you should see a decrease in available disk size.

Mirko

MShabara, 2023-10-18:
> The model I tested this on is very similar to your RM3 model, with the exception that I use a rotary PTO instead of a linear PTO.

I just tested the RM3 MCR, using rotational PTO, and the temp file was few MB. I don't know why in your case it was in the order of GB. If this issue occurred again with other users in the future we will consider your suggestion. 
Currently we will consider the issue Closed.

closed 2023-10-18 by MShabara

mprevisic1, 2023-10-18:
That's surprising - which temp file did you check? Did you check the
remaining disk space to see if you are slowly filling up your disk?