WEC-Sim/WEC-Sim #1116: [BUG] Phase angle of wave excitation loads for offset body, and added mass and radiation damping matrix for multibodies in AQWA

hachikoi1, 2023-09-03:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
_A clear and concise description of what the perceived bug is._

Dear [salhus](https://github.com/salhus) and [dforbush2](https://github.com/dforbush2)

I open a new issue here to talk about the Phase angle of wave excitation loads for offset body, and added mass and radiation damping matrix for multibodies. The issue is was originally posed in **A designed two-body hinged raft wave energy converter: From experimental study to annual power prediction for the EMEC site using WEC-Sim** by Siya.

Let me briefly describe the problem. For an offset body with its COG away from the origin of global coordinate system in the XY plane (this configuration is usually found in multibody cases), the phase angle output by AQWA is defined as the local parameter phase angle. But the phase angle output by WAMIT is defined as the global phase angle, which is equal to the local parameter phase angle + global wave phase angle (global wave phase angle is defined as the phase angle of the local coordinate origin with relative to the global coordinate origin, and is related to the spatial location of the body)

AQWA
[image]

WAMIT
[image]
[image]

Based on previous discussions (https://github.com/WEC-Sim/WEC-Sim/issues/1090#issue-1828915005), for a body with its COG located at the origin of global coordinate system in the XY plane

 PhaseAngleWAMIT = -PhaseAngleAQWA

But the definition is not perfect, because it is given based on the premise that the wave phase angle is zero (COGx = COGy = 0). For an offset body, the correct phase angle should be 

PhaseAngleWAMIT = -PhaseAngleAQWA - k * [x * cos(beta) + y * sin(beta)]
where PhaseAngleAQWA is the phase angle ouput by AQWA, k is the wave number, x and y are locations of the body, and beta is the wave heading. The corrected code in **readAQWA.m** is:

[image]

The simulation configuration is the same as Figure 1 in https://github.com/WEC-Sim/WEC-Sim/issues/1114#issuecomment-1704300869.

The real parts of wave excitation loads in 0 degrees wave heading obtained based on **hydro(F).ex_ph(((ind(1)-1) * 6+1):(ind(1) * 6),ind(2),ind(3)) = -tmp2 * pi/180** are 

[image]

The corrected results in 0 degrees wave heading are 

[image]

The corrected results in 10 degrees wave heading are 

[image]

% ======================================
Another issue arises in the added mass and radiation damping matrices

In WAMIT, the hydro.A(1,7,1) is -1.808348000000000e+03, and hydro.A(7,1,1) is -1.811229000000000e+03

[image]
[image]
[image]

In AQWA, the dimensioned hydro.A(1,7,1) * 1025 is -1839700, and hydro.A(7,1,1) * 1025 is -1733900. However, they are different from the results in .AH1 file
[image]
[image]
[image]

The corrected code in **readAQWA.m** is
[image]

I hope these efforts meet with your approval

Best regards,
Tianyuan

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to '...'_
2. _Click on '....'_
3. _Scroll down to '....'_
4. _See error_

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

salhus, 2023-09-06:
@hachikoi1 
Hi Tianyuan,

I noticed you are using the older WEC-Sim releases? Does the recent fix address this? I think I add a phase of $\pi$ instead of just negative sign. I should have thought about the wave-heading argument when trying to convince questions about wave-directions.

Let me know if v5.0.1 resolves this?

Cheers,
sal

hachikoi1, 2023-09-06:
Hi [salhus](https://github.com/salhus)

The current WEC-Sim v5.0.1 does not seem to address the issue. The recent fix does not consider the case of an offset body with its COG away from the origin of global coordinate system in the XY plane.

Best regards,

Tianyuan

hachikoi1, 2023-09-24:
Hi [salhus](https://github.com/salhus)
Perhaps I made a mistake in the treatment of added mass and radiation damping. In WEC-Sim, the cross coupling of boty-to-body case always place the accelerations of BODY 1 at A (1:6), and place the accelerations of BODY 2 at A (7:12), and so on.
[image]
Therefore, the added mass (1,2,wave frequency) in AQWA .AH1 file should be placed to (7:12,1:6) of added mass matrix during the time-domain caculations, and added mass matrix (7:12,1:6) would multiply with A (1:6). The results of added mass matrix (7:12,1:6) * A (1:6) represent the effects of the radiation wave caused by the motions of the first body on the second body.
From this point, the conversion of radiation coefficient in Bemio has no problem.

But I still think the phase angle is not right for multibody cases

Please let me know your thoughts

Best regards,
Tianyuan

salhus, 2023-09-28:
@hachikoi1 
Hi Tianyuan,

Thanks for investigating that. I am looking into this with another developer to identify differences between WAMIT/AQWA. Once done, there will be comprehensive documentation and address this issue (hopefully) for good.

Cheers,
Sal

akeeste, 2023-10-17:
@salhus What is the status on this issue? Do we need a bugfix for our upcoming release?

salhus, 2024-03-20:
@hachikoi1 

Hi Tianyuan,

Apologies for the long radio silence. After extensive code comparisons we may have a resolution.

There are two main differences between AQWA and WAMIT that result in phase differences between the two codes.
## 1. Different sign in the exponential kernel,
We have already established that the sign conventions in AQWA and WAMIT are opposite in the comments above,
## 2. When the first body is not located at the origin of the global frame,
AQWA assigns the global frame to the local CG frame of the first body. This was rather puzzling but after extensive tests it has become evident that the AQWA assigns the origin of the global frame at the CG of the first body being analyzed. 

We would present all our test cases and results summary in the next few weeks.

Cheers,
sal

PR #1253 (2024-09-03): [BUG] Update readAQWA to fix excitation phase and b2b loading issues - main branch
This PR fixes 2 bugs in the readAQWA.m script to resolve #1116 :
1. The excitation phase in Ansys AQWA is given relative to the center of gravity of each body and a positive phase angle represents the excitation lagging behind the wave.

[image]

On the other hand, WEC-Sim follows WAMIT's standard, which is to define the excitation phase relative to the global origin and a positive phase angle meaning the excitation leading the wave.

[image]

The current readAQWA.m script negates the phase correctly to adjust t [... cut]