WEC-Sim/WEC-Sim #1117: [WEC-Sim Applications] optimal damping/gain for PAWEC system

benhopkinson, 2023-09-04:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**

Multiple Run Conditions

**Describe the bug or issue**

I have a PAWEC system and I am trying to determine the optimal damping for the system. In github pages there is reference to an "optimalgainCalc.m" file  and a "mcrBuildGains.m", but I am unsure where to find these/ if they exist. I am wanting to graph the mean power v damping as seen in the screenshots below. Any assistance would be greatly appreciated. 

**To Reproduce**

I cannot find the .m file from the screenshots provided, or any other method to reproduce.

**Expected behavior**
Graphing mean power v damping to find the optimal damping for the system.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: [MacOS Monterey ]
 - [MATLAB/Simulink Release][2022a]
 - [WEC-Sim Release][5.0.1]

jtgrasb, 2023-09-05:
Hi @benhopkinson, thank you for your question. The control examples are found within the [WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/dev) repo. Since they are a new addition to the repo, they are only on the dev branch (which I linked to) instead of master. Within these control examples are the scripts for the optimal gain calculation and mcr setup.

I want to note that the optimal gain calculation calculates the optimal gains for mechanical power in regular waves. A more accurate determination of optimal damping would require a detailed PTO model to calculate electrical power.

benhopkinson, 2023-09-06:
@jtgrasb Thank you for your response, I have attempted to run the example sphere passive and reactive systems found in the control file. When I do run the mcrBuildGains.m file then run wecSimMCR, the results I achieve are a flat plane. I am, unsure what I have done wrong in setting up the test file, if anyone could assist that would be great
[image]

See attached image of the reactive control file resulting graph

[Passive (P) copy.zip](https://github.com/WEC-Sim/WEC-Sim/files/12530728/Passive.P.copy.zip)
I have attached the zip file of the passive file with the geometry files in it.

I am unsure why the results are not changing for the varying damping

hachikoi1, 2023-09-06:
> Screenshots
Dear [benhopkinson](https://github.com/benhopkinson)
I didn't find the location of the screenshots, could you please share the link with me?

Many thanks

Tianyuan

jtgrasb, 2023-09-18:
@benhopkinson Are you still having these issues?

benhopkinson, 2023-09-20:
Hi @jtgrasb 
Yes I am still having these issues
I am using WEC-Sim máster and matlab ver 2022a

jtgrasb, 2023-09-20:
Hi @benhopkinson. This one is confusing since it seems to be working on my end (although I'm on windows), but there are some things to check. 
Is the mcrCases.mat file being created in the directory after you run mcrBuildGains.m? If so, is the mcr variable being loaded into the workspace when you run wecSimMCR? And is mcr.header and mcr.cases correctly defined in the workspace?

jtgrasb, 2023-10-02:
Are there any updates on this issue?

closed 2023-10-16 by jtgrasb