WEC-Sim/WEC-Sim #1118: [Theory or Implementation] Error when switching to CIC wave option

mprevisic1, 2023-09-07:
**Is your question request related to a problem? Please describe.**
I setup a 2-body problem for a hinged barge type WEC. It runs fine when I use the wave class 'regular' as follows:

waves = waveClass('regular');               % Initialize Wave Class and Specify Type                                 
waves.height = 3;                                 % Wave Height [m]
waves.period = 6;                                 % Wave Period [s]

However, when I switch to regular waves with CIC or irregular waves, the simulation fails giving the following error:
[image]

I am not sure why that is and I am not sure where this variable is set. 

**Describe the theory or implementation approach that you have a question**
Please find above.

**Personal project context**
Please find above.

**Describe the type of conclusion or resolution you are looking for**
Trying to resolve the error.

**Additional context**
See files included to reproduce. 
[Debug.zip](https://github.com/WEC-Sim/WEC-Sim/files/12551746/Debug.zip)

jtgrasb, 2023-09-08:
Hi @mprevisic1 thank you for your question. The [convolution integral calculation](https://wec-sim.github.io/WEC-Sim/dev/theory/theory.html#convolution-integral-formulation) (CIC) relies on the impulse response function (IRF), which is calculated by bemio when reading the hydrodata. In order to solve this issue, in bemio.m, the second number (tEnd) in the call to radiationIRF() should be increased to at least equal to the desired cicEndTime. Alternatively, the cicEndTime can be specified in the input file to be smaller if desired.

mprevisic1, 2023-09-08:
Thank you for that clarification. It all works now.

Mirko

closed 2023-09-11 by jtgrasb