WEC-Sim/WEC-Sim #1119: [WEC-Sim Applications] WECCCOMP NMPC question: NMPC block

brittlyd, 2023-09-08:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
WECCCOMP Nonlinear Model Predictive Control

**Describe the bug or issue**
I have a question about inputs to the RTI-NMPC algorithm block for the WECCCOMP NMPC example. When looking at the function block (screenshot 1), there are only three inputs to the function block (xk0, excM_pred, and currenttime), but opening the function, there are many inputs not shown by the block (screenshot 2). I noticed all the variables not shown in the block in screenshot 1 are defined in controller_init.m. My question is how are there more variables in the function of screenshot 2 than what the block shows in screenshot 1, and how does the function have access to variables defined in controller_init.m?

**To Reproduce**
N/A

**Expected behavior**
I thought that all the inputs defined in the function all showed up outside the block.

**Screenshots**
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) 2022a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) v5

**Additional context**
This may be a question about Simulink more than WEC-Sim in particular.

jtgrasb, 2023-09-11:
Hi @brittlyd thank you for your question! Usually, the input variables of a MATLAB Function block are shown as inputs to the block but not always. The variables that are not shown on the block are all constants throughout the simulation and don't need to be input into the block at each timestep. Thus, these variables were set as parameters instead of inputs which allows the block to read the values directly from the MATLAB workspace. Within the MATLAB function block, in the modeling tab, if you open up model explorer, you can see the different inputs, outputs, and parameters.
[image]

nathanmtom, 2023-09-20:
Thanks for responding @jtgrasb, @brittlyd does this answer your issue or do you have further questions based on @jtgrasb response? We'd like to close this issue at the end of this week if there is no further discussion or activity.

closed 2023-09-27 by jtgrasb