WEC-Sim/WEC-Sim #1121: [BUG] Error evaluating 'StopFcn' callback of Frames block (mask)

magm1, 2023-09-14:
Hello, I am trying to simulate an array of WECs and I keep getting this error:

Error using wecSim
Error evaluating 'StopFcn' callback of Frames block (mask) 'array16/Global Reference Frame'.
Callback string is 'run('stopWecSim')'
Caused by:
    Error using wecSim
    Subscripted assignment between dissimilar structures.
    
 How can I solve this? 
Thank you.

nathanmtom, 2023-09-15:
@magm1 Thank you for your question. When creating an issue, please use the GitHub templates:

[image]
[image]
[image]

This information is important for us to accurately assess what an issue is, how it occurs, and most quickly create a solution for you. It also helps us categorize our issues to determine areas of future improvement. Please maintain the template structure (bold headings). ​

Please either modify your initial issue to follow the template or close this issue and re-open a new issue.

nathanmtom, 2023-09-20:
@magm1 Please update your issue to reflect our issues template or we will close this issue at the end of the week.

nathanmtom, 2023-09-27:
Closing because of inactivity

closed 2023-09-27 by nathanmtom