WEC-Sim/WEC-Sim #1122: Rotational PTO Implementation Issue [Theory or Implementation]

mprevisic1, 2023-09-20:
**Is your question request related to a problem? Please describe.**
I implemented a rotational PTO on a two-body system and am trying to add a PTO system with time-domain behavior. I am using the "Rotational PTO Actuation Torque" Simulink block. The system works fine when I use the "Rotational PTO" block, but when I switch in the "Rotational PTO Actuation Torque", things don't seem to work anymore. Here is the Simulink model. 

[image]
When setting the feedback term to 0, we should see PTO forces and power values become zero as well. However, they stay the same no matter what values I use. Here is the Input file section on the PTO. As shown below the stiffness and damping is set to zero. 
[image]
Computing the average power output, I seem to consistently get the same value, no matter what I change as a damping value. 
[image]

**Describe the theory or implementation approach that you have a question**
See above

**Personal project context**
NA

**Describe the type of conclusion or resolution you are looking for**
I am uncertain if I configured this system properly. 

**Additional context**
Please find enclosed the model files. 
[Hinged_Barge2_PTOLimit.zip](https://github.com/WEC-Sim/WEC-Sim/files/12677886/Hinged_Barge2_PTOLimit.zip)

jtgrasb, 2023-09-21:
Hi @mprevisic1,

I just ran your model and I am not able to replicate the issue. The average PTO power output is 0 when I run it. Could you provide more details on how you are running WEC-Sim? Master or dev? run from simulink? etc. I would guess that the PTO damping and stiffness aren't being properly set to 0, but it is hard to tell why this would be the case.

jtgrasb, 2023-10-02:
Hi @mprevisic1, are you still having this issue?

mprevisic1, 2023-10-02:
Sorry, I thought I responded. The issue is resolved. A reboot seems to have
done the trick.

closed 2023-10-02 by jtgrasb