WEC-Sim/WEC-Sim #1124: [WEC-Sim Applications] Wave Direction issue

magm1, 2023-09-24:
Hello,

I am trying to analyse different wave directions for a WEC array, for which I'm running irregular waves for incidence angles of 0º, 22.5º, and 45º. I have the WAMIT .out file already with these three incidence angles, and also the .h5 file. When trying to run WEC-Sim using waves.direction = [0; 22.5; 45] it is giving the following error:

Error using  () 
Query coordinates input arrays must have the same size.

Error in interp2 (line 156)
    Vq = F(Xq,Yq);

Error in bodyClass/irrExcitation (line 570)
                    obj.hydroForce.fExt.re(:,:,ii) = interp2(X, Y, squeeze(re(ii,:,:)), wv, direction);

Error in bodyClass/hydroForcePre (line 302)
                    obj.irrExcitation(w,bemCount,direction,rho,g);

Error in initializeWecSim (line 249)
        body(it).hydroForcePre(waves.omega,waves.direction,simu.cicTime,waves.bem.count,simu.dt,...

Error in run (line 91)
evalin('caller', strcat(script, ';'));

Error in wecSim (line 43)
run('initializeWecSim');

My goal is to analyse these directions separately to compute the Power output of the WEC, how can I solve this issue?

Thank you.

nathanmtom, 2023-09-25:
@magm1 Thank you for your question. When creating an issue, please use the GitHub templates:

[image]
[image]
[image]

This information is important for us to accurately assess what an issue is, how it occurs, and most quickly create a solution for you. It also helps us categorize our issues to determine areas of future improvement. Please maintain the template structure (bold headings). ​

Please either modify your initial issue to follow the template or close this issue and re-open a new issue.

nathanmtom, 2023-10-02:
@magm1 Please update your issue to follow our template described above. We will close this issue at the end of this week if no action is observed.

closed 2023-10-17 by dav-og