WEC-Sim/WEC-Sim #1125: [Theory or Implementation] Non-PTO damping between bodies

ckberinger, 2023-09-25:
**Is your question request related to a problem? Please describe.**
I am trying to implement a viscous damping force that represents losses in the PTO. My PTO is between two floating bodies. I want this viscous damping to be separate from the standard PTO damping that power is calculated from. I am not sure of the best way to implement this as I want the damping to be applied to the relative velocities of the bodies, not any single body velocity.

**Personal project context**
This is for the Laboratory Upgrade Point Absorber. There is a float and a spar with a PTO between them. 

**Describe the type of conclusion or resolution you are looking for**
I think there are multiple ways to achieve this but I wanted advice or to know if someone else has tried this. Could it be done with two PTOs in series and naming one "power capture" and the other "losses"? Or with a PTO and mooring matrix block in series and implementing damping in a "mooring" force matrix?

nathanmtom, 2023-09-27:
@ckberinger Thank you for your question and I'll do my best to answer your question. 

The option that would provide you the best option is to use the "Translation PTO Actuation Force" block in the PTOs library. This allows you to create a subsystem where you can use any damping gains desired to be multiplied against the PTO relative velocity or displacement if you have a spring force. This way you can use one gain for the generator damping force and one gain for the component friction. A good example would be our WECCCOMP applications which demonstrates how you can have multiple signals be used to determine an output force for the PTO to exert on the two connected bodies.

ckberinger, 2023-09-28:
Hi Nathan,

Thank you for your advice. I have successfully set up this system. I have uploaded my Simulink file for others to reference. 
In the wecSimInputFile, I define the constant _ptoDampingLoss = -350; % [N/(m/s)]_
The term needs to be negative to get the correct force to be applied in the PTO as it is a viscous loss. 
[image]
[image]

[LUPAsim.zip](https://github.com/WEC-Sim/WEC-Sim/files/12753771/LUPAsim.zip)

closed 2023-09-28 by ckberinger