WEC-Sim/WEC-Sim #1128: [BUG] WEC-Sim tests failing for MATLAB latest (2023b)

jtgrasb, 2023-09-27:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
The CI test is failing for MATLAB latest on windows-latest.

**WEC-Sim file(s)**
N/A

**To Reproduce**
_Steps to reproduce the behavior:_
1. Look at most recent dev branch test results

**Expected behavior**
Tests pass

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
N/A

**Additional context**
I looked into this issue and found that the last tests to pass were when MATLAB 2023a was the latest release. When MATLAB 2023b became MATLAB latest, the test started failing. It is currently occurring on the dev branch but I expect it will fail on Master as well if the tests were to be run again.

akeeste, 2023-10-17:
It appears that this issue is related to the implementation of R2023b on matlab-actions and not an issue with the WEC-Sim source.  I moved this to the project board. I suggest we close this particular issue and follow up on the matlab-actions organization. I made a [project card ](https://github.com/WEC-Sim/WEC-Sim/projects/58#card-90861704)to follow-up on this at a later date.

closed 2023-10-17 by akeeste

kmruehl, 2023-10-17:
Thanks @akeeste I was thinking about leaving the issue open until it was resolved externally, but let's try the project card. I'll update it to include the action to update WEC-Sim applications as well.

akeeste, 2023-10-17:
That's fair @kmruehl. My motivation to follow-up and eliminate any bug-related issues or PRs before our pending release. I'm open to either method of tracking this