WEC-Sim/WEC-Sim #1131: [Theory or Implementation] Degenerate mass distribution error

magm1, 2023-09-28:
Hello, 

I am trying to simulate an array of one body buoys. I read in the documentation that, in order to join the PTO with a constraint, an insignificantly small mass between the two joints should be included. However, even though I added a mass between them, I keep getting the 'CONSTRAINT (Planar Joint)' has a degenerate mass distribution on its follower side' error. I tried this for only one of this buoys and it worked, but not for the array.
Anyone knows what the problem could be and how to solve it?

Thank you.

salhus, 2023-09-29:
Hi @magm1,

Depends on the overall configuration of the model. If the bodies are connected in a way that forces motion that the planar joint won't allow there maybe issues.
As suggested in the previous raised by you (see https://github.com/WEC-Sim/WEC-Sim/issues/1121). Please follow the template and share the relevant files so that further discussion is possible.

Another, however, more risky approach would be to break the library and 
1. Go down to the level of SimScape's body block and add additional frames: https://www.mathworks.com/help/sm/ug/frames-and-frame-transforms.html
2. Add additional out-ports through all the levels of masks, all the way up, (Note these out-ports are for Simscape signals so would look like diamond-shaped instead of the usual Simulink's elliptical out-ports,
3. Now use the additional frames to connect the other constraints/PTOs.

We would be adding proper documentation for this degenerate-mass issue by the end of the year.

Cheers,
Sal

akeeste, 2023-10-17:
Hi @magm1,

Is there any additional information on your question that you can share? If you can please update your original comment to include one of our Issue templates, our team can better support your issue.

nathanmtom, 2023-10-27:
Closing because of inactivity.

closed 2023-10-27 by nathanmtom