WEC-Sim/WEC-Sim #1138: [Theory or Implementation] Question about re-creating added mass torque for OSWEC

brittlyd, 2023-10-10:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
This question is related to the OSWEC example. All the settings are default, and the input is a regular wave with H = 2.5m, T = 8s.

**Describe the theory or implementation approach that you have a question**
I am trying to re-create the radiation damping torque, added mass torque, and excitation torque using the BEM coefficients based on the equations in the Theory Manual. The equations I'm using are: 

F_{rad,damp} = -B * X_dot, where B is the 6x6 radiation damping matrix and X_dot is the 6xn velocity vector (n is number of time steps)
F_{added mass} = -A * X_dot_dot, where A is the 6x6 added mass matrix and X_dot_dot is the 6xn acceleration vector
F_{exc} = real(H/2 * F_{e,coeff} * exp(i*omega*t)), where H is the wave height, F_{e,coeff} is the excitation force coefficient, omega is the wave frequency, and t is time

Once I calcuated each force/torque in the 6 degrees of freedom, I chose the pitch direction and compared it to the forces saved in the output.bodies(1) structure. I was able to re-create the radiation damping and added mass torques, but not the added mass (see pictures attached).  I'm wondering if there is another step I'm missing in calculating this torque, maybe having something to do with where the torque is calculated? Or what else could be causing the discrepancy?

**Personal project context**
We are interested in extracting hydrodynamics coefficients from experimental data and want to test our algorithms on WEC-Sim data.

**Describe the type of conclusion or resolution you are looking for**
I am hoping to be able to match the added mass torque in the pitch direction more accurately.

**Additional context**
For each plot, the blue lines are torques from the output.bodies structure, and the orange lines are the recreated torques using the above equations and the BEM coefficients. 

[image]
[image]
[image]

nathanmtom, 2023-10-13:
@brittlyd Thank you for your question and I will do my best to respond. 

First, I'd like to ensure you saw our Added Mass implementation guide, https://wec-sim.github.io/WEC-Sim/master/developer/advanced_features.html#added-mass, which explains the additional steps that WEC-Sim completes when implementing added mass. Also, suggest reviewing the implementation under `Rigid Body`->`Hydrodynamic Body` to see the workflow. In addition, the `postProcessWecSim.m` and `bodyClass.m` show how we adjust the added mass force after the simulation is completed.

Second, can you confirm that the A and B are the coefficients at the wave period of 8 s? Also that the simulation case you are running is `regular` and not `regularCIC`?

My expectation is that our transport delay is slightly changing how the added mass is being calculated compared to the equation you have written above. Although unclear why the torque has a second harmonic if the coefficients are static which would indicate the acceleration is non-sinusoidal.

brittlyd, 2023-10-13:
@nathanmtom Thanks for your response! I did not look at the Added Mass implementation guide, so thank you for linking that and I'll look over it. 

A and B are at the frequency closest to the frequency corresponding to an 8 second wave period. And I am using the `regular` setting.

I'm not sure why the added mass torque has a second harmonic either.. The blue line in that plot comes straight from `output.bodies(1).forceAddedMass(:,5)`.  I made another plot (attached) where the blue line is the angular acceleration (`output.bodies(1).acceleration(:,5)`) and the orange line is added mass torque (`output.bodies(1).forceAddedMass(:,5)`) and it is clear that the acceleration does not have the second harmonic but the added mass torque does. Is there something I'm missing as to why that is the case? I ran the case straight from the example settings, so I don't think I changed anything there. 
[image]

nathanmtom, 2023-10-18:
@brittlyd Thank you for the follow-up to my initial comments. 

After confirming your simulation parameters and running the example case, I was able to reproduce your results. Now after review the results I have a few additional comments:

- In response to multiple harmonics in the pitch added mass torque, I realized we need to be cautious that the pitch acceleration is just one component that is multiplied by the added mass matrix to generate the torque. With the surge, heave, and pitch motions combined with coupling terms the torque time signal does not have to match the pitch acceleration profile alone. 
- The mismatch in pitch torque I believe still comes down to how our implementation of the time shift and correcting for the added mass based on the documentation previously shared with you. However, related to your other issue if we are moving the forces to about the hinge, in this example the surge added mass force multiplied by the lever arm from the flap center of gravity to the rotation point is going to dominate the added mass torque in this case (see plot below). If we are able to reconstruct the surge added mass force in this example, this will have a greater influence on the flap dynamics. If we look at the y-axis scale we can see the pitch added torque is around a few percent at most. Therefore, in this case I do not expect that diving deeper into how to improve matching of the pitch added mass torque would improve your system estimation.

Please let me know your thoughts on this issue and if this answer is sufficient to close the issue.

[image]

brittlyd, 2023-10-26:
Thank you for your response! This makes sense and I have no follow-up questions at this time.

closed 2023-10-26 by nathanmtom