WEC-Sim/WEC-Sim #1139: [BUG] MoorDyn Update Loading Issue

yman95, 2023-10-10:
Hello,

Thanks for updating MoorDyn. I see that MoorDyn v2 is live as @RyanDavies19 mentioned. 

It might be a simple fix, but right now, WEC-Sim still wants to load 'lines.dll' instead of 'libmoordyn.dll' in the new library. How do I change which file WEC-Sim reads? Can't seem to find the code using the search function. Thanks.

[image]

Yi

RyanDavies19, 2023-10-10:
@yman95 MoorDyn v2 is now live on the dev branch of WECSim, it has not yet been merged into the master branch. I will defer to others on whether or not to recommend the dev branch, but that is the status of the MoorDyn v2 coupling.

jtgrasb, 2023-10-12:
@RyanDavies19 is correct. We are in the process of a new WEC-Sim release, so MoorDyn v2 will be on the master branch soon. In the mean time, you should use the dev branch to work with MoorDyn v2.

jtgrasb, 2023-10-12:
The updates to the mooring library are in the Properties/Callbacks of WECSim_Lib_Moorings/MoorDyn/MoorDyn/MoorDyn. These updates changed the initFcn, stopFcn, and closeFcn to use the new library name ('libmoordyn') and the new MoorDyn.h header file which serves as a wrapper for MoorDyn. This is detailed in PR #1134. 

These updates are present in the dev branch. Please let me know if the dev branch is not working with MoorDyn v2.

yman95, 2023-10-12:
@jtgrasb it seems that I have run into an environment problem. And people suggest that it's impossible to run a win32 .dll with 64bits MATLAB. Also, it seems to relate to Visual Studio, but I'm not sure if VS and MATLAB is related in such setting.

`---------------Starting MoorDyn-----------
Error using wecSim
Error evaluating 'InitFcn' callback of SubSystem block 'singleBody_3DOF_MoorDyn/MoorDyn/MoorDyn/MoorDyn'.
Callback string is '% Initialize MoorDyn Lib (Windows:dll or OSX:dylib)
disp('---------------Starting MoorDyn-----------')

if libisloaded('libmoordyn')
    calllib('libmoordyn', 'MoorDynClose');
    unloadlibrary libmoordyn;
end

if ismac
    loadlibrary('libmoordyn.dylib','MoorDyn.h');
elseif ispc
    loadlibrary('libmoordyn.dll','MoorDyn.h');
elseif isunix
    loadlibrary('libmoordyn.so','MoorDyn.h');
else
    disp('Cannot run MoorDyn in this platform');
end

calllib('libmoordyn', 'MoorDynInit', mooring.orientation, [0 0 0 0 0 0], mooring.moorDynInputFile);
disp('MoorDyn Initialized. Now time stepping...')
'
Caused by:
    Error using wecSim
    C:\Users\llora\AppData\Local\Temp\tpb0d8db40_73d1_4df8_888d_87aba47be678\libmoordyn_thunk_pcwin64.dll is not a valid Win32 application.

 `
Please advise. Thanks,
Yi

RyanDavies19, 2023-10-12:
@yman95 My suggestion would be to try compiling MoorDyn yourself and then moving the library to the appropriate folder: `WEC-Sim/source/functions/moordDyn/`. The easiest way to do this is using the release executables found [here](https://github.com/FloatingArrayDesign/MoorDyn/releases/tag/v2.2.2). Downloading and executing the `Moordyn-2.2.2-win64.exe` file will compile the library on your machine and install files in the folder where you execute it. There will be three library binary files created in the lib folder, but the two with version numbers are just pointers to `libmoordyn.dll`. The MoorDyn header file can be the same one as what is in the WEC-Sim/MoorDyn/ repo, it can also be found in the include/moordyn/ folder. 

If this doesn't work, then I would suggest trying to compile MoorDyn yourself from the source code following the instructions in the documentation for [compiling as a simple library](https://moordyn.readthedocs.io/en/latest/compiling.html#compile-as-a-simple-library).

If that still doesn't work there is also an option to compile MoorDyn from the source code [using cmake](https://moordyn.readthedocs.io/en/latest/compiling.html#compile-using-cmake).

yman95, 2023-10-16:
@RyanDavies19 Thanks for the detailed instruction. I'm a bit confused in the process still.

Sorry for not having prior experience in compiling files, but I don't see the files you mentioned from using the executables. 
The installed library has three folders:
[image]
In `lib`, there are cmake source files:
[image]
And I can't find libmoordyn.dll anywhere, so I'm unable to move the compiled library to `WEC-Sim/source/functions/moordDyn/`

Tried mixing files in the `WEC-Sim/MoorDyn/` repo with files from `\MoorDyn-2.2\include\moordyn` but it doesn't seem to work. Have I missed something?

RyanDavies19, 2023-10-16:
@yman95 I am shooting a little bit in the dark here because I use Mac and not windows. You could try renaming the file moordyn.dll (in the bin folder) to libmoordyn.dll and then copying that to the appropriate WEC-Sim folders. Let us know if that works.

yman95, 2023-10-19:
@RyanDavies19  Renaming moordyn.dll to `libmoordyn.dll` still lacked `moordyn.h`, so I included the files from `include`, however it gave the same error message of `libmoordyn_thunk_pcwin64.dll is not a valid Win32 application`

Error log:
[image]

Seems like I need to recompile MoorDyn into the 64bit version. Was MoorDyn v2 from repo compiled as the 32 bit version? If that's the case does cmake and git allow the user to compile from source into a 64bit version? Thanks.

RyanDavies19, 2023-10-19:
Ah bummer, thanks for trying that. Yes, you can compile MoorDyn from source quite easily. If you are familiar with cmake then I would suggest using this compile method: [compile using cmake](https://moordyn.readthedocs.io/en/latest/compiling.html#compile-using-cmakel). There is also the option to use an order compiling routine that just builds the dll: [compile as a simple library](https://moordyn.readthedocs.io/en/latest/compiling.html#compile-as-a-simple-library). 

After you follow those steps, move the dll file and the moordyn.h file to the proper place and try again. I am not sure what version the dll was compiled to, @jtgrasb did the file in the WEC-Sim/MoorDyn repo and GitHub build the dll I shared as part of the release.

RyanDavies19, 2023-10-19:
@yman95 correct me if I am wrong but the error you are getting seems like MoorDyn is being built for 64bit systems but your system requires a 32 bit application.

RyanDavies19, 2023-10-19:
@yman95 you might find this issue useful: FloatingArrayDesign/MoorDyn#78

yman95, 2023-10-19:
@RyanDavies19 My system and MATLAB are 64 bit, but seemed like MoorDyn was compiled to be 32bit.

Thanks for the info. I'll try to compile v2 as a simple library.

jtgrasb, 2023-10-19:
@yman95 I compiled the moordyn dll on Windows 64 bit, so the error you are getting is strange to me. Please let me know if compiling v2 works.

yman95, 2023-10-26:
@jtgrasb @RyanDavies19 Hi, after some digging through my system environment, I found that the system was using an older version of c++ compiler from another project, and the 6.3.0 c++ compiler was installed as 32 bits somehow. Fixed those.

However, there seemed to be a pathing issue with my setup for WEC-Sim is unable to find the input file. I checked my system environment variables and couldn't find a path for just `C:\Program`, any idea on where the command might be called?

The full error output:
[image]

Thanks for your help,
Yi

jtgrasb, 2023-10-26:
This seems to be related to the path for mingw-64, which does not support spaces in its path - https://www.mathworks.com/matlabcentral/answers/1994983-loadlibrary-function-error-failed-to-preprocess. You can check the path with `getenv MW_MINGW64_LOC`. You should install the mingw-64 add on in a path without spaces.

yman95, 2023-10-26:
> This seems to be related to the path for mingw-64, which does not support spaces in its path - [...]

fixed, thanks. Although MoorDyn didn't run properly, it's no longer an environment issue, but within WEC-Sim and MoorDyn coding itself.

closed 2023-10-26 by yman95