WEC-Sim/WEC-Sim #1140: [Theory or Implementation] Question about OSWEC example coordinate systems

brittlyd, 2023-10-12:
**Is your question request related to a problem? Please describe.**
My question is related to the OSWEC example, but could apply to any other example or problem. 

**Describe the theory or implementation approach that you have a question**
I was hoping to better understand where the coordinate systems are located in the OSWEC example in WEC-Sim. First, by looking at the position values, it looks like position is measured relative to the free surface, similarly to how we define the location of the PTO, moorings, etc. Are forces and torques also measured about this point? Or are they measured at the center of gravity? And if they are measured at the center of gravity, does that coordinate system rotate with the flap, or do the directions remain aligned with the global coordinate system?

All of these questions are part of my main question of how do I translate the forces and torques to be about the hinge? We have an experimental setup where we can measure force and torques at the hinge of the device, and we want to be able to accurately compare the experimental and WEC-Sim results in the future. 

**Personal project context**
I am trying to translate the force and torque output of the OSWEC example into a coordinate system located at the OSWEC hinge to compare to other simulation and experimental data. 

**Describe the type of conclusion or resolution you are looking for**
I am hoping to get a better understanding of the coordinate systems in WEC-Sim and ideally learn how to translate the force and torque values to be about the OSWEC hinge.

nathanmtom, 2023-10-13:
@brittlyd Thank you for your question and I will do my best to answer.

You are correct that the position values are referenced to the (0,0,0) origin which represents the free surface. 

The forces and torques are calculated about the center of gravity and not the (0,0,0) origin which is common for other hydrodynamic solvers. The default simulation parameters are such that the body coordinate system remains aligned with the global coordinate system. The passive yaw feature provides the option to update certain degrees of freedom based on the rotation about the vertical z-axis. 

For your desire to translate the forces and torques, given you have the 3 DOF of translational forces at the center of gravity and the lever arm between the center of gravity and the hinge you can translate the forces to the hinge. Alternatively, you could use a PTO block to send in a force time history that oscillates the flap in the noWaveCIC condition and you'd be able to calculate the force terms in phase with velocity and acceleration to get the pitch hydrodynamics about the hinge.

brittlyd, 2023-10-13:
@nathanmtom Thank you for your response! I think that makes sense. I've been trying to draw out the system to make sure I am accounting for the forces and torques correctly, and attached the drawing I've been using (let me know if this is confusing). I'm considering the flap as a mass on a weightless rod that is hinged. The forces drawn at the mass are in green and represent what WEC-Sim outputs (from the output.bodies(1).forceTotal matrix). The reaction forces at the hinge are shown in purple, and they are what I assume are reported in the output.ptos.forceConstraint structure for surge and heave? If this is the case, then I wrote out the force balance at the COG which basically just says the sum of the total force at the COG and the constraint force at the hinge is equal to mass times acceleration in the respective direction. However, when I plot that, I get the plot attached. There is good agreement in surge, but an offset in heave and neither match exactly. I'm curious if this setup is correct, or if I'm missing something that's causing the discrepancy. 
[image]
[image]

nathanmtom, 2023-10-18:
@brittlyd Thank you for that detailed description and I agree with the analysis. When I run the simulation given the parameters you have described, I am getting a much closer match, almost exact, compared to your plots (see plots below). This might be down to which variables you are using in the analysis as the pto and the constraint will report different values. Please review and let me know if you have further questions as my reproduction does not identify an issue or discrepancy in theory.

[image]

[image]

brittlyd, 2023-10-26:
Thank you for your response! The results you have make sense to me, and they are what I would expect from mine. I'm not sure why the results I got are different. I will investigate further and follow up if I have another question. Thank you.

nathanmtom, 2023-11-08:
@brittlyd Thank you for the feedback on the issue. I am checking to see if we can close this issue or if there are any immediate questions to keep this issue open?

nathanmtom, 2023-11-13:
@brittlyd I am going to close this issue for now, but please reopen if you have an update.

closed 2023-11-13 by nathanmtom

hachikoi1, 2023-12-18:
Hi @nathanmtom 
Regarding PTO location, the docs https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#modifying-constraints-and-ptos clarifies that the PTO motions and loads are relative to the local coordinate system. But for a multibody system, I am not sure if the origin of local coordinate system refers to the center of gravity of the base or the follow.

-  For example, the COG of platform (base) is (0,0,-14.4), but the COG of the left WEC (follow) is (-51.75,0,0). Thus the `pto(1).location`                 should be (-51.75,0,14.4) or (0,0,0)?

[image]

- When performing free decay tests, apart from `body.centerGravity` and `body(1).initial`,  is it necessary to setup `constraint(1).location`, `constraint(1).initial.displacement`,  `pto(1).location` and `pto(1).initial.displacement` ? 
From https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#decay-tests, both body, constraint, pto and mooring should have correct initial conditions. But in WEC-Sim free decay test case `\WEC-Sim v6.0\applications\Free_Decay\`, I found the setting of constraint, pto and mooring doesn't seem so necessary.

Best regards,

hachikoi1, 2023-12-18:
Hi @nathanmtom 
I post my input file below:
```
% Platform
Control.PlatformCOG                                   =  [0.0 0.0 -14.4];
Control.PlatformPRPInitTranDisplacement               =  [0.0 0.0 0.0];
Control.PlatformInitRotDisplacement                   =  0; % deg
Control.PlatformInitRotAxis                           =  [0 1 0];

% WECLeft
Control.WECLeftCOG                                    =  [-51.75 0.0 0.0];
Control.WECLeftPRPInitTranDisplacement                =  [0.0 0.0 0.0];
Control.WECLeftInitRotDisplacement                    =  0; % deg
Control.WECLeftInitRotAxis                            =  [0 1 0];

% WECLower
Control.WECLowerCOG                                   =  [51.75*cos(pi/3) -51.75*sin(pi/3) 0.0];
Control.WECLowerPRPInitTranDisplacement               =  [0.0 0.0 0.0];
Control.WECLowerInitRotDisplacement                   =  0; % deg
Control.WECLowerInitRotAxis                           =  [0 1 0];

% WECUpper
Control.WECUpperCOG                                   =  [51.75*cos(pi/3) 51.75*sin(pi/3) 0.0];
Control.WECUpperPRPInitTranDisplacement               =  [0.0 0.0 0.0];
Control.WECUpperInitRotDisplacement                   =  0; % deg
Control.WECUpperInitRotAxis                           =  [0 1 0];

constraint(1)                   =  constraintClass('Constraint1');
constraint(1).location          =  Control.PlatformCOG;
constraint(1).initial.displacement =  body(1).initial.displacement;

% WECLeft Translational PTO
pto(1)                          =  ptoClass('PTO1');
pto(1).location                 =  Control.WECLeftCOG-Control.PlatformCOG; % Location in the local coordinate system, see https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#modifying-constraints-and-ptos
pto(1).hardStops                =  struct('upperLimitSpecify',               'on',...               
                                          'upperLimitBound',                 15-(Control.WECHeight-Control.WECDraft), ...                  
                                          'upperLimitStiffness',             1e10, ...                
                                          'upperLimitDamping',               1e10, ...                
                                          'upperLimitTransitionRegionWidth', 1, ...               
                                          'lowerLimitSpecify',               'on',...               
                                          'lowerLimitBound',                 -13+Control.WECDraft, ...                
                                          'lowerLimitStiffness',             1e10, ...              
                                          'lowerLimitDamping',               1e10, ...               
                                          'lowerLimitTransitionRegionWidth', 1);

% WECLower Translational PTO
pto(2)                          =  ptoClass('PTO2');
pto(2).location                 =  Control.WECLowerCOG-Control.PlatformCOG; % Location in the local coordinate system, see https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#modifying-constraints-and-ptos
pto(2).hardStops                =  pto(1).hardStops;

% WECUpper Translational PTO
pto(3)                          =  ptoClass('PTO3');
pto(3).location                 =  Control.WECUpperCOG-Control.PlatformCOG; % Location in the local coordinate system, see https://wec-sim.github.io/WEC-Sim/master/user/advanced_features.html#modifying-constraints-and-ptos
pto(3).hardStops                =  pto(1).hardStops;

for i = 1:length(pto)
    pto(i).initial.displacement =  body(i+1).initial.displacement;
end
```