WEC-Sim/WEC-Sim #1150: Locating the force equations

Khalid-1994-06-17, 2023-10-26:
Hello, again how can I locate the code(s) which uses the force equations to plot the force and displacement graphs?

nathanmtom, 2023-10-27:
@Khalid-1994-06-17 Thank you for your question. We require users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "**Theory or implementation clarification**" template and include more details on your setup and error message, I am happy to help. Please maintain the template structure and the issue type in the title.

nathanmtom, 2023-10-30:
Closing as discussion has moved to issue #1153.

closed 2023-10-30 by nathanmtom