WEC-Sim/WEC-Sim #1151: [Theory or Implementation] Question about PM vs. JS for irregular wave spectrum

brittlyd, 2023-10-26:
**Is your question request related to a problem? Please describe.**
This applies to all examples, but I used the OSWEC example. 

**Describe the theory or implementation approach that you have a question**
When I run the OSWEC example in irregular waves and run one case using "PM" for a Pierson-Moscovitz spectrum and another with "JS" for a JONSWAP with the same parameters otherwise, the resulting spectrum in the waves structure is the same for both cases (see figure attached). In other words, in the wecSimInputFile, under waves.spectrumType I ran two cases: one where I inputted "PM" for Pierson-Moscovitz, and other where I inputted "JS" for JONSWAP. After running both cases I plotted the spectrum defined in the waves structure (waves.spectrum) and found that they are exactly the same, even though I inputted different spectrums in the input file (and confirmed they were inputted in the waves structure under waves.spectrumType). I'm not sure if this is expected by design, or if there is an additional parameters to change the contents of the spectrum between PM and JONSWAP.

**Personal project context**
I'm interested in testing out how well a modeling method works for modeling OSWEC behavior in different types of irregular wave spectrums. 

**Describe the type of conclusion or resolution you are looking for**
I am hoping to understand if the "PM" and "JS" option for waves.spectrumType in the wecSimInputFile should result in the same spectrum in the waves structure, or if there is something I am doing wrong when defining the problem, like not defining a certain constant, or something like that. 
[image]

nathanmtom, 2023-10-26:
@brittlyd Thank you for your question and I will do my best to respond. 

- First can you confirm what the significant wave height and peak period were used in your simulations? 
- Second I'd like to point to our documentation on PM and JS, https://wec-sim.github.io/WEC-Sim/master/theory/theory.html#jonswap-js-spectrum, with focus on the \gamma variable. If this is equal to one then the PM and JS will be equal and the `waves.gamma` is either defined by the user or automatically calculated based on the expression shown on our documentation page. If you did not define `waves.gamma` and the ratio of peak period to the square root of significant wave height is greater than 5, `waves.gamma` is set to 1 and so you the resulting PM and JS spectrums should be the same. 

I believe after getting answers and comments to the points above we can clarify what you are experiencing.

brittlyd, 2023-10-26:
@nathanmtom Thank you for your quick response! I used the following parameters: Hs = 2.5m and Tp = 8s. This means that Tp/sqrt(Hs) = 5.05, so that's why gamma is set to 1 and they are the same. Thank you for clearing this up!

closed 2023-10-26 by nathanmtom