WEC-Sim/WEC-Sim #1152: WEC-Sim Simulink OSWEC

HadeelKhalid, 2023-10-27:
hello, 
I'm facing a problem running the Simulink simulation of the OSWEC example, I face errors that say "variable body does not exist" for all parts in my Simulink file.
@kmruehl 
@nathanmtom 
[image]
[image]

nathanmtom, 2023-10-27:
@HadeelKhalid Thank you for your question. We require users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "Theory or implementation clarification" template and include more details on your setup and error message, I am happy to help. Please maintain the template structure and the issue type in the title.

HadeelKhalid, 2023-10-28:
Unfortunately, I cannot change the template. here are more details about the problem.

the wecSim run works perfectly, however, the problem appears when I try to run the Simulink file.
[image]
[simulation.log](https://github.com/WEC-Sim/WEC-Sim/files/13194656/simulation.log)

nathanmtom, 2023-10-30:
@HadeelKhalid Appreciate the additional description; however, we still cannot respond to your question without modifying the issue to meet one of our templates. You are able to edit your original post by going to the top right corner, click on the three dots `...`, scroll down and click `edit`. See image below for reference.

[image]

nathanmtom, 2023-11-08:
@HadeelKhalid This is a request to please edit your issue to follow our templates as shown in my response above. If we do not see  the edits changed by the end of this week we will close this issue.