WEC-Sim/WEC-Sim #1153: [Theory or Implementation]

Khalid-1994-06-17, 2023-10-27:
**Personal project context**

**Additional context**

Hello, how can I locate the code(s) which uses the force and respnse equations to plot the force and response graphs? I need the equations for then force and response code files. I want to call these functions for the forces and response to Python to help with my project for Reinforcement Learning, my states are the responses for each time period, the action is to control the PTO force to get the energy as the reward. This is because when I called the entire simulation to Python the simulation worked but I am unable to perform reinforcement learning on the model. I realized that all I needed are the codes for the equations of the forces and responses, do I have to look into Simulink or something. Please help me

MShabara, 2023-10-30:
To locate the code for the force and response equations used to plot force and response graphs, you can refer to the WEC-Sim source code. Specifically, WEC-Sim provides built-in methods inside the response class and wave class that can assist you in processing its output. You should look for the following methods: output.plotForces and output.plotResponse. You can find these classes in the (WEC-Sim\source\objects) file. These methods are instrumental in visualizing and analyzing the forces and responses of your wave energy converter (WEC) system. 

Now, regarding your goal of integrating WEC-Sim with your RL project, To my understanding, you use Python to run and stop WEC-Sim each predefined RL time step, this might be time consuming, can can cause difficulty for your RL mode to converge.

To address this, I suggest considering an alternative approach. Instead of opening and closing WEC-Sim at every time step, which can indeed be time-consuming, you could use a more streamlined method. One potential solution is to use a persistent WEC-Sim environment within your Python code.

There are various ways to achieve this, and your choice may depend on your specific project requirements. You might consider creating a Python wrapper around WEC-Sim that allows you to control the simulation and interact with it more efficiently. This wrapper could maintain the WEC-Sim environment in memory, reducing the overhead of repeatedly starting and stopping the simulation.

Furthermore, you could explore the use of reinforcement learning libraries, such as OpenAI Gym, to structure and manage your RL training process. This can help you seamlessly integrate WEC-Sim into your RL environment, making it easier to train your RL agents on the response data from WEC-Sim.

Here's a simplified example Class of what a WEC-Sim wrapper class might look like:
class WECWrapper:
    def __init__(self, config_file):
        # Initialize WEC-Sim with the provided configuration file.
        self.simulator = WEC_Sim(config_file)
    
    def start_simulation(self):
        # Start the WEC-Sim simulation.
        self.simulator.run()

    def step_simulation(self):
        # Advance the simulation by one time step.
        self.simulator.step()

    def stop_simulation(self):
        # Stop the simulation.
        self.simulator.stop()

    def get_response(self):
        # Retrieve response data from the simulation.
        return self.simulator.get_response_data()

    def get_forces(self):
        # Retrieve force data from the simulation.
        return self.simulator.get_force_data()

MShabara, 2023-10-30:
To locate the code for the force and response equations used to plot force and response graphs, you can refer to the WEC-Sim source code. Specifically, WEC-Sim provides built-in methods inside the response class and wave class that can assist you in processing its output. You should look for the following methods: output.plotForces and output.plotResponse. You can find these classes in the (WEC-Sim\source\objects) file. These methods are instrumental in visualizing and analyzing the forces and responses of your wave energy converter (WEC) system.

Now, regarding your goal of integrating WEC-Sim with your RL project, To my understanding, you use Python to run and stop WEC-Sim each predefined RL time step, this might be time consuming, can can cause difficulty for your RL mode to converge. 

To address this, I suggest considering an alternative approach. Instead of opening and closing WEC-Sim at every time step, which can indeed be time-consuming, you could use a more streamlined method. One potential solution is to use a persistent WEC-Sim environment within your Python code.

There are various ways to achieve this, and your choice may depend on your specific project requirements. You might consider creating a Python wrapper around WEC-Sim that allows you to control the simulation and interact with it more efficiently. This wrapper could maintain the WEC-Sim environment in memory, reducing the overhead of repeatedly starting and stopping the simulation.

Furthermore, you could explore the use of reinforcement learning libraries, such as OpenAI Gym, to structure and manage your RL training process. This can help you seamlessly integrate WEC-Sim into your RL environment, making it easier to train your RL agents on the response data from WEC-Sim.

**Here's a simplified example Class of what a WEC-Sim wrapper class might look like:**
class WECWrapper:
        def init(self, config_file):
        # Initialize WEC-Sim with the provided configuration file.
        self.simulator = WEC_Sim(config_file)

def start_simulation(self):
    # Start the WEC-Sim simulation.
    self.simulator.run()

def step_simulation(self):
    # Advance the simulation by one time step.
    self.simulator.step()

def stop_simulation(self):
    # Stop the simulation.
    self.simulator.stop()

def get_response(self):
    # Retrieve response data from the simulation.
    return self.simulator.get_response_data()

def get_forces(self):
    # Retrieve force data from the simulation.
    return self.simulator.get_force_data()

Khalid-1994-06-17, 2023-10-30:
Thanks I'll look into it

Khalid-1994-06-17, 2023-10-31:
Also I want to get the data for the waves, response and forces as a matrix and not a graph, how can I do that?

MShabara, 2023-11-02:
Try to pause the execution at line 220 in the responseClass.m file then run the code, 
[image]

Then check the workspace, you will find a new variable called "obj", obj consists of a set of structures, among these structures two called "bodies" and ""wave".  There you will find all the responses, forces, wave properties you might be looking for.
[image]

I hope this answers your questions.

closed 2023-11-02 by MShabara

Khalid-1994-06-17, 2023-11-03:
Hi I tried to do what I got from you but I got an error:

>> responseClass
Not enough input arguments.

Error in responseClass (line 148)
            obj.wave.type = waveOutput.type;

What should I do my deadline is after 6 weeks

Khalid-1994-06-17, 2023-11-03:
[image][image]

Those are the errors I am getting

MShabara, 2023-11-03:
Sure, here's a modified response that should be easier to understand:

"Running a function or class in Matlab, like 'responseClass.m,' by itself won't work. 

To address your problem, follow these steps:
1. Follow the steps specified in my previous response.
2. Don't try to run 'responseClass.m' directly. Instead, run your model as you normally would using the 'wecSim' command.
3. If you encounter an issue, it will likely pause at a specific location in the class where you can inspect the problem. This is a good way to understand what's going wrong.
_____________________________________
Another easier option, is to run your model first. Once it's done, look in the workspace for an object named 'output'. This object contains various data structures related to the response and waves you're interested in.

To postprocess these variables, you can write your code in the 'userDefinedFunctions.m' file found in the case file.

I understand your deadline is approaching in six weeks. And we're here to help and provide guidance,

I hope this clarifies the process for you. Feel free to ask if you have any questions or need further assistance.

Khalid-1994-06-17, 2023-11-04:
Thanks, I ran the code with the keyboard function and I managed to get the values for anything such as the responses. If there is anything else I have I'll let you know thanks

Khalid-1994-06-17, 2023-11-04:
Also, how could I write the codes in userDefinedFunctions.m for example?

Khalid-1994-06-17, 2023-11-04:
Plus for the wecSimInputFile.m I want to try to create something where I want to input variables such as the start and end time from Python. 

Furthermore I want to have different time periods with different variables for example:

Period 1 (0-100): PTO Stiffness: 500
Period 2 (101-200): PTO Stiffness: 300

And so on

Khalid-1994-06-17, 2023-11-09:
Okay since I already found the values for the output for the responses, waves and time how can I use these values to calculate the work/energy on the WEC.

MShabara, 2023-11-09:
If you're experiencing problems or have questions specifically related to WECSim, feel free to let us know, and we'll do our best to assist you. However, if your inquiries involve learning how to use MATLAB or delve into wave energy basics, I'd recommend posting your questions on the [MATLAB forums](https://www.mathworks.com/matlabcentral/answers/index) for MATLAB-related queries. For academic guidance on calculating energy or power, reaching out to your academic advisor would be a fantastic option.

MShabara, 2023-11-09:
Also, please use the template provided to post future questions.

Khalid-1994-06-17, 2023-11-09:
But that was from the template I used before like maybe two weeks ago

Khalid-1994-06-17, 2023-12-14:
Hello thank you very much for all of the support I got from you guys in helping me with my project 😃