WEC-Sim/WEC-Sim #1154: Issue with simulation

HadeelKhalid, 2023-10-28:
hello, 
I'm working on a project similar to the OSWEC however I'm using a different design of the plate. I'm facing a problem regarding the wave size, when I run the simulation the device is huge compared to the waves, and I tried to change wave class settings in the input file but nothing changed. Here are pictures of the problem and the input file.
@nathanmtom 
Thank you!
[image]
[image]
[image]
[image]

jtgrasb, 2023-10-30:
@HadeelKhalid  Thank you for your question. We require users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "Theory or implementation clarification" template and include more details on your setup and error message, I am happy to help. Please maintain the template structure and the issue type in the title. As for the large size, please make sure to export your stl in units of m instead of mm.

jtgrasb, 2023-11-14:
Closing due to inactivity. Please open a new issue using the template if help is still needed.

closed 2023-11-14 by jtgrasb