WEC-Sim/WEC-Sim #1159: [WEC-Sim Applications]

AndreasAsiikkis, 2023-11-03:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
The case files are attached (including the source code).
[TestCase.zip](https://github.com/WEC-Sim/WEC-Sim/files/13250637/TestCase.zip)
This folder does not include the .h5 file due to its large size. This file can be accessed via this google drive link:
https://drive.google.com/open?id=1yKH7S38ptc2yo7IywJS9jW0MLrbzEr4d&usp=drive_fs

**Describe the bug or issue**
I am utilizing WEC-Sim to study the response of an array of 6 floaters (with 6 DOFs) in a circular configuration. Two configurations were tested so far: a circular arrangement and a row arrangement. The BEM results were obtained separately for each configuration using Capytaine. While the row arrangement simulation runs without issues, the simulation for the circular arrangement crashes.

Notably, an examination of the added mass of bodies over time revealed different behavior based on their positioning along the x-axis. For bodies aligned along the x-axis, the added mass values in the degrees of freedom (DOFs) of heave, surge, and pitch are normal, while they are significantly smaller for other DOFs. Conversely, for bodies not aligned along the x-axis, all six DOFs exhibit very high added mass values, which I suspect may be contributing to the instability observed in the circular configuration simulation.

It is noted that when 3 DOFs are used as a constrained, the simulation does not crash.

I am seeking advice on resolving this issue or, if possible, understanding the cause of this observed behavior. Could this be a bug in WEC-Sim or, if not, might it be a bug originating from Capytaine?

_A big thanks to WEC-Sim developers for making this amazing tool available to the public._

**To Reproduce**
Download the zip folder and extract all the files.
Add the .h5 file from the google drive link to the hydrodata folder
Type wecSim in command window

**Expected behavior**
The expectation was for the simulation to run smoothly and provide the response of the 6 floaters in the circular configuration, similar to the successful simulation observed with the row configuration.

**Screenshots**
Here I attach some of the results but more can be found after running the model.

**_Configuration of the floaters:_**
[image]

**_Response of the floaters over time (floater 1 and 6 are along the x axis):_**
[image]
[image]
[image]

**_Added mass force for heave and sway:_**
[image]
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2023b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v6.0]

**Additional context**
Actions already taken and tested:

1. Enabled b2b
2. Tried using waveClass('regularCIC') instead of waveClass('regular')
3. Tried using simu.solver = 'ode15s' instead of simu.solver = 'ode4' considering that the model might be stiff
4. Tried changing the adjMassFactor.
5. Tried running a finer mesh and finer frequency range in Capytaine.
6. Tried a smaller time-step
7. Tried different wave conditions.
8. Tried using moordyn
9. Tried adding cables
10. The list goes on...

My final goal: To build a model of 18 floaters (inner circle: 6 floaters & outer circle: 12 floaters) which will move in all the 6 DOFs connected with cables to the seabed.

MShabara, 2023-11-29:
Sorry for the late response Andreas, were you able to solve the problem or you still need support with it?

closed 2023-11-29 by MShabara

AndreasAsiikkis, 2023-11-30:
Dear @MShabara,

Unfortunately, I did not manage to solve the problem. Any help would be highly appreciated!