WEC-Sim/WEC-Sim #1162: [WEC-Sim Applications] Adding delays to the wave propagation for control purposes

eliesh17, 2023-11-08:
Hi all, 
I am working on implementing an MPC controller in wecSim. 
the MPC works under the assumption that it can "see" future waves. 
I have a normal hydrobody, connected to a PTO block which is connected to a reference frame. 
Setting the PTO to work with a spring damper works well and is validated against tank data. 
For the MPC to work, I basically need to delay the wave elevation hitting the hydro body. This is sort of simulating the controller with future knowledge. 
My questions are:
What is the best method to introduce this delay? 
Where is the wave elevation timeseries in simulink? 

Thank you, and awesome work on wecSim v6! Super helpful especially with Moordyn v2.

jtgrasb, 2023-11-08:
Thank you for your question at @eliesh17. We have begun requiring users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "**WEC-Sim Applications**" template and include more details on your setup and error message, I am happy to help. Please maintain the template structure and the issue type in the title. When creating an issue, please use the GitHub templates.

This information is important for us to accurately assess what an issue is, how it occurs, and most quickly create a solution for you. It also helps us categorize our issues to determine areas of future improvement. Please maintain the template structure (bold headings). ​

jtgrasb, 2023-12-13:
Closing due to inactivity. Please open a new issue with the template if you still have questions.

closed 2023-12-13 by jtgrasb