WEC-Sim/WEC-Sim #1164: [BUG] MoorDyn v2 reports an error

hachikoi1, 2023-11-10:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
Hello,
I downloaded MoorDyn and placed it in \WEC-Sim v6.0\source\functions\moorDyn\.  But When I run \applications\Mooring\MoorDyn\ case, WEC-Sim reports a bug. It seems function calllib does not find libmoordyn.dll, but I did put the path  \WEC-Sim v6.0\source\functions\moorDyn\ in the MATLAB environment variable. 

Does anyone else encounter this problem?

Best regards,
Tianyuan
[image]

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._

**To Reproduce**
_Steps to reproduce the behavior:_
1. _Go to '...'_
2. _Click on '....'_
3. _Scroll down to '....'_
4. _See error_

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

yman95, 2023-11-13:
Hi, is the repo you downloaded from `https://github.com/WEC-Sim/MoorDyn` ? Make sure it has libmoordyn.dll in the folder.

Mine looked something like this:
[image]

hachikoi1, 2023-11-14:
Hi [yman95](https://github.com/yman95)
Yes, I am sure the libmoordyn.dll is in the folder. I'm not sure if it's just me, there's no way to access the moordyn download page now.

[image]

Best regards,
Tianyuan

jtgrasb, 2023-11-14:
Hi @hachikoi1, thank you for your question. I am not able to replicate this error. Are you running on Windows? And could you translate the error message so I can see exactly what MATLAB is outputting?
--Best,
Jeff

hachikoi1, 2023-11-14:
> Hi @hachikoi1, thank you for your question. I am not able to replicate this error. Are you running on Windows? And could you translate the error message so I can see exactly what MATLAB is outputting? --Best, Jeff

Hi [jtgrasb](https://github.com/jtgrasb) 
Here are the converted error:
```
Simulating the WEC device defined in the SimMechanics model C:\Program Files\WEC-Sim\WEC-Sim v6.0\applications\Mooring\MoorDyn\RM3MoorDyn.slx...   
Elapsed time is 1.815694 seconds.
---------------Starting MoorDyn-----------
Error using wecSim
Error evaluating 'InitFcn' callback of SubSystem block 'RM3MoorDyn/MoorDyn/MoorDyn/MoorDyn'.
Callback string is '% Initialize MoorDyn Lib (Windows:dll or OSX:dylib)
disp('---------------Starting MoorDyn-----------')

if libisloaded('libmoordyn')
    calllib('libmoordyn', 'MoorDynClose');
    unloadlibrary libmoordyn;
end

if ismac
    loadlibrary('libmoordyn.dylib','MoorDyn.h');
elseif ispc
    loadlibrary('libmoordyn.dll','MoorDyn.h');
elseif isunix
    loadlibrary('libmoordyn.so','MoorDyn.h');
else
    disp('Cannot run MoorDyn in this platform');
end

calllib('libmoordyn', 'MoorDynInit', mooring.orientation, [0 0 0 0 0 0], mooring.moorDynInputFile);
disp('MoorDyn Initialized. Now time stepping...')
'
Caused by:
    Error using wecSim
    There was an error loading the library "C:\Program Files\WEC-Sim\WEC-Sim v6.0\source\functions\moorDyn\libmoordyn.dll"
    The specified module could not be found.

        Error using wecSim
        The specified module could not be found.
```
 My Matlab version is R2022a, in windows 11. Could you send me your moordyn code zip, I can't access it right now.

Best regards,
Tianyuan

hachikoi1, 2023-11-14:
Hello Hi [jtgrasb](https://github.com/jtgrasb)

As expected, your shared moorDyn code still does not work. I try to call `loadlibrary('libmoordyn.dll','MoorDyn.h');` or `loadlibrary('C:\Program Files\WEC-Sim\WEC-Sim v6.0\source\functions\moorDyn\libmoordyn.dll','C:\Program Files\WEC-Sim\WEC-Sim v6.0\source\functions\moorDyn\MoorDyn.h');`, but it still reports the same error. 

I successfully run the same case on the other PC. I don't know why...

jtgrasb, 2023-11-14:
The `libmoordyn.dll` is specific to operating system, but it should work if you are working on a Windows 64-bit system. One solution would be to [compile MoorDyn from the source](https://moordyn.readthedocs.io/en/master/compiling.html#compile-as-a-simple-library), which will create a new `libmoordyn.dll` that is specific to your PC. You can then copy the new `libmoordyn.dll` and replace the one in `source\functions\moorDyn`

hachikoi1, 2023-11-14:
Hi [jtgrasb](https://github.com/jtgrasb)
I seem to have found the reason. I update my Matlab to R2023b and it works. As you say, the issue probably has a lot to do with matlab compiler.

Best regards,
Tianyuan

jtgrasb, 2023-11-14:
Interesting, I was not aware that the dll required a certain version of MATLAB. Thank you for bringing this issue to our attention. We will work on making it more compatible with other versions.

Glad you were able to resolve your issue!

closed 2023-11-14 by jtgrasb