WEC-Sim/WEC-Sim #1165: [Theory or Implementation] Gravity issue

ttammine, 2023-11-10:
**Is your question request related to a problem? Please describe.**
The gravity force is not working correctly in the simulation.

**Describe the theory or implementation approach that you have a question**
Simulation Class Gravity. According to the documentation, the default value is 9.81 m/s2. However, the direction is not defined. It is also unclear whether is it considered by default in the simulations or not. Looking at the mask of the global reference frame the gravity at least there is set to zero.

**Personal project context**
I am trying to model a simple pendulum on a rolling hull. Thus, the hull rolling in waves should excite the simple pendulum to swing. However, immediately after the simulation starts the pendulum swings 180 degrees. Thus, pointing to the sky. Which, obviously is physically not correct. I have tried to change the gravity direction by setting simu.gravity = 9.81 and also -9.81 in the input file. But after adding positive g = 9.81 line the simulation does not run through anymore. The simulation ends without explanation right after it has started. This is very strange because adding the line should not change anything as it is same as the default value.
When testing with negative g = 9.81 the whole system started to roll over and over. This either is not physically possible.

I have attached two videos explaining these:

https://github.com/WEC-Sim/WEC-Sim/assets/147230484/8876f4e4-5964-45cb-bb32-330796961919

https://github.com/WEC-Sim/WEC-Sim/assets/147230484/6582c36d-b17d-458f-86ac-b9ee6cf2e8d2

The Simulink model is as:
[image]

The input file is as follows:
%% Simulation Data
simu = simulationClass();               % Initialize Simulation Class
simu.simMechanicsFile = 'Box.slx';      % Specify Simulink Model File
simu.mode = 'normal';                   % Specify Simulation Mode ('normal','accelerator','rapid-accelerator')
simu.explorer = 'on';                   % Turn SimMechanics Explorer (on/off)
simu.startTime = 0;                     % Simulation Start Time [s]
simu.rampTime = 10;                    % Wave Ramp Time [s]
simu.endTime = 200;                     % Simulation End Time [s]
simu.solver = 'ode4';                   % simu.solver = 'ode4' for fixed step & simu.solver = 'ode45' for variable step 
simu.dt = 0.1; 							% Simulation time-step [s]
simu.outputDir = 'output'
simu.gravity = -9.81
 
% Regular Waves  
waves = waveClass('regular');           % Initialize Wave Class and Specify Type                                 
waves.height = 2.5;                     % Wave Height [m]
waves.period = 8;                       % Wave Period [s]
waves.direction = 90; %[deg]

%% Body Data
% Hull
body(1) = bodyClass('hydroData/Hull.h5');      
    % Create the body(1) Variable, Set Location of Hydrodynamic Data File 
    % and Body Number Within this File.   
body(1).geometryFile = 'geometry/Structure.stl';    % Location of Geomtry File
body(1).mass = 'equilibrium';                   
    % Body Mass. The 'equilibrium' Option Sets it to the Displaced Water 
    % Weight.
body(1).inertia = [342094 5077594 5227500];  % Moment of Inertia [kg*m^2]     

% Support
body(2) = bodyClass('');      
body(2).geometryFile = 'geometry/support.stl';    % Location of Geomtry File
body(2).mass = 100;                  % Weight.
body(2).centerGravity = [0 0 6]
body(2).inertia = [1000 1000 1000]
body(2).volume = 1
body(2).nonHydro = 1

% Pendulum strut
body(3) = bodyClass('');      
body(3).geometryFile = 'geometry/Pstrut.stl';    % Location of Geomtry File
body(3).mass = 10;                  % Weight.
body(3).centerGravity = [0 0 7]
body(3).inertia = [100 100 100]
body(3).volume = 1
body(3).nonHydro = 1

%% PTO and Constraint Parameters
% Translation 1
constraint(1) = constraintClass('Constraint1'); % Initialize Constraint Class for Constraint1
constraint(1).location = [0 0 0];               % Constraint Location [m]
constraint(1).orientation.z=[1,0,0];
constraint(1).orientation.y=[0,-1,0];

% Translation 2
constraint(2) = constraintClass('Constraint2'); % Initialize Constraint Class for Constraint1
constraint(2).location = [0 0 0];               % Constraint Location [m]
constraint(2).orientation.z=[0,1,0];
constraint(2).orientation.y=[1,0,0];

% Rotational PTO
pto(1) = ptoClass('PTO1');                      % Initialize ptoClass for PTO1
pto(1).stiffness = 0;                           % PTO Stiffness Coeff [Nm/rad]
pto(1).damping = 0;                         % PTO Damping Coeff [Nsm/rad]
pto(1).location = [0 0 0];                   % PTO Location [m]
pto(1).orientation.z=[0,0,1];
pto(1).orientation.y=[1,0,0];

% Fixed 1
constraint(3) = constraintClass('Constraint3'); % Initialize Constraint Class for Constraint1
constraint(3).location = [0 0 0];               % Constraint Location [m]

% Fixed 2
constraint(4) = constraintClass('Constraint4'); % Initialize Constraint Class for Constraint1
constraint(4).location = [0 0 12];               % Constraint Location [m]

% Rotational PTO
pto(2) = ptoClass('PTO2');                      % Initialize ptoClass for PTO1
pto(2).stiffness = 0;                           % PTO Stiffness Coeff [Nm/rad]
pto(2).damping = 0;                         % PTO Damping Coeff [Nsm/rad]
pto(2).location = [0 0 12];                   % PTO Location [m]
pto(2).orientation.z=[0,0,1];
pto(2).orientation.y=[1,0,0];

**Describe the type of conclusion or resolution you are looking for**
I would like to know how to correctly apply gravity to the model and how to simulate a simple pendulum on a rolling hull.

nathanmtom, 2023-11-15:
@ttammine Thank you for your question in regards to trouble running your simulation. 

In the past, pendulum WECs have been modeled so my first reaction is that gravity is not the culprit here. Rather when reviewing your `wecSimInputFile.m` I notice for your non-hydro bodies you have given them both a volume of 1 m3, which provides 1023 kg of buoyant force (assuming sea water density) which is greater than your 100 kg and 10 kg of material weight. In this case since the buoyancy is calculated still for non-hydro bodies the simulation is accurate and it should rotate vertically. Therefore, in this case if the non-hydro bodies are not in the water the volume should be set to 0 so that buoyancy forces are not calculated. 

Please implement these changes to see if your pendulum no longer rotates and stays vertical. Let us know if you are still having issues after this implementation.

ttammine, 2023-11-18:
Thank you, this solved the problem.

closed 2023-11-18 by ttammine