WEC-Sim/WEC-Sim #1167: [Theory or Implementation] MoorDyn - Error in stopWecSim

yman95, 2023-11-15:
**Is your question request related to a problem? Please describe.**
WEC-Sim identifies the problem as `Line.out` can't be found. In the ../mooring folder, there is `lines_Line1.out` but with no data. I suspect MoorDyn was not successfully run, hence no output data. It might be the input file `lines.txt` or the definition in `wecSimInputFile.m`

**Describe the theory or implementation approach that you have a question**
I've tried to come up with a correctly defined input file after reading through the MoorDyn document, but there is likely things I've missed. 

It seems like MoorDyn successfully initialized and ran through but have problem post processing, but I doubt there were any data or results at all. The project folder is attached at the end.

Error Message:
[image]

**Personal project context**
Testing and trying using MoorDyn for mooring simulation. Geometry is a hexagon with circumdiameter of 10m. For now, I'm trying simulate a simple case where one mooring line attaches to the center bottom of the geometry. 

Thanks for any help offered,
Yi

Using:
Windows OS
MATLAB R2022b
GCC 6.4.0 C compiler
WEC-Sim 6.0
MoorDyn v2
[WEC-Sim_hexagon_10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/13359213/WEC-Sim_hexagon_10m.zip)

jtgrasb, 2023-11-15:
Hi @yman95, thank you for your question, and thank you for including the .zip file - this makes it much easier to debug. 

I believe your issue is because the input file syntax you are using is for MoorDyn V1 instead of [MoorDyn V2](https://moordyn.readthedocs.io/en/master/inputs.html#the-v2-input-file). Please change the syntax of lines.txt to be consistent with v2 and let me know if this solves your issue.

yman95, 2023-11-15:
@jtgrasb  There were two mistakes I fixed, namely an extra 0.5 on the line of LINE TYPES, and OPTIONS instead of SOLVER OPTIONS, but it produces the same issue. Updated lines.txt is attached.

I thought I followed v2's syntax well. Since it's just a simple mooring line, the ROD TYPES, BODIES, RODS, FAILURE, sections were omitted. Is there something I missed using the v2 syntax? Thanks.

[lines.txt](https://github.com/WEC-Sim/WEC-Sim/files/13369548/lines.txt)

yman95, 2023-11-27:
@jtgrasb 
I remade the shell and BEM calculation, the restoring coefficients are now positive. It might be the reason that the shell I assumed was too light and not properly submerged. The shell is now heavier with about 2/3 of it submerged. It seems like the geometry pitches a lot and the restoring force is dominating in pitch. Not sure if they are correct behaviors; can you provide some insights?

Ok, I will try with the mooring matrix block for now. I thought MoorDyn was supposed to be easier since I don't really understand the matrices about the coefficients and parameters regarding the mooring lines.

A questions:
For the constraint in the input file:
`constraint(1) = constraintClass('Constraint1'); % Initialize Constraint Class for Constraint`
`constraint(1).location = [0 0 0];               % Constraint Location [m]`
Is the location supposed to be at the CG, CB, or wherever I specified in the BEM calculation (in this case, the CG)? At this point, I'm assuming CG locations as long as it is below CB.

Shell info in now in the zip file (zip file is larger than 25mb of the github limit)
https://drive.google.com/file/d/11-3CnvRVnePoAOoeIwSetdRKQkKlYn8L/view?usp=sharing

Let me know if the link works. Thanks,
Yi

jtgrasb, 2023-11-28:
Hi @yman95,
The constraint should be at the location about which the body is pitching which would be the center of buoyancy (although the 3dof constraint block handles the location of rotation automatically). Your geometry is pitching a lot. It is difficult to know if this is accurate or not without knowing what you are physically trying to model. Your inertia does seem a bit high though, lowering it could help reduce the pitch. 

As for the mooring matrix, it is easier to work with than MoorDyn, and we have an [example](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Mooring/MooringMatrix) on the WEC-Sim Applications repo. You just need to specify the location, the damping and stiffness in each direction, and the pretension. In the example case, the mooring line is just a stiffness in the surge direction to prevent the WEC from drifting.

yman95, 2023-12-01:
@jtgrasb Following your suggestion, I started to investigate the MOI of the geometry. I was taking the MOI with mass calculation from FreeCAD plugin FCInfo. To validate the their calculation, I loaded the RM3 example to see if FCinfo's MOI calculations agree with WEC-Sim's. It seems like it's in fact not accurate. I'll move to SolidWorks for MOI calculation. 

I did a manual approximation of the MOI, and I think the result looks (intuitively) better? MOI is now (3969424.06 3969424.06 4536484.64)
[WEC-Sim_hexagon_10m.zip](https://github.com/WEC-Sim/WEC-Sim/files/13531555/WEC-Sim_hexagon_10m.zip)

I'll try with SolidWorks and mooring matrix and report back,
Yi

jtgrasb, 2023-12-13:
Hi @yman95, is your issue resolved?

jtgrasb, 2023-12-13:
Hi @yman95, there are a few issues with the lines.txt file in the attached .zip. 
1. The BODIES section is missing and needs to be added to represent the Hexagon body in MoorDyn. 
2. The POINTS IDs should be in consecutive, ascending order. Your 2nd point ID should be 2 instead of 11. This changes the LINES section as well.
3. The Attachment for the 2nd point should be Body1 instead of vessel. According to https://moordyn.readthedocs.io/en/master/structure.html#id2, the vessel attachment refers to using a driver program to control the point. The other point should also be called `Fixed` with a capital F.
4. Please add `2        writeLog     - Write a log file` and `0        WriteUnits   - 0: do not write the units header on the output files` to the OPTIONS section, which will output the MoorDyn outputs to a log file (not including the units so WEC-Sim can read it correctly), and you can see more details on what the issues may be.
5. Lastly, on the WEC-Sim side, the mooring(1).initial.displacement should be set equal to the initial displacement of your body (0 in this case).

yman95, 2023-12-18:
@jtgrasb Thanks for pointing out the mistakes. I think MoorDyn is working now (at least finished without crashing), but I need to be sure I'm plugging in the correct numbers. If you could take a look and see if numbers are reasonable, it'd be much appreciated.

Answering your points and a few questions:
1.  I thought BODIES section mean any additional bodies other than main geometry. In this section, without empirical or experimental data, How does one obtain the drag coefficient and added mass coefficient of the body? Some literature suggested that CdA is 0.6-1.2 and Ca is 1.0-2.0, so I assumed 1.0 and 1.5 for them respectively.
2. The other thread (#1109) pointed out I didn't need to number 2-10 of the nodes, but it's probably different for v2. Added now.
3. Does WEC-Sim report back on the wave celerity? I checked that the wave number is in MATLAB and I can calculate the wave length from that (k=2pi/L), but can't seem to find how fast is the wave propagating given only height and period are defined.

File:
https://drive.google.com/file/d/1bBPjotUH3Rm2niIVKwvzipzzDXZGvCBq/view?usp=drive_link

jtgrasb, 2023-12-19:
Hi @yman95, glad to hear you were able to get it functioning. 
1. The BODIES section should include the bodies that mooring lines are being attached to. MoorDyn is also a standalone software, so it allows you to define the dynamics of different bodies in terms of their drag coefficient and added mass. Since WEC-Sim is handling the body-wave dynamics (MoorDyn is only being used to define the mooring line dynamics), these quantities should be set to 0.
2. That is correct, you do not need to define nodes 2-10, but those are different than the points (which are for specifying attachment locations). Thus there are only two points and they should be number 1 and 2. The node locations will be inferred by MoorDyn.
3. WEC-Sim does not report the wave celerity, but it seems the [celerity can be calculated](https://www.usna.edu/NAOE/_files/documents/Courses/EN455/AY20_Notes/EN455CourseNotesAY20_Chapter3.pdf) based on gravity, wave number, and water depth.

yman95, 2024-01-02:
Happy new year @jtgrasb, hope all is well.
Both point 1 and 2 are fixed. Thanks for the course notes as well.

Final question about the body property before closing this issue:
I had confusion about the coordinate convention between WAMIT and NEMOH. I understand that WAMIT uses geometric center for origin, while NEMOH uses water surface as origin. When I specify the CG of the body to be (0,0,-4), as also specified in NEMOH's Mesh.cal, I meant it to be -4 below the water surface. Because WEC-Sim want the origin to be at the geometric center, I wonder if this convention is kept the same in WEC-Sim's calculation? i.e., CG in .h5 file and lines.txt for MoorDyn, etc. 

It got complicated because the .stl file I put in WEC-Sim's folder is translated by (0, 0, 2.3), otherwise the whole hexagon is submerged. It does appear that the CG position is kept the same, the geometry position is just visual.

In other words, I want to make sure the (0,0,-4) in WEC-Sim files also points to (0, 0, -4) below water line
[image]

Translated .stl file (currently in use, made specifically for WEC-Sim input and visual):
[image]
Original .stl file (used in BEM calculation):
[image]
Submerged part for NEMOH's calculation (black box is CG, blue box is CB):
[image]
[image]

Let me know if you need further clarification,
Yi

jtgrasb, 2024-01-03:
Hi @yman95 , WEC-Sim's BEMIO process has a separate function for loading in WAMIT and NEMOH results. This means the difference in origin is accounted for automatically during the BEMIO process. 

WEC-Sim uses the free surface as the origin, so when the body cg is at [0, 0, -4], this is 4 m below the free surface. As for the geometry files, these are loaded in with their origin being placed at the center of gravity. Thus, the origin of any geometry file(s) should be at the body’s center of gravity.

closed 2024-01-03 by yman95