WEC-Sim/WEC-Sim #1169: [Theory or Implementation]

cc200528, 2023-11-19:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/master/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/master/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
_Hi, I have read your WEC-Sim Documentation, if I want to use MOST to generate displacement time history, velocity time history, acceleration time history, etc., for some points on the top of the floating offshore wind turbines, what should I do._

**Describe the theory or implementation approach that you have a question**
_N/A_

**Personal project context**
_I am trying to simulate the motion of a floating offshore wind turbines using Wec-sim's MOST function and obtain the motion time history of certain points on the top of the fan_

**Describe the type of conclusion or resolution you are looking for**
_[Ex.]I am still unclear on how an displacement time history of the top of the floating offshore wind turbines is generated within WEC-Sim and would appreciate being pointed to where in the documentation and code where I can find additional information._

**Additional context**
_N/A._

jtgrasb, 2023-11-20:
Hi @cc200528 thank you for using a template. I would recommend exploring the Wind turbine block more in Simulink. Within the Wind turbine subsystem, there is a SignalOutput block that contains all the existing outputs of the turbine (tower load, blade pitch, etc.). 
[image]
A good example to show where these outputs are coming from is the nacelle IMU block. This uses a transform block to detect the relative motion between the nacelle and the world frame and outputs the velocity and acceleration. A similar block could be added to the tower, hub, etc. to output the response at various points in the system. If you need this for specific points besides the center of different bodies, a rigid transform can be used to select a location relative to a body.
[image]
[image]

jtgrasb, 2023-12-13:
Closing due to inactivity. Let me know if there are any further questions.

closed 2023-12-13 by jtgrasb