WEC-Sim/WEC-Sim #1171: [BUG] Moordyn Issue with V2 Matlab 2023b

timmaflan, 2023-11-22:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug**
_A clear and concise description of what the perceived bug is._
I am experiencing a similar to issue #1164. When I run wecSim v6 with matlab 2023b I get the following error:

---------------Starting MoorDyn-----------
Error using wecSim
Error evaluating 'InitFcn' callback of SubSystem block 'RM3MoorDyn/MoorDyn/MoorDyn/MoorDyn'.
Callback string is '% Initialize MoorDyn Lib (Windows:dll or OSX:dylib)
disp('---------------Starting MoorDyn-----------')

if libisloaded('libmoordyn')
    calllib('libmoordyn', 'MoorDynClose');
    unloadlibrary libmoordyn;
end

if ismac
    loadlibrary('libmoordyn.dylib','MoorDyn.h');
elseif ispc
    loadlibrary('libmoordyn.dll','MoorDyn.h');
elseif isunix
    loadlibrary('libmoordyn.so','MoorDyn.h');
else
    disp('Cannot run MoorDyn in this platform');
end

calllib('libmoordyn', 'MoorDynInit', mooring.orientation, [0 0 0 0 0 0], mooring.moorDynInputFile);
disp('MoorDyn Initialized. Now time stepping...')
'
Caused by:
    Error using wecSim
    There was an error loading the library
    "...\Documents\WEC-Sim\source\functions\moorDyn\libmoordyn.dll"
    The specified module could not be found.

        Error using wecSim
        The specified module could not be found.

**WEC-Sim file(s)**
_If known, please identify the the WEC-Sim files causing the bug or error:_
_Please upload relevant case files if possible to speed debugging._
It seems for some reason, that matlab is unable to find libmoordyn.dll, or that there is an error with libmoordyn.dll. 

**To Reproduce**
_Steps to reproduce the behavior:_
1. place moordyn in the correct folder & ensure it is in the path. (Documents\WEC-Sim\source\functions\moorDyn)
[image]
2. set up the moodyn example in wecSim applications so that it has a geometry and hydro folder from wecSim's RM3 example. 
[image]
3. ensure that all necessary files are in the path and try to run wecSim. 

_State method of running WEC-Sim (command line, from simulink with input file, from simulink with custom parameters)_
I have been running wecSim through the command line, by typing wecSim

**Expected behavior**
_A clear and concise description of what you expected to happen._
My expected behavior is that wecSim will run (it does) and that moordyn will be compiled (it does not). Instead it throws an error related.

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 10
 - MATLAB/Simulink Release: 2023b
 - WEC-Sim Release: v6 with moordyn v2

**Additional context**
_Add any other context about the problem here._
My collegues have tried to compile moordyn from the source as suggested here: https://moordyn.readthedocs.io/en/master/compiling.html#compile-as-a-simple-library

jtgrasb, 2023-11-28:
Hi @timmaflan, thank you for your question. I am not sure why there are issues with MoorDyn in different versions of MATLAB. I am working to resolve this issue. In the meantime, I would recommend compiling from the moordyn source as if possible. This would ensure the .dll file is set up for your system.

timmaflan, 2023-11-28:
Thanks! One of my coworkers had the same problem and did try to compile from the source. I will also give that a try today. Thanks, Timma

jtgrasb, 2023-11-29:
Hi @timmaflan, just to be sure, you are running a 64-bit version of MATLAB, correct?

timmaflan, 2023-11-29:
Yes I am using the 64-bit version.

jtgrasb, 2023-11-30:
@timmaflan I updated my MATLAB to R2023b and the loadlibrary function is working fine with libmoordyn.dll. In order to debug this problem, can you type `mex -setup` in the command window and paste the output here? This will let us see what your default c compiler is.

timmaflan, 2023-11-30:
Renamed options file 'C:\Users\AppData\Roaming\MathWorks\MATLAB\R2023b\mex_C_win64.xml' to 'C:\Users\AppData\Roaming\MathWorks\MATLAB\R2023b\mex_C_win64_backup.xml'.
MEX configured to use 'Microsoft Visual C++ 2022 (C)' for C language compilation.

To choose a different C compiler, select one from the following:
MinGW64 Compiler (C)  mex -setup:'C:\Program Files\MATLAB\R2023b\bin\win64\mexopts\mingw64.xml' C
Microsoft Visual C++ 2022 (C)  mex -setup:C:\Users\AppData\Roaming\MathWorks\MATLAB\R2023b\mex_C_win64.xml C

To choose a different language, select one from the following:
 mex -setup C++ 
 mex -setup FORTRAN
>>

jtgrasb, 2023-11-30:
Thanks for checking. I would recommend trying the MinGW64 Compiler as well (although I expect it to give the same result). 

I am still unable to replicate your error. I will check with the rest of the team to see if anyone else encounters similar errors.

timmaflan, 2023-12-01:
@jtgrasb just to clarify, as of yet, I haven't compiled moordyn from the source, but am still using the one that is simply downloaded through github. Are you recommending I try using the MinGW64 compiler for moordyn from the source - as per the instructions? Or using the compiler for some other use? My coworker used the instructions provided online to compile moordyn (MinGW64) but it provided the same results. I'll update you if we see any changes.

jtgrasb, 2023-12-01:
@timmaflan Sorry for the confusion. I am suggesting to click on MinGW64 Compiler (C) after you enter `mex -setup` in the MATLAB command window. This will change MATLAB's default C compiler to MinGW64 instead of Microsoft Visual C++ 2022. Like I said, I don't expect this to fix your error, but it is good to check. 

As for compiling moordyn from the source, that is strange that your coworker got the error from loadlibrary even when compiled on the same machine. This makes me think it may be an issue on the MATLAB side. We'll keep looking into this.

hachikoi1, 2023-12-05:
Hi @jtgrasb
I start a new thread https://github.com/WEC-Sim/WEC-Sim/issues/1178, regarding the input file of MoorDyn V2. I would appreciate your help. 

Best regards,

Tianyuan

jtgrasb, 2023-12-05:
Hi @hachikoi1, thank you for pointing this out. The lines.txt file is being updated in [PR #54](https://github.com/WEC-Sim/WEC-Sim_Applications/pull/54). It should be consistent with MoorDyn v2's convention now. I see you created a new issue and I will review soon. 

Hi @timmaflan, thank you for fixing this. I updated the wecSim.m script in PR #1177. And yes, the WEC-Sim/MoorDyn repository is specifically for WEC-Sim, but those instructions are for the official MoorDyn repository.

I am glad your issue is resolved. Were you able to get the library to load using the dll in the [WEC-Sim MoorDyn repo](https://github.com/WEC-Sim/MoorDyn) or just the one you compiled yourself?

hachikoi1, 2023-12-05:
Hi @jtgrasb
Thanks for your work. Please don't forget to update \applications\Paraview_Visualization\RM3_MoorDyn_Viz\Mooring\lines.txt for the sake of uniformity

Best regards,
Tianyuan

timmaflan, 2023-12-05:
Got it! Thanks for the clarification on the REPO, perhaps that clarification could be made in the "Read the docs" file, as I was unfamiliar with the other Repo. 
I've only tried to load the Moordyn repo I compiled myself.

jtgrasb, 2023-12-06:
@timmaflan Thanks for clarifying. This has been communicated to the MoorDyn team. 

Could you try the individual steps with the dll file from the [WEC-Sim MoorDyn Repo](https://github.com/WEC-Sim/MoorDyn) again? I would like to know if closing and unloading the library successfully resolves your original issue. 

calllib('libmoordyn', 'MoorDynClose');
loadlibrary('libmoordyn.dll','MoorDyn.h');
unloadlibrary libmoordyn;

jtgrasb, 2023-12-20:
Closing due to inactivity

closed 2023-12-20 by jtgrasb