WEC-Sim/WEC-Sim #1172: Wec sim application run problem

chakramtalla, 2023-11-24:
Index exceeds the number of array elements. Index must not exceed 0.

Error in Read_AQWA (line 100)
            hydro(F).cg(:,i) = tmp(2:4); % Center of gravity

Error in bemio (line 4)
hydro = Read_AQWA(hydro,'ANALYSIS.AH1','ANALYSIS.LIS');

i am getting an error like this  any one know about this

nathanmtom, 2023-11-27:
Thank you for your question at @chakramtalla . We have begun requiring users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "**Bug Report**" template and include more details on your setup and error message, we are happy to help. Please maintain the template structure and the issue type in the title. When creating an issue, please use the GitHub templates.

This information is important for us to accurately assess what an issue is, how it occurs, and most quickly create a solution for you. It also helps us categorize our issues to determine areas of future improvement. Please maintain the template structure (bold headings). ​

nathanmtom, 2023-12-06:
@chakramtalla Please update the issue template by the end of this week or we will close this issue because of inactivity.

chakramtalla, 2023-12-06:
[image]
please check this

nathanmtom, 2023-12-11:
@chakramtalla We need more information on the files you are using in order to investigate this issue. This is why we require issue submitters to follow the issues Template which provides the information required to diagnosis if this is an issue or user error. Since the issue has not be updated to follow one of our issues templates I am closing this issue.

closed 2023-12-11 by nathanmtom