WEC-Sim/WEC-Sim #1173: Assistance Required to Replicate Power Output Plot from OSTI Document in RM3 WEC-Sim Example

sherlocksilver, 2023-11-26:
Hello WEC-Sim Developers and Users,

I am new to WEC-Sim and I am learning how to analyze and interpret the simulation results. I have been working through the RM3 example and my goal is to create a power output plot similar to the one found on page 39 of this osti.gov document (there are 3 figures): [https://www.osti.gov/servlets/purl/1335545](https://www.osti.gov/servlets/purl/1335545).

The plot I am interested in replicates the power output in kW, showing both instantaneous power and time-averaged power over a period. I came across [Issue #879](https://github.com/WEC-Sim/WEC-Sim/issues/879) on the WEC-Sim GitHub, which discusses power output analysis. Despite reading through [Issue #879](https://github.com/WEC-Sim/WEC-Sim/issues/879) on the WEC-Sim GitHub, I am still unclear on how to proceed with generating a similar plot with the RM3 case results. 

Could anyone provide a step-by-step guide or advice on how to extract the necessary data from the WEC-Sim output and plot it accordingly? Here are some specific questions I have:

1. What are the correct steps to process the `pto.powerInternalMechanics` data for the RM3 case to plot power output in kW?
2. How can I calculate and plot the time-averaged power alongside the instantaneous power?
3. Are there any recommended MATLAB functions or scripts that I should use to achieve this?

Here is the MATLAB code snippet I've used for plotting, but it doesn't produce the expected positive power values:

```
ptoPower = output.ptos.powerInternalMechanics;  % Assuming this is the correct field
ptoPower_kW = sum(ptoPower, 2) / 1000;  % Convert power from W to kW
plot(output.ptos.time, ptoPower_kW);
xlabel('Time (s)');
ylabel('Power (kW)');
title('PTO Power Output over Time');
```

I am looking to generate a plot that could help me visualize the performance of the wave energy converter similar to the document referenced. I would greatly appreciate any guidance or examples of MATLAB code that could help me.

Thank you for your support!

nathanmtom, 2023-11-29:
Thank you for your question at @sherlocksilver. We have begun requiring users to open GitHub issues using an appropriate template. This is helpful for the development team to classify issues, and address questions more thoroughly and efficiently. We hope this will be beneficial for developers, users and the WEC-Sim community in general.

If you can please reopen or edit this issue to use the "WEC-Sim applications specific issue or question" template and include more details on your setup and error message, we are happy to help. Please maintain the template structure and the issue type in the title. When creating an issue, please use the GitHub templates.

This information is important for us to accurately assess what an issue is, how it occurs, and most quickly create a solution for you. It also helps us categorize our issues to determine areas of future improvement. Please maintain the template structure (bold headings). ​

sherlocksilver, 2023-11-29:
@nathanmtom , Thank you for your reply. I have created a new issue https://github.com/WEC-Sim/WEC-Sim/issues/1175.