WEC-Sim/WEC-Sim #1174: Unable to add wec-Sim path to Matlab [BUG]

BinoBK, 2023-11-27:
**Describe the bug**
I am unable to add the wecSim path to Matlab even after following the instructions as per documentation.

**WEC-Sim file(s)**
_addWecSimSource.m

**To Reproduce**
_Steps to reproduce the behavior:_
1. Add WEC-Sim to the MATLAB Path as per Option 1 or Option 2 of Documentation. 
2. Verify the path by typing "path" in command window. 

**Expected behavior**
_I am trying to add the source directory 'C:\WaveWatts\3.0 WEC-Sim v6.0\source'. 
The source directory, and its subfolders should be listed in Matlab path. 

**Screenshots**
1. Screenshot of 'Set path' in Matlab where intended 'source' directory is not getting populated.
[image]

2. Screenshot of results after typing "path" in command window where intended 'source' directory is not getting populated.
[image]

3. Screenshot of Simulink window where wecSim Library is not getting populated. 
[image]

**Workstation Information:**
 - OS: Windows 11
 - MATLAB/Simulink Release 2023b
 - WEC-Sim Release v6.0

**Additional context**
_NA

nathanmtom, 2023-11-27:
Hello @BinoBK thank you for your question about the installation process and getting the path directory set. 

Reviewing your issue description, I believe you are not following the guidance properly. 

For Option 1 you need to put the `startup.m` in the MATLAB startup folder which should be your...\bino2\onedrive\Documents\MATLAB. The `startup.m` will not be set-up if kept in the WEC-Sim source directory. 

For Option 2, you do not edit `addWECSimSource.m` and need to navigate to the WEC-Sim Source Directory within MATLAB and run the function. The easiest way to do this is close MATLAB and then double click on `addWECSimSource.m` which will open MATLAB pointing to the folder where the function is located. 

Please try to implement these changes to your process to confirm you are able to add the path correctly.

BinoBK, 2023-11-27:
Hello @nathanmtom, Thank you for the reply. I already tried both the options but in vain. 

For Option1, I pasted startup.m in the MATLAB startup folder. Screenshot is below: 
[image]

For Option2, I executed the file addWECSimSource.m. Screenshot is here:
[image]

 I am still not able to add wecSim to MATLAB path. Kindly advice and correct me if something I have done is wrong.

nathanmtom, 2023-11-29:
@BinoBK Thank you for following up after my recommendation. 

It is hard to see from your screenshots as the Set Path overlaps with your MATLAB interface. I am going to focus on Option 2 as this should not require ensuring that the startup.m is in the right folder. 

I want to make sure that the following steps are being followed: 

1. The `addWecSimSource.m` is unmodified and is sitting in the folder above source (see red arrow in image below)
2. You run `addWECSimSource.m` with MATLAB pointing to the folder that holds all of the WECSim files (see green circle below).
3. Type `path` in the command line and see the WECSim added to path.

[image]

Based on your images below it looks like you moved `addWECSimSource.m` into the source folder which is not the appropriate installation procedure. 

Please confirm or attempt these steps and report back on your progress.

BinoBK, 2023-11-29:
@nathanmtom The issue has been resolved. I had incorrectly placed addWecSimSource.m file inside 'source' folder. 

Thank you.

closed 2023-11-29 by BinoBK