WEC-Sim/WEC-Sim #1175: [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) Assistance Required to Replicate Power Output Plot from OSTI Document in RM3 WEC-Sim Example

sherlocksilver, 2023-11-29:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Describe the bug or issue**
I am new to WEC-Sim and I am learning how to analyze and interpret the simulation results. I have been working through the RM3 example, and my goal is to create a power output plot similar to the one found on page 39 of this [osti.gov document](https://www.osti.gov/servlets/purl/1335545) (Screenshot attached).

**To Reproduce**
Steps to reproduce the behavior:

-Go to the RM3 example in WEC-Sim.
-Run the simulation.
-Attempt to create a power output plot as described in the document.

**Expected behavior**
I expect to generate a power output plot in kW, showing both instantaneous power and time-averaged power over a period, similar to the plot in the referenced document.

**Screenshots**
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2021b]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v6.0]

**Additional context**
I came across Issue #879 on the WEC-Sim GitHub, which discusses power output analysis. Despite reading through Issue #879 on the WEC-Sim GitHub, I am still unclear on how to proceed with generating a similar plot with the RM3 case results.

Could anyone provide a step-by-step guide or advice on how to extract the necessary data from the WEC-Sim output and plot it accordingly? Here are some specific questions I have:

1. What are the correct steps to process the `pto.powerInternalMechanics` data for the RM3 case to plot power output in kW?
2. How can I calculate and plot the time-averaged power alongside the instantaneous power?
3. Are there any recommended MATLAB functions or scripts that I should use to achieve this?
4. Additionally, how can I plot the three specific types of figures shown on page 39 of the OSTI document (I have also attached)?

Here is the MATLAB code snippet I've used for plotting, but it doesn't produce the expected positive power values:

```
ptoPower = output.ptos.powerInternalMechanics;  
ptoPower_kW = sum(ptoPower, 2) / 1000;  % Convert power from W to kW
plot(output.ptos.time, ptoPower_kW);
xlabel('Time (s)');
ylabel('Power (kW)');
title('PTO Power Output over Time');

```
I am looking to generate a plot that could help me visualize the performance of the wave energy converter similar to the document referenced. I would greatly appreciate any guidance or examples of MATLAB code that could help me.

Thank you for your support!

sherlocksilver, 2023-12-01:
@nathanmtom, I am eagerly waiting for the answer. 

Regards.

nathanmtom, 2023-12-04:
@sherlocksilver Thank you for your questions about using WEC-Sim and I will do my best to answer them.

1. The output found in `output.ptos.powerInternalMechanics` will only have values in the 3rd column as this is a translational PTO. The output is the instantaneous power from the PTO and plotting directly would correspond to the red line in the WEC-Sim power output; however the exact time history may not match depending on the sea state run. 
2. The time-averaged power is just the mean over the time history so this value is easy to calculate and then plotted as a constant as shown in the instantaneous power plot. 
3. Please see our WEC-Sim Applications Case on Multiple Condition Runs (MCR) **[here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs)**. There will be several functions and scripts included to assist.
4. The time history values can be found in the `output` variable when completing your run and will be straightforward to plot against time. The power matrix requires running multiple sea states and an example on how to implement this is found in the MCR applications case, **[here](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs)**.

Looking at the snippet of code, if you want the time history you do not sum the power to get energy. If you want to plot the time history of power in kWs the following should be used.

`plot(output.ptos.time, output.ptos.powerInternalMechanics(:,3)/1000);`
`xlabel('Time (s)');`
`ylabel('Power (kW)');`
`title('PTO Power Output over Time');`

nathanmtom, 2023-12-20:
@sherlocksilver Please let us know if my response above answered your questions about how to use our existing RM3 model to complete the work you have described. If we do not hear from you by the end of the week we will close this issue because of inactivity.

sherlocksilver, 2023-12-21:
@nathanmtom, Thank you for the information. I have no further questions at the moment. I appreciate your assistance.

closed 2023-12-21 by nathanmtom