WEC-Sim/WEC-Sim #1179: Simulating cylinder with 6DOF constraint returns 'Derivative not finite' error but 3DOF constraint does not [BUG]

ajay-menon-iitkgp, 2023-12-06:
**Describe the bug**
Hi there. I am simulating a floating cylinder (horizontally inclined) in WEC-Sim and would like to calculate its 6DOF motions. I used NEMOH to generate the hydroData and then ran the `bemio` command on it. Since I'm merely interested in analysing the body's dynamics, I first used a 3DOF constraint when setting up my Simulink file. When running the `wecSim` command, the simulation runs perfectly with no warnings or errors. The range of motions are also reasonable, and nothing stands out.

However, when I use a 6DOF constraint, the simulation terminates prematurely with the following error message displayed:

> Error using wecSim
>
> Derivative of state 'cylinder.Floating_6DOF.CONSTRAINT_Bushing_Joint.Px.v' in block 'cylinder/Floating (6DOF)/CONSTRAINT (Bushing Joint)' at time 20.6 is not finite. The simulation will be stopped. There may be a singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances)

I tried this for all the default regular & irregular wave classes in `wecSimInputFile.m`, reduced the step size all the way down to 0.001, changed the simu duration between 200-500 seconds, and changed the ramp time between 20-150 seconds. But the error persists in all these cases!

While not useful to the work I'm doing, it's worth noting that the 'noWave' case runs perfectly without an error - and I assume that's because there are "no waves". What's interesting is that when the simulation terminates with the error, the cylinder behaves absurdly right at the end - spinning uncontrollably and eventually disappearing. I can't add a video of this owing to the file size, but if you run my files, you should be able to observe it in the last few seconds of the 6DOF case.

**WEC-Sim file(s)**
Please find attached my $CASE directory files. There are 2 folders - 3DOF and 6DOF simulation. My raw NEMOH output files are also included.
[WEC-Sim Issue - 3DOF, 6DOF Files.zip](https://github.com/WEC-Sim/WEC-Sim/files/13575479/WEC-Sim.Issue.-.3DOF.6DOF.Files.zip)

**To Reproduce**
_Steps to reproduce the behavior:_
1. Extract both case directories from the zip folder attached above
2. Execute the `wecSim` command in the 3DOF $CASE directory
3. The 3DOF simulation returns no error and the results appear acceptable
4. Now, execute the `wecSim` command in the 6DOF $CASE directory
5. The above error appears in the command line

I run WEC-Sim from the command line and have customised `wecSimInputFile.m` to increase the domain size.

**Expected behavior**
The 6DOF simulation should also execute without any errors, since all other parameters remain the same.

**Screenshots**
1. Cylinder half-geometry (up to the waterline) - mesh generated and used by NEMOH.
[image]

2. Simulink setup for the 3DOF simulation.
[image]

3. 3DOF simulation executes without any errors.
[image]

4. Simulink setup for the 6DOF simulation.
[image]

5. The command line throws up an error for the 6DOF simulation.
[image]

**Workstation Information (this section MUST BE completed):**
 - OS: Windows 11
 - MATLAB Release: 2023b Home
 - WEC-Sim Release: v6.0

**Additional context**
N/A

ttammine, 2023-12-14:
I have the same issue. I am also using Nemoh with a similar workstation setup. I have been simulating rolling of a hull. When I use one PTO to allow rolling and then two fixed constraints for lateral movement the simulation runs and rolling seems reasonable. But the constraint forces are huge. Thus, the model must be far from being in balance and I believe that's why the mathematics are not working.   But I have no explanation why the model is not balance or how to balance it.

ajay-menon-iitkgp, 2023-12-19:
Hi there! Quick update: the problem still persists. Since posting this, I've tried debugging the issue by diving into the code for the Simulink 3DOF/6DOF blocks. Since this issue only occurs when I use a 6DOF component, I figured the issue must be related to:

- How I setup the problem in Nemoh, or
- How Nemoh processes the other 3DOFs.

Unfortunately, nothing stands out and I'm not making much headway. Looking forward to the WEC-Sim team's thoughts on why this error might be cropping up. Cheers!

salhus, 2023-12-19:
@ajay-menon-iitkgp @ttammine 

Hi,

Sorry I couldn't get to this sooner.
Some thoughts,

1. The mesh has issues, _viz_., The mesh is not closed and has inconsistent panels.
[image]

I would recommend using a mesh that has quadrangularoid panels.

2. I noticed that the IRFs are not clean, making the time-domain unstable,
[image]
The IRFs are not clean because, the hydrodynamic coefficients are truncated improperly,
[image]

I would recommend running the BEM solver to at least $15-20 rad/s$ or when the hydrodynamic coefficients converge. I would also strongly advise to conduct a mesh convergence study and checking the IRFs.

Cheers,
Sal

--
Salman Husain, M.Sc., Ph.D.
National Renewable Energy Laboratory (NREL)

ajay-menon-iitkgp, 2023-12-21:
Thank you for the comments, @salhus.

### Improving the Mesh:
I was using FreeCAD to model the body, but it is frankly quite tough to generate and refine meshes. So, I've generated the mesh using Capytaine, clipped the region above the free surface (`keep_immersed_parts()`), exported it into a .stl file, and used BEM Rosetta to obtain the .dat format. A direct export to .dat from Capytaine uses Diodore, and I wasn't too sure about using it, so I took the more circuitous route. Plus, I need the .stl file for the WEC-Sim geometry folder.

The panels are quadrilateral and the overall mesh quality is better refined now. I conducted a mesh convergence study in Capytaine, and this resolution appears to yield stable BEM results.

[image]

**Question:** Could you please elaborate on what you meant by the mesh isn't closed? I thought tools such as Nemoh required the top panels below the free surface to be removed?

### Increasing the Frequency Range & Resolution:
I have increased the frequency range to 0.1-30 rad/s with 1000 components. The Nemoh plots for IRFs, added mass, and radiation damping are:

[image]

### New Error - Possibly due to Non-linear Hydrodynamics:
Unfortunately, the issue with the **constraint error still persists**. As you pointed out in capytaine/capytaine#444 , I need to use non-linear hydrodynamics because of the characteristic length of the cylinder. I implemented this in `wecSimInputFile.m` by opting to use viscous damping and have added the following code to the 'Body' section:
```
body(1).nonlinearHydro = 2;                                   % Non-linear hydro on/off
body(1).quadDrag.cd = [1 0 1 0 1 0];                          % Quadratic drag - drag coefficient (assumed)
body(1).quadDrag.area = [0.05 0 0.36 0 0.36 0];               % Quadratic drag - characteristic area (approximated)
```

I've also generated a fine mesh to compute the non-linear pressure forces accurately. However, the problem seems to be with how I'm implementing non-linear hydrodynamics. Now, I get the same error for both 3DOF and 6DOF systems. I tried different combinations of regular/irregular waves with 3DOF/6DOF constraints, and have summarized the results it below.

| Case                    | 3-DOF (linear)                        | 6-DOF (linear) | 3-DOF (non-linear) | 6-DOF (non-linear) |
| ------------------ |:--------------------------------:|:----------------:|:--------------------:|:----------------------:|
| Regular wave      | Executes with absurd results | FAIL                  | FAIL                        | FAIL                         |
| Irregular wave    | Executes with absurd results  | FAIL                  | FAIL                        | FAIL                         |

By absurd results for the 3DOF-linear cases, I mean the body oscillates in an uncontrolled fashion, and that likely occurs because there is no non-linear damping applied (Morison element/viscous damping).

My new WEC-Sim files are below. Capytaine and Nemoh both yield similar results, so I've provided Capytaine's BEM output in the hydroData folder since the Nemoh output is over 2Gb (courtesy of the pressure files).
[Cylinder - 6DOF-updated - Copy.zip](https://github.com/WEC-Sim/WEC-Sim/files/13740391/Cylinder.-.6DOF-updated.-.Copy.zip)

The Capytaine python files used to generate the hydro data & export the mesh are in the hydroData folder.

Thank you for taking the time to review these files. On my end, I will try alternate approaches to modelling the problem in WEC-Sim.

Cheers!

salhus, 2024-01-04:
Hi @ajay-menon-iitkgp ,

Apologies for the delayed response, as I was on time-off.

re: Mesh and hydrodynamic coefficients,
--------------------------------------------

The process you describe, and the coefficients all look acceptable. By the surface panels comment I meant that previously the surface had panels between the diagonal and the surface piercing perimeter of the horizontal cylinder.

re: WEC-Sim inputs and linear hydrostatics,
-----------------------------------------------

I could make the model work with linear hydrostatics, and smaller wave amplitudes. The thing is that a wave amplitude of $1 m$ is just too large compared to the diameter of the cylinder. With an amplitude that large, none of the BEM assumptions hold. An amplitude of $\approx 0.1 m$ works but still the highest you want to be going given that is the radius of the cylinder, also the cylinder itself is only $1.8 m$ long which shorter than even the crest to trough distance of $2 m$ for an input wave of $1 m$ amplitude.

The other thing I would mention is that the cylinder is pinned to its center, which would inevitably result in a torque about the central hinge. I am not sure how/what real system this emulates -- and if it is realistic at all. Given the high buoyancy of the cylinder ($\rho_{cylinder} = 0.5\rho_{water}$), and its length, any input at the bow of the cylinder will result in such behavior, and one doesn't need WEC-Sim or modeling tools to predict that.

re: Non-linear hydrostatics,
------------------------------

I am not confident why, but I think this is due to the scaling of the stl file. FreeCAD I am afraid is encoding the units in $mm$. I made a new .stl file and I think the MoIs needed an update. And the model seems to run (using small amplitude obv.).
I am attaching the modified folder, which seems to work for linear and non-linear hydrostatics.
[Cylinder - 6DOF-updated - Copy.zip](https://github.com/WEC-Sim/WEC-Sim/files/13834489/Cylinder.-.6DOF-updated.-.Copy.zip)
Hope that helps.

Cheers,
sal

--
Salman Husain, MSc, PhD.
National Renewable Energy Laboratory.

salhus, 2024-01-05:
@ajay-menon-iitkgp 
Just to confirm. This issue can be closed?

ajay-menon-iitkgp, 2024-01-08:
@salhus, Apologies for not being able to get back earlier. I'll run a simulation using the new files you have supplied and will revert ASAP.
I can't believe I missed out the logic behind why the simulations weren't running (relative dimensions) and your comments seem to pin-point the reason. Thank you!

ajay-menon-iitkgp, 2024-01-09:
@salhus Thank you very much for your comments. I've made the changes you recommended (scaling down the wave height) and have obtained reasonable results. The files you sent were also very helpful. As you pointed out, the issue was with the relative size of the wave and body. Tuning the wave height and period to within the linear region allows me to use this with multiple bodies and the 6DOF constraint as well. That was an oversight on my part when setting up the simulation.

I will close the issue as solved. Good day!

closed 2024-01-09 by ajay-menon-iitkgp