WEC-Sim/WEC-Sim #1180: [WEC-Sim Applications]When simulating the moorDyn the programme runs with no red errors. but it finally says that "error loading the library "Lines.dll" And the specific module has not found. How to overcome?

HareendraAmaratunge, 2023-12-08:
[image]
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**
_Please identify the WEC-Sim applications case in question._

**Describe the bug or issue**
_A clear and concise description of what the perceived bug or issue is._

**To Reproduce**
_Steps to reproduce the behavior:_

1. _Go to '...'_
1. _Click on '....'_
1. _Scroll down to '....'_
1. _See error_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Workstation Information (this section MUST BE completed):**
 - OS: [e.g. iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html) [2019b, 2020a, 2020b, 2021a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html) [v3.1, v4.0, v4.1]

**Additional context**
_Add any other context about the problem here._

hachikoi1, 2023-12-08:
To use MoorDyn in WEC-Sim, please

1. Download moorDyn from repo https://github.com/WEC-Sim/moorDyn
2. Place all the files and folders under WEC-Sim/source/functions/moorDyn folder

jtgrasb, 2023-12-11:
@HareendraAmaratunge Thank you for your question. As @hachikoi1 said, you can download the MoorDyn library, which is configured to work with WEC-Sim, from the link provided. 

As for your specific error, I believe you are working with an old version of WEC-Sim. You should update your WEC-Sim version to v6.0, which will use the correct library `libmoordyn.dll`.

jtgrasb, 2024-01-03:
@HareendraAmaratunge have your issues been resolved? If so, I will close this issue.

HareendraAmaratunge, 2024-01-04:
Thanks for sending me a solution. Due to the heavy work of other studdies,
I am still unable to stick into the WEC Sim. Can you please give sometime
until this January end, then I ll tell you the progress.
Thank you!

jtgrasb, 2024-01-04:
No problem. I will close this issue for now. Please feel free to reopen this if you still have problems or open a new issue.

closed 2024-01-04 by jtgrasb

HareendraAmaratunge, 2024-01-04:
Okey. Thanks for the support. i ll contact you again if there would any
problmes